morrownr/USB-WiFi #219: Need help with RLT8832AU drivers

smeep0, 2023-02-24:
I coudln't seem to find proper Linux drivers for this adapter. Is there any place or possibility I can get this USB WiFi adapter to work on something like Kali Linux?

bjlockie, 2023-02-24:
https://github.com/lwfinger/rtl8852au

But it is an extremely poor driver and may not compile on recent kernels.
Realtek is really bad so I would return it if you can and buy one with a mediatek chip.

morrownr, 2023-02-24:
Hi @smeep0 

@bjlockie speaks words of wisdom. You have several good chipsets available, this is not one of them.

codebling, 2023-02-25:
@smeep0 You'll need to install Kali's linux kernel header source package, then download the driver from the repo mentioned above and compile it. If you *do* get it to compile, cross your fingers and hope that the driver loads and recognizes your device. If it does, you are in business, and now you can look forward to all of the problems listed in **morrownr**'s [review in the other thread you posted in](https://github.com/morrownr/USB-WiFi/discussions/25). Since you're running Kali, it may be worth noting that you'll likely not get monitor mode to work with that device, but best of luck to you. 

Do yourself a favor and get yourself a [Mediatek MT7922u or MT7921au](https://github.com/morrownr/USB-WiFi/issues/87)!