morrownr/USB-WiFi #221: Help with MT7922 card

fs30000, 2023-02-25:
Hello all,

I got a MT7922 PCIe card, but i'm having trouble make it work.  hostapd fails with:

-------------------------------------
wlp4s0: interface state UNINITIALIZED->COUNTRY_UPDATE
Frequency 5180 (primary) not allowed for AP mode, flags: 0x853 NO-IR
Primary frequency not allowed
wlp4s0: IEEE 802.11 Configured channel (36) or frequency (5180) (secondary_channel=1) not found from the channel list of the current mode (2) IEEE 802.11a
wlp4s0: IEEE 802.11 Hardware does not support configured channel
Could not select hw_mode and channel. (-3)
wlp4s0: interface state COUNTRY_UPDATE->DISABLED
wlp4s0: AP-DISABLED
wlp4s0: interface state DISABLED->DISABLED
wlp4s0: AP-DISABLED
wlp4s0: CTRL-EVENT-TERMINATING
hostapd_free_hapd_data: Interface wlp4s0 wasn't started
nl80211: deinit ifname=wlp4s0 disabled_11b_rates=0
------------------------------

I uncommented the respective lines in hostapd.conf for MT7921, which should work for this MT7922 (but does not):

------------------------------------------------------
#
interface=wlp4s0
driver=nl80211
bridge=br0

ssid=1
wpa_passphrase=12345678
hw_mode=a
channel=36
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42
country_code=PT
# Advertises the country_code and the set of allowed channels and transmit power levels
ieee80211d=1
# Enables support for 5GHz DFS channels (requires ieee80211d=1)
ieee80211h=1

macaddr_acl=0

# Security
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK

# Note: Capabilities can vary even between adapters with the same chipset.
#
# Note: Only one ht_capab= line and one vht_capab= should be active. The
# content of these lines is determined by the capabilities of your adapter.
#
# IEEE 802.11n
ieee80211n=1
wmm_enabled=1
#
# mt7921au
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]

# IEEE 802.11ac
ieee80211ac=1
#
# mt7921au (VHT Cap. 339071b2)
vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][SU-BEAMFORMEE][BF-ANTENNA-4][MAX-A-MPDU-LEN-EXP7][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
----------------------------------------------------------

[... 670 lines cut ...]
                 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * mesh point: 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-device: 0x40 0xd0
        Supported extended features:
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
                * [ BEACON_RATE_HT ]: HT beacon rate setting
                * [ BEACON_RATE_VHT ]: VHT beacon rate setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support`

Any ideas guys?

P.S. - I tried to add the output as code, but it takes out the spaces, making it more unreadable.

fs30000, 2023-02-25:
[iw list MT7922.txt](https://github.com/morrownr/USB-WiFi/files/10831638/iw.list.MT7922.txt)
[hostapd.conf.txt](https://github.com/morrownr/USB-WiFi/files/10831639/hostapd.conf.txt)

bjlockie, 2023-02-25:
Do you need wpa3 for wifi6?

morrownr, 2023-02-26:
> Oh my apologies to this formating mess. How can i past the config file without github formating it? 

```
I did this to help. See how I put the little whachamacallums above and below the test?
```

> I got a MT7922 PCIe card, but i'm having trouble make it work. hostapd fails with...

Okay. This is like being taken out of elementary school so you can start work on a PhD.

I just saw where the mt7922e cards were shipping recently. What I think I know about the mt7922:

- it is the mt7921 except that it supports 160 MHz channel width
- we may see a mt7922u at some point but let's not forget that USB is always last
- the mt7922 is supported by the mt7921 driver
- the mt7922 has its own firmware (not the same as the mt7921) (have you upgraded it?)

Does the adapter work in managed mode? If so, which bands have you tested?

What does `iw dev` show?

You must have the driver going because I see:

```
Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
```

So, that seems to indicate AP mode is supported.

> modinfo shows only 1 parameter possible to change: disable_aspm. Any ideas guys?

Not a clue at this point. Cross that bridge when we need to.

Let me suggest this: How about you start with a WiFi 4 2.4 GHz band AP setup and then when that is working well, move on to the next step.

```
ssid=my2g
wpa_passphrase=myPW
hw_mode=g
channel=6
country_code=PT

bridge=br0
interface=wlp4s0
driver=nl80211

auth_algs=3
macaddr_acl=0
ignore_broadcast_ssid=0
wpa=2
wpa_pairwise=CCMP
#rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK

# IEEE 802.11n
ieee80211n=1
wmm_enabled=1
#
# generic 20 MHz setting
ht_capab=[SHORT-GI-20]

```

You did not say what hardware or distro you are using. Enlighten us?
What guide are you using?
Have you looked up to see which bands and channels are supported in your country? regdb?

fs30000, 2023-02-27:
> Do you need wpa3 for wifi6?

Hmm no. Unless there's something wrong with WPA2.

morrownr, 2023-02-27:
> Do you need wpa3 for wifi6?

No. WPA3 is mandatory for WiFi 6e but not WiFi 6.

morrownr, 2023-02-27:
> Rocky linux 9.1. Just a mini-pc with this chip on a PCIe card. Guide? 

I have the Bridged Wireless Access Point guide in the Main Menu but I don't know if that is what you want to do.

I am not familiar with Rocky. AP mode, like monitor mode, will have problems if certain procs are left running... Network Manager for example.

Have you thought about trying OpenWRT? If has the mt7921e driver backported to kernel 5.10. You might have to add the firmware but I have a guide up for that.

fs30000, 2023-02-27:
> > Do you need wpa3 for wifi6?
> 
> No. WPA3 is mandatory for WiFi 6e but not WiFi 6.

Ok, so does the mt7921 driver/module support wpa3? Does this mt7922 support it? hostapd?  I'm assuming yes, otherwise they wouldn't market it a 6e capable device, right?

fs30000, 2023-02-27:
> I have the Bridged Wireless Access Point guide in the Main Menu but I don't know if that is what you want to do.
>
> I am not familiar with Rocky. AP mode, like monitor mode, will have problems if [...]

I managed to have a TPLink usb wifi dongle working on another pc with your config. However on this one i'm having trouble have it working. Same hostapd.conf. Same usb dongle. I suspect the kernel module. I'm on 6.2.1. Realtek drivers already on it.

morrownr, 2023-02-27:
> Ok, so does the mt7921 driver/module support wpa3?

Yes. In fact, all in-kernel drivers that I have tested support WPA3. That includes adapters all the way back to ra3070, ra2870, and more. I actually have a lot of adapters.

> Does this mt7922 support it?

Yes

> hostapd?

It depends on the version you are using.

 > I'm assuming yes, otherwise they wouldn't market it a 6e capable device, right?

Well, marketing, logic and truth have a loose relationship.

> I managed to have a TPLink usb wifi dongle working on another pc with your config. However on this one i'm having trouble have it working. Same hostapd.conf. Same usb dongle. I suspect the kernel module. I'm on 6.2.1. Realtek drivers already on it.

The guilty parties might be wpa_supplicant or Network Manager or something else in the distro. You can run the kernel up to the latest but that does not upgrade others things are needed to be modern for WPA3 and other modern things to work. My basic advice is if you have a distro that was released after mid-2022, you should be good. There are exceptions. What are you using?

fs30000, 2023-02-27:
> > Ok, so does the mt7921 driver/module support wpa3?
> 
> Yes. In fact, all in-kernel drivers that I have tested support WPA3. That includes adapters all the way back to ra3070, ra2870, and more. I actually have a lot of adapters.

So, for wap3, i just need to have wpa=2 and wpa_key_mgmt=SAE, right?

 
> 
> > Does this mt7922 support it?
> 
> Yes
> 
> > hostapd?
> 
> It depends on the version you are using.

2.10

> Well, marketing, logic and truth have a loose relationship.
>
>
> The guilty parties might be wpa_supplicant or Network Manager or something else in the distro. You can run the kernel up to the latest [...]

Downgraded the kernel to 6.1.9, which doesn't have the realtek drivers and compiled your drivers, 88x2bu-20210702, the tplink dongle is working. Not at full blast, but it's ok. However, i would like to have this mt7922 card working. Maybe the mediatek drivers in the kernel 6.2.1 suck just like the realtek ones?

fs30000, 2023-02-28:
So now the card works. No idea why it started working. Anyway here is my hostapd.conf file. 

`#
interface=wlp4s0
bridge=br0
driver=nl80211

ssid=teste
wpa_passphrase=12345678
country_code=PT

auth_algs=3
macaddr_acl=0
ignore_broadcast_ssid=0
wpa=2
rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK
#wpa3
#wpa_key_mgmt=SAE

# Advertises the country_code and the set of allowed channels and transmit power levels
ieee80211d=1
# Enables support for 5GHz DFS channels (requires ieee80211d=1)
ieee80211h=1

#############################
############################# SPEEDS
#g --> 2GHz   a --> 5Ghz
hw_mode=a
channel=36

########################### 2GHz

# IEEE 802.11n
ieee80211n=1
wmm_enabled=1

# mediatek
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]

########################## 5GHz
ieee80211ac=1

# 0 = 20 or 40 MHz operating Channel width
# 1 = 80 MHz channel width
# 2 = 160 MHz channel width
# 3 = 80+80 MHz channel width
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42

#mediatek
vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][SU-BEAMFORMEE][BF-ANTENNA-4][MAX-A-MPDU-LEN-EXP7][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][VHT160][SHORT-GI-160]
`

So i added [VHT160][SHORT-GI-160] there. No 160Mhz connection. Both phone and laptop connect using 802.11ac, no ax. going to search here for a hostapd.conf with 160Mhz enabled. Both phone and laptop are wifi6 capable. 

Tried wpa_key_mgmt=SAE for wpa3, no connection for phone and laptop (intel ax201)

Tried vht_oper_chwidth=2 and 3, hostapd doesn't start.

morrownr, 2023-02-28:
> So, for wap3, i just need to have wpa=2 and wpa_key_mgmt=SAE, right?

No, there are 4 things I check:

```
auth_algs=3
wpa=2
wpa_key_mgmt=SAE
ieee80211w=2
```
ieee80211w is mandatory for WPA3 so it has to be on.

Here is a longer explanation:

```
# Security
macaddr_acl=0
# auth_algs=3 is required for WPA3-SAE and WPA3-SAE Transition mode
auth_algs=3
ignore_broadcast_ssid=0
# wpa=2 is required for WPA2 and WPA3 (read the docs)
wpa=2
wpa_pairwise=CCMP
#wpa_pairwise=CCMP CCMP-256
rsn_pairwise=CCMP
#rsn_pairwise=CCMP CCMP-256
#group_cipher=CCMP
# only one wpa_key_mgmt= line should be active.
# wpa_key_mgmt=WPA-PSK is required for WPA2-AES
#wpa_key_mgmt=WPA-PSK
# wpa_key_mgmt=SAE WPA-PSK is required for WPA3-SAE Transition mode
#wpa_key_mgmt=SAE WPA-PSK
# wpa_key_mgmt=SAE is required for WPA3-SAE
wpa_key_mgmt=SAE
#sae_pwe=2
#wpa_group_rekey=1800
# ieee80211w=1 is required for WPA3-SAE Transition mode
# ieee80211w=2 is required for WPA3-SAE
ieee80211w=2
# sae_require_mfp=1 is required for WPA3-SAE Transition mode
#sae_require_mfp=1
# If parameter is not set, 19 is the default value.
#sae_groups=19 20 21 25 26
# If parameter is not set, 5 is the default value.
#sae_anti_clogging_threshold=10
```

fs30000, 2023-02-28:
Ok, but i'm giving up on this card. I guess the drivers are still bad. I used the 6.2.1 kernel and latest firmware.  The connection is kind of "funky",  also 20ms+ pings. Sometimes hostapd does not start, etc. No wpa_supplicant and Networkmanager was stopped.

Funny because you were saying mediatek is more stable than realtek. I'm finding the exact opposite. The TPLink dongle i got, is much more stable than this card and comfast 953-ax. 2-3ms pings. Plus, i'm compiling the module for Red hat original kernel, 5.14, not mailine. That comfast dongle is even worse than the card. Complete disapointment. I tried, i really tried.

Maybe i screwed up with the PCIe card, because the antenna plugs were so damn tiny, but at a time iperf3 was showing 40-50MB/s, normal and reverse. So afterall the antennas are correctly working. As for the comfast dongle, it has no excuse to suck so much :) Very unstable.

Rocky Linux 9.1
elrepo kernel-ml 6.2.1, it can't be more recent than this.

**Realtek 2 Mediatek 0**

morrownr, 2023-02-28:
> Red hat original kernel...

> Rocky Linux 9.1

This could explain a lot. There is a reason that I specify that Red Hat/RHEL and any distro that is based on RHEL is not supported by the Realtek drivers we have up here. There two reasons:

- Red Hat does some things with their kernels that make it very difficult to support with out-of-kernel drivers. There are many good distros for running hostapd, RHEL is not one of them.

- Red Hat prioritizes extreme stability. That is good for their target market. Running hostapd is about as far away from Red Hat's target market as you can get. A large part of that stability comes from basing on kernels and other parts that are old. That makes trying get WiFi6/6e and WPA3 going almost impossible...without a deep dive into the various components that support those functions. I'm amazed that you got as far as you did. There is far more than a stable driver that is required to make what you were trying to do work.

> Maybe i screwed up with the PCIe card

I don't know that this is what happened as I don't think you can properly judge that based on the distro you are using. I've been running Linux since 1994 and if you asked me to do what you are trying to do with RHEL, I would just break out in a cold sweat and take off running. 

What I have found is that if users that have a specific project in mind , they should ask for suggestions first, there are those here than can make recommendations that are likely to increase the chances of success. Do I think you can have success with your card? Yes, but it is not likely right now due to the use of bleeding edge hardware on a distro that can be problematic with 10 year old wifi hardware on your end and a lack of knowledge of what you are really wanting to accomplish on our end.

If you decide to give things another try, first explain what end goal/end state you are trying to accomplish and we can see where it goes.

Good luck

fs30000, 2023-03-02:
So, ok. My end goal is to set an AP. 5Ghz only.  Decent speeds, decent pings.

I have 3 devices i can use:

- MT7922  m.2 card (i said before it was pcie, my mistake)
- Comfast CF-953AX with MT7921
- TP-Link T3U Plus

MT7922:
Even tough i'm running Rocky 9.1, the kernel, as i said, was mainline 6.2.1, released one week ago or something. So, the argument  Red Hat is playing safe, does not apply here.
 The card/glue patched antennas are correctly installed, i got 50 MB/s on iperf. And again, these are crap antennas. However pings were crap. When i modprobed the module with disable_aspm, the system freezed. I find it strange, the modules have just one or two parameters.

Comfast:
Too much crapness, very unstable connection, sometimes it didn't even connected. On hostapd.conf i used the config for mt7921. Cheap is cheap i guess.

TPlink:
Realtek sucks eh? With your drivers 88x2bu-20210702, not here. I'm using Rocky/Red Hat's default kernel,  5.14.0-162.18.1.el9_1.x86_64. Pings are low and stable, speeds are enough.

Realtek wins. Mediatek can update their drivers as often as they want. For AP, the chips suck. In my experience, mediatek always sucked, i have tried phones with their chips. They sucked.

Maybe i'll try qualcomm later. Pitty intel does not support AP in 5Ghz.

Apologies if i sound mad, that's because i am :) I bought the mediatek stuff, because of recomendations here.

morrownr, 2023-03-03:
> Apologies if i sound mad, that's because i am :)

I get that. It happens.

I started this site with a primary goal of getting good information to the Linux community. WiFi is a complicated subject and there are many ways that USB WiFi adapters can be used. At times, I have to make changes to the information that is available. If the information here has caused you to see less than excellent results, I'm all for getting to the reason so that the information here can be modified.

> My end goal is to set an AP. 5Ghz only. Decent speeds, decent pings.

The speeds and pings depend on a lot of things besides driver and wifi chipset.

If I know your country then we can look at what 5 GHz channels are available to you?
Have you taken a look at the congestion levels on the various 5 GHz channels?
Are you able to use DFS channels? IOW, do you have clients that can't use DFS?

I have the following speed test up:

https://github.com/morrownr/USB-WiFi/blob/main/home/Speed_Comparison_Test.md

That test is conducted on a DFS channel that has no other AP operating. It is clean air. If I were to do the same tests on channel 36 or 149, oh my, the results would look much different because there are other APs working. Remember that WiFi is a technology based on sharing.

Should we perform an analysis to see what channels would be best for you to use?
I rescanned this thread but do not see what kind of computer you are using?
Are you locked into Rocky or are you open to other distros?

morrownr, 2023-03-03:
Ping test:

Adapter based on the 8812bu chipset (like your TP-Link adapter)

```
64 bytes from 192.168.1.1: icmp_seq=10 ttl=64 time=2.16 ms
64 bytes from 192.168.1.1: icmp_seq=11 ttl=64 time=2.38 ms
64 bytes from 192.168.1.1: icmp_seq=12 ttl=64 time=2.67 ms
64 bytes from 192.168.1.1: icmp_seq=13 ttl=64 time=2.50 ms
64 bytes from 192.168.1.1: icmp_seq=14 ttl=64 time=1.86 ms
64 bytes from 192.168.1.1: icmp_seq=15 ttl=64 time=2.34 ms

```

Adapter based on the 7921au chipset (like your Comfast adapter)
```
64 bytes from 192.168.1.1: icmp_seq=21 ttl=64 time=1.48 ms
64 bytes from 192.168.1.1: icmp_seq=22 ttl=64 time=2.21 ms
64 bytes from 192.168.1.1: icmp_seq=23 ttl=64 time=1.53 ms
64 bytes from 192.168.1.1: icmp_seq=24 ttl=64 time=2.50 ms
64 bytes from 192.168.1.1: icmp_seq=25 ttl=64 time=1.91 ms
64 bytes from 192.168.1.1: icmp_seq=26 ttl=64 time=2.70 ms

```

Not much difference.

Here is the result from my Powerline AV2 adapters

```
64 bytes from 192.168.1.1: icmp_seq=14 ttl=64 time=3.90 ms
64 bytes from 192.168.1.1: icmp_seq=15 ttl=64 time=3.10 ms
64 bytes from 192.168.1.1: icmp_seq=16 ttl=64 time=3.82 ms
64 bytes from 192.168.1.1: icmp_seq=17 ttl=64 time=3.21 ms
64 bytes from 192.168.1.1: icmp_seq=18 ttl=64 time=3.18 ms
64 bytes from 192.168.1.1: icmp_seq=19 ttl=64 time=3.22 ms
64 bytes from 192.168.1.1: icmp_seq=20 ttl=64 time=3.07 ms

```

Not as good but not that bad. Wireless is faster.

fs30000, 2023-03-07:
The problem with the mediatek stuff, was not the bands. I was using the same hostapd.conf file i use for the tplink adapter, just with ht and vht settings specific for the mediatek stuff. tplink works ok, mediatek not. Latest drivers and firmware. Pc is a mini-pc.

In resume, with the m.2 card, pings were high, but speeds were actually good, 50 MB/s. Strangely enough, my phone showed a 5 icon instead of 6. My laptop too, showed AC instead of AX. The chip is supposed to support wifi6/ax.

The comfast adapter was just crap, sometimes hostapd wouldn't even start. It was bad.

I'm happy with tplink, maybe later i'll try again with mediatek.

morrownr, 2023-03-07:
> The chip is supposed to support wifi6/ax.

It does.

That does not mean it is easy to configure for those without a high level of skill yet. For hostapd, you need to download and compile the dev version in order for AX to even be available to be turned on. Not to mention numerous others .config settings that have to be correct. You can't just install `hostapd` and expect it to work with WiFi 6 yet. I have a guide about how to do this.

The bottom line is that the hardware and driver support WiFi 6 and WiFi 6e in AP mode. Very few distros have versions of hostapd and other requires things that are at levels to support it yet. You have to compile you way to it. The latest RasPiOS was released about 2 weeks ago. It STILL does not have a version of hostapd that will support WiFi 6. I had to compile hostapd for my Pi4B to use my 951AX in WiFi 6 mode. Hell, I had to compile the entire kernel as the RasPiOS still comes with kernel 5.15.

In Main Menu item 2, I wrote at length telling anyone that read it that if a person wants plug and play, they need to move on down to the AC1200 section. This still holds true. If you are technically proficient, go for it. Otherwise, stay away. This is a hard upgrade because 3 things are happening at once: Wifi 6, WPA2 and WiFi6. This is not a Linux, Windows or MacOS problem. It is an overall problem because of the complexities.

The tp-link adapter you are comparing with is a WiFi 5 adapter. You are comparing apples to oranges. Grab a WiFi 6 usb Realtek adapter and compare it to the 953AX. If you want something to compare to the tp-link adapter, grab and Alfa ACM and tell me which adapter is easier to use and maintain after about 6 months. That would be comparing apples to apples.

Cheers

morrownr, 2023-03-09:
>Can you please share the guide's page? Already compiled hostapd, but don't know the settings for wifi6.

[Upgrade_hostapd.tar.gz](https://github.com/morrownr/USB-WiFi/files/10928296/Upgrade_hostapd.tar.gz)

The guide is tested on Raspberry Pi OS which means it is Debian oriented but you can likely work through that. Ask if you don't understand something.

I just thought about this: you will need to compile wpa_supplicant` as well to get WiFi 6 support. Need a guide? This depends on your distro as Ubuntu 22.10 and later are all moved from wpa_supplicant to IWD so this doesn't apply.

WiFi 6e is likely a bridge too far right now. Have you checked reg.db to see if the 6e band is legal in your country? This is a real problem around the world as many countries around the world have not even passed laws making 6e legal yet. The EU seems to be far ahead of the rest of us.

Right now the goal should be getting WiFi 6 going. 6e is another level of complexity. I'm not even up to speed on 6e in AP mode yet. I'm working on it. 

My day job is rocket science but this is far more difficult. Just kidding about my day job but not about the level of difficulty. My guess is that this stuff will mostly be sorted out by the time we see Ubuntu 24.04. We have good chipsets with the mt7921 and mt7922 and the Linux driver is already good and improving. It is all the other stuff that is coming up to speed slowly. Remember that you are on the bleeding edge... me too.

bjlockie, 2023-03-16:
What is "normal" and "reverse"?
How do you get MB/s from iperf, I get Mbit/sec.

fs30000, 2023-03-16:
Even tough, iw list shows this card supports 160Mhz and 80+80, it fails with vht_oper_chwidth=2 and 3, for 160Mhz and 80+80 Mhz respectively.

````
Configured VHT capability [VHT_CAP_SUPP_CHAN_WIDTH_MASK] exceeds max value supported by the driver (2 > 1)
````

On  vht_capab it accepts [SHORT-GI-160][VHT160], but not [VHT160-80PLUS80].

fs30000, 2023-03-16:
> What is "normal" and "reverse"? How do you get MB/s from iperf, I get Mbit/sec.

I'm talking about iperf3.
iperf3 -f M -c server will show in MB/s.

In normal mode, iperf client sends data, with reverse (-R), iperf downloads from server.
iperf3 -f M -c server -R

morrownr, 2023-03-17:
> iw list shows this card supports 160Mhz and 80+80

I think I remember seeing a patch 1-2 weeks ago in linux-wireless that removed 80+80. I think the discussion was that it supports 160 but not 80+80, at least at this time. So, that tells me the capability was added by mistake.

FYI: There is a lot of ongoing work on the mt7921 driver. We should see P2P capability in the wild soon. Since last fall, we have been seeing new firmware files coming monthly... never seen that before but then wifi drivers have never been this complicated before. The various versions of the mt792x chipset are going into stuff all over the place... TV's, etc.

Astrako, 2023-03-31:
Hi @fs30000!

I saw you in your hostapd example than you've set Portugal as country and it has same bands locked as Spain (where I am).... May I ask you how you were able to start hostapd in channel 36? I saw you in your first log "wlp4s0: IEEE 802.11 Hardware does not support configured channel"

I ask because I have a AX210 card and I just can start hostapd for 5GHz in channels from 149 to 157 or similar, but in this case is due Intel shitty LAR implementation. And these channels have a very limited power output so they are useless for a home made router.

I was thinking on buying a mt7922 card in aliexpress but I don't know if it will work well as I want.

Thanks in advance!

fs30000, 2023-04-03:
Yes, 36 works here. In the begining it was failing because i did't have the right config done. I have it now, working fine. However just wifi5, no AX. For wifi6, like Nick said, you need a recent build of hostapd. 5 is enough for me.

````
interface=wlp4s0
bridge=br0
driver=nl80211

ssid=1
wpa_passphrase=pass
country_code=PT

auth_algs=3
macaddr_acl=0
ignore_broadcast_ssid=0
wpa=2
rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK

# Advertises the country_code and the set of allowed channels and transmit power levels
ieee80211d=1
# Enables support for 5GHz DFS channels (requires ieee80211d=1)
ieee80211h=1

hw_mode=a
channel=36

ieee80211n=1
wmm_enabled=1

#mediatek pcie
ht_capab=[RXLDPC][HT40+][SHORT-GI-20][SHORT-GI-40][GF][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40][SU-BEAMFORMEE]

ieee80211ac=1
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42

#mediatek pcie
vht_capab=[RXLDPC][HT40+][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40][MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][SHORT-GI-160][VHT160]
````

Maybe i could tweak it even more, but it's enough for me.

UnderEu, 2023-09-26:
> EDIT: I got it at this link: https://www.aliexpress.com/item/1005005042304492.html
>
I bought the exact same card but I'm getting slow speeds (for a Wi-Fi 6 card, I mean).

I used this exact conf as a template (modifying the interface id and country code, only) and I'm getting ~150Mbps during iperf3 tests from Wi-Fi 6 client to the host, in both directions (send / receive).

```
interface=wlp1s0
bridge=br0
driver=nl80211

ssid=Test
wpa_passphrase=87654321
country_code=BR

auth_algs=3
macaddr_acl=0
ignore_broadcast_ssid=0
wpa=2
rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK

# Advertises the country_code and the set of allowed channels and transmit power levels
ieee80211d=1
# Enables support for 5GHz DFS channels (requires ieee80211d=1)
ieee80211h=1

hw_mode=a
channel=36

ieee80211n=1
wmm_enabled=1

#mediatek pcie
ht_capab=[RXLDPC][HT40+][SHORT-GI-20][SHORT-GI-40][GF][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40][SU-BEAMFORMEE]

ieee80211ac=1
ieee80211ax=1
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42

#mediatek pcie
vht_capab=[RXLDPC][HT40+][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40][MAX-MPDU-11454][SHORT-GI-80][TX-STBC-2BY1][SU-BEAMFORMEE][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][SHORT-GI-160][VHT160]
```

I tested the card in client mode and I get top speed from my previous Wi-Fi 5 (2x2) hotspot = ~600Mbps

Also tested different distros and another host machine, just in case, but the results were the exact same. My primary goal was using it on OpenWrt but I can't manage to get it working - I want to believe it's because 23.05 is on rc stage, didn't test on 22.03.

morrownr, 2023-09-27:
Hola @UnderEu 

> I tested the card in client mode and I get top speed from my previous Wi-Fi 5 (2x2) hotspot = ~600Mbps

I am seeing 750 - 800 Mbps (iperf3) with my mt7922e pcie card. It is in managed mode running WiFi 6 (not 6 Ghz) on channel 36 but that channel is has some congestion. It is showing 160 Mhz channel width but with the congestion and my router, which only has a dual core 1350 Mhz processor, the limitation is likely not the wifi.

With that said, I'll suggest that for AP mode, you limit yourself to the following right now:

- 5 Ghz
- WiFi 6
- 80 Mhz channel width

A user posted the other day that right now the mt7922 does 160 Mhz channel width in managed mode but on 80 in AP mode. I ran `iw list` and he is right. I had not tested mt card in AP mode yet. I suspect that 80 is not the goal as the driver and firmware are still under heavy work. I saw some patches going into linux-wireless today that made me think.

UnderEu, 2023-09-30:
Did more testing during the last days, found some client devices with newer (better) Wi-Fi and here's what I got:
** The client devices were placed right on the side of the AP host, to give the best numbers possible.

While configured for Wi-Fi 6 (ieee80211ax=1), in my iperf3 tests:
- Downstream: 300Mb/s avg., 380Mb/s peak
- Upstream: 320Mb/s avg., 329Mb/s peak

If I disable Wi-Fi 6 (comment ieee80211ax=1, leaving the rest untouched):
- Downstream: 550Mb/s avg., 640Mb/s peak
- Upstream: 90Mb/s avg., 95Mb/s peak

- Latency (ping): 24ms min. with higher variations, sometimes >100ms - both modes (ac and ax)

file conf
```
# Hardware
interface=wlp1s0
bridge=vmbr0
driver=nl80211

# Regulatory
country_code=BR
ieee80211d=1
ieee80211h=1

# WMM Support
wmm_enabled=1

# WMM QoS Low priority / AC_BK = background
wmm_ac_bk_cwmin=4
wmm_ac_bk_cwmax=10
wmm_ac_bk_aifs=7
wmm_ac_bk_txop_limit=0
wmm_ac_bk_acm=0

# WMM QoS Normal priority / AC_BE = best effort
wmm_ac_be_aifs=3
wmm_ac_be_cwmin=4
wmm_ac_be_cwmax=10
wmm_ac_be_txop_limit=0
wmm_ac_be_acm=0

# WMM QoS High priority / AC_VI = video
wmm_ac_vi_aifs=2
wmm_ac_vi_cwmin=3
wmm_ac_vi_cwmax=4
wmm_ac_vi_txop_limit=94
wmm_ac_vi_acm=0

# WMM QoS Highest priority / AC_VO = voice
wmm_ac_vo_aifs=2
wmm_ac_vo_cwmin=2
wmm_ac_vo_cwmax=3
wmm_ac_vo_txop_limit=47
wmm_ac_vo_acm=0

# Radio Mode (5GHz - Wi-Fi 6)
hw_mode=a
ieee80211n=1
ieee80211ac=1
ieee80211ax=1

# Channel Selection (80MHz Channel Width)
channel=149
vht_oper_centr_freq_seg0_idx=155
vht_oper_chwidth=1

# HT Capabilities
ht_capab=[RXLDPC][HT40+][SHORT-GI-20][SHORT-GI-40][GF][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40][SU-BEAMFORMEE]

[... 1317 lines cut ...]
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		* [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
```

iw reg get
```
global
country BR: DFS-FCC
	(2402 - 2482 @ 40), (N/A, 20), (N/A)
	(5170 - 5250 @ 80), (N/A, 17), (N/A), AUTO-BW
	(5250 - 5330 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
	(5490 - 5730 @ 160), (N/A, 24), (0 ms), DFS
	(5735 - 5835 @ 80), (N/A, 30), (N/A)
```

OS: Debian 12 / Proxmox VE 8.0
Kernel: 6.2.16-14-pve

morrownr, 2023-09-30:
I did not see these in your hostapd.conf:

he_oper_chwidth=1
he_oper_centr_freq_seg0_idx=155

I think your current conf is keeping WiFi 6 at 40 Mhz channel width.

Your 6 Ghz channels are disabled as the reg.db is showing them as not authorized in your country so WiFi 6 on 5 Ghz is about the best you can do. It can be fast as I see 750 to 800 Mbps with my mt7922 based card using WiFi 6 on 5 Ghz.

Something to consider: Putting your client next to the AP is probably not going to yield the best performance as there is likely to be signal distortion. I'd recommend no closer than maybe 2 meters.

UnderEu, 2023-09-30:
> I did not see these in your hostapd.conf:
> 
> he_oper_chwidth=1
> he_oper_centr_freq_seg0_idx=155
> 
> I think your current conf is keeping WiFi 6 at 40 Mhz channel width.

Indeed it was but I don't remeber seeing those params in documentation.
Added to the conf, will bring test results later.

> Your 6 Ghz channels are disabled as the reg.db is showing them as not authorized in your country so WiFi 6 on 5 Ghz is about the best you can do. It can be fast as I see 750 to 800 Mbps with my mt7922 based card using WiFi 6 on 5 Ghz.

Outdated database, our regulatory agency approved Wi-Fi 6E to operate since February 2021
https://www.csagroup.org/global-certification-regulatory-update/brazil-anatel-act-no-1306-wifi-6e

Not too worried to use 6GHz right now because I have no 6GHz compatible devices (yet).

> Something to consider: Putting your client next to the AP is probably not going to yield the best performance as there is likely to be signal distortion. I'd recommend no closer than maybe 2 meters.

It's for bench testing purposes, my box will be placed properly after all the testing is done.

morrownr, 2023-09-30:
> Indeed it was but I don't remeber seeing those params in documentation.

I assume you are looking at this example that I have on the Main Menu:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi6.conf

If so, look in the last section, WiFi 6.

> Added to the conf, here's the results: Downstream: 780Mb/s avg., 940Mb/s peak

Looking better. If you can find any improvements, let me know or submit a PR.

> Not worried to use 6GHz right now because I have no 6GHz compatible devices (yet).

Now is the time to get the changes into the db so... can I get the contact info for the db maintainer to you?

> It's for bench testing purposes, my box will be placed properly after all the testing is done.

I guess I was not clear in what I was saying. 

https://www.astro.co.uk/home-wi-fi-network-mistakes/#:~:text=Working%20with%20a%20wireless%20device,a%20detrimental%20affect%20on%20performance.

Working with a wireless device too close to the router can have a detrimental affect on performance.

Like I said, I would not put a client any closer than about 2 meters no matter the situation.

morrownr, 2023-09-30:
@UnderEu

Updating the WiFi Regulatory db

The Maintainer:

https://www.forshee.me/about/

Webpage:

 git://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git

Download:

 git clone git://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git

If you have kernel version 4.15 or newer, the regulatory database is loaded using the same mechanism as the firmware files. The files can be found at

https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/

You'll need two files, regulatory.db and regulatory.db.p7s. Place them into /lib/firmware/ directory.

$ sudo cp regulatory.db /lib/firmware/
$ sudo cp regulatory.db.p7s /lib/firmware/
$ sudo reboot

-----

Getting the regdb updated now would be a really good idea as it can take time for updates to work their way through the system.

UnderEu, 2023-09-30:
> [...]
> Getting the regdb updated now would be a really good idea as it can take time for updates to work their way through the system.

Yes, I manually updated the db files and now it shows all available channels for my region.
```
global
country BR: DFS-FCC
	(2400 - 2483 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 27), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 27), (0 ms), DFS, AUTO-BW
	(5725 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
	(5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
	(57000 - 71000 @ 2160), (N/A, 40), (N/A)
```

I'll continue my testing and let you know if I manage to improve things further.