morrownr/USB-WiFi #225: The list contains no gain information.

navid-zamani, 2023-03-14:
I have a very long range WIFI connection (across the street, two windows and 1-2 drywalls, and sometimes a truck ;), and was looking for a good adapter here. (Other solutions, like moving something closer, are unfortunately unavailable.)

But the list contains no information about the gain achieved by those antennas. Specifically, the dBi (and maybe dBm) value.
So I have no information to find out which adapter has the biggest gain / best reception, let alone the best one for my price limit.

**Can you add gain information?**
(Or just put everything into a big CSV/ODS, so I can filter/sort by a dBi column myself.)

---- 
Side note:
I wish the list was more like [the awesome power of the Geizhals comparison site (translated from German)](https://geizhals-de.translate.goog/?cat=nwpcie&xf=14065_80211-Adapter~14066_extern+kabelgebunden&asuch=dbi&asd=on&bpmin&bpmax&v=e&hloc=at&hloc=de&plz&dist&mail&fcols=17982&fcols=14072&fcols=18669&fcols=14073&fcols=14074&fcols=14080&t=v&sort=p&bl1_id=30&_x_tr_sl=de&_x_tr_tl=en&_x_tr_hl=de&_x_tr_pto=wapp).
Which doesn’t have dBi as a column either, but at least you can search for it, and it is highlighted when you do a comparison.
(It has even more properties to filter by, but hides them when there is no choice anymore due to already chosen filters.)

navid-zamani, 2023-03-14:
E.g. the ALFA AWUS036ACHM, which is said do have a “Very Long” range, has 5dBi but not even any MIMO. I noticed that MIMO makes reception *way* better in my situation. (From 50kB/s with a high-gain antenna USB stick to 1,1MB/s with a bog-standard Fritz!Repeater 300 with no external antenna)

morrownr, 2023-03-14:
Hi @navid-zamani 

https://github.com/morrownr/USB-WiFi/blob/main/home/Performance_Comparison.md

That test is a little dated but the Alfa ACHM and ACH are still the best long range adapters that I am aware of. There is a good probability that Alfa will be releasing a MAX Power WiFi 6e adapter later this year from what I have heard but it will be a tad expensive.

The following antenna combined with an ACHM might provide the results you are looking for. Of course, testing is the only way to know for sure.

https://store.rokland.com/collections/wi-fi-antennas/products/alfa-apa-m25-2-4-5-ghz-dual-band-directional-10-dbi-panel-antenna

@morrownr

navid-zamani, 2023-05-13:
@morrownr:

Just wanted to come back and say your tip was a great one! I bought the AWUS036ACM, and I get 100Mb/s with no issues. Just the standard antennas, using a band with less competition, and taking care with positioning it well. (It *does* have MIMO, btw. Hence two antennas.)

Thank you very much!

navid-zamani, 2023-05-13:
Oh, I forgot that you may have wanted to keep this open as a reminder to add gain information. Hmm … I assume you are able re-open it in that case.

closed 2023-05-13 by navid-zamani