morrownr/USB-WiFi #229: how to capture raw ampdu packet

serial115200, 2023-03-17:
Hi everyone

The wireless driver will split the ampdu packet into multiple mpdu packets.

How to capture raw ampdu packet. Modify the driver or purchase other devices for packet capture

any suggestion, thanks.

morrownr, 2023-03-17:
Hi @serial115200 

I may not be able to answer this question but I think those that can will need more info.

What adapter/driver are you using now?
What app are you using to capture packets currently?
Are you using a guide? If so, can you post a link?

dubhater, 2023-03-22:
> The wireless driver will split the ampdu packet into multiple mpdu packets.

Maybe you are confused about what AMPDU is.

Back in the 802.11g days, transmissions between two stations looked something like this:
* station A sends packet 1 to station B
* station B sends ack (acknowledgement) for packet 1
* station A sends packet 2 to station B
* station B sends ack for packet 2
* station A sends packet 3 to station B
* station B sends ack for packet 3
* station A sends packet 4 to station B
* station B sends ack for packet 4
* station A sends packet 5 to station B
* station B sends ack for packet 5

A lot of time was spent just sending ack.

802.11n introduced AMPDU:
* station A sends packet 1 to station B
* station A sends packet 2 to station B
* station A sends packet 3 to station B
* station A sends packet 4 to station B
* station A sends packet 5 to station B
* station B sends block ack for packets 1-5

This is MPDU aggregation aka AMPDU. The packets are still sent one at a time, but without waiting for ack after each one.

serial115200, 2023-05-10:
I see.

Thank you.

The image explains A-MPDU and A-MSDU.

[image]

closed 2023-05-10 by serial115200