morrownr/USB-WiFi #230: Overview of WiFi 6 USB adapters and their chipsets available in Europe

mendel5, 2023-03-18:
I was looking for USB WiFi adapters that support modern speeds (at least WiFi 6 / 802.11ax) and are available in Europe.

The German/ European website Geizhals has nice filtering options to search for products. The link to the filtered list of USB Wifi adapters is: https://geizhals.eu/?cat=nwpcie&xf=14065_80211-Adapter%7E14066_USB-Stick%7E14072_WLAN+802ax11

I did some research on which chipsets these USB WiFi adapters use and wanted to share my findings with you.

**Netgear AXE3000 (A8000)**
- Advertised speed: WiFi 6E
- Chipset: `mt7921au`
- Source for chipset information: https://patchwork.kernel.org/project/linux-wireless/patch/20230123090555.21415-1-git@qrsnap.io/
- Link to adapter: https://geizhals.eu/netgear-axe3000-wi-fi-6e-adapter-a8000-a2818234.html

**ASUS USB-AX56 (90IG06H0-MO0R00)**
- Advertised speed: WiFi 6
- Chipset: `RTL8832AU`
- Source for chipset information: http://en.techinfodepot.shoutwiki.com/wiki/ASUS_USB-AX56
- Link to adapter: https://geizhals.eu/asus-usb-ax56-90ig06h0-mo0r00-a2582054.html

**D-Link AX1800 (DWA-X1850)**
- Advertised speed: WiFi 6
- Chipset: `RTL8832AU`
- Source for chipset information: http://en.techinfodepot.shoutwiki.com/wiki/D-Link_DWA-X1850_rev_A1
- Link to adapter: https://geizhals.eu/d-link-ax1800-dwa-x1850-a2608455.html

**Allnet ALL-WA1800AX (207515)**
- Advertised speed: WiFi 6
- Chipset: `RTL8832AU`
- Source for chipset information: https://gzhls.at/blob/ldb/9/0/8/5/7303802bc1a36f05bf1c538beb4744ba7032.pdf
- Link to adapter: https://geizhals.eu/allnet-all-wa1800ax-207515-a2750885.html

**Tenda U18**
- Advertised speed: WiFi 6
- Chipset: possibly `RTL8832AU` (unsure)
- Source for chipset information: See Geizhals link of this adapter.
- Link to adapter: https://geizhals.eu/tenda-u18-a2768770.html

**Edimax AX1800 (EW-7822UMX)**
- Advertised speed: WiFi 6
- Chipset: `RTL8832BU`
- Source for chipset information: Contact with Edimax support.
- Link to adapter: https://geizhals.eu/edimax-ax1800-ew-7822umx-a2768803.html

**ASUS USB-AX55 Nano (90IG06X0-MO0B00)**
- Advertised speed: WiFi 6
- Chipset: `RTL8852BU`
- Source for chipset information: http://en.techinfodepot.shoutwiki.com/wiki/ASUS_USB-AX55_Nano
- Link to adapter: https://geizhals.eu/asus-usb-ax55-nano-90ig06x0-mo0b00-a2850211.html

**ZyXEL AX1800 (NWD7605-EU0101F)**
- Advertised speed: WiFi 6
- Chipset: `RTL8852BU`
- Source for chipset information: Contact with ZyXEL support.
- Link to adapter: https://geizhals.eu/zyxel-ax1800-dualband-nwd7605-eu0101f-a2805390.html

In total there are four different chipsets: `mt7921au`, `RTL8832AU`, `RTL8832BU` and `RTL8852BU`.
Information about these chipsets can be found here: https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Chipsets.md

Other relevant web pages:
- http://en.techinfodepot.shoutwiki.com/wiki/MediaTek
- http://en.techinfodepot.shoutwiki.com/wiki/Realtek/Realtek_PCIe-USB-SDIO
- https://wikidevi.wi-cat.ru/MediaTek
- https://wikidevi.wi-cat.ru/Realtek

bjlockie, 2023-03-18:
Does Realtek have a 6e chip yet? :-)

morrownr, 2023-03-20:
@mendel5 

Good thread. I'll post what I think I know.

> Netgear AXE3000 (A8000)

If my count is correct, I think we have seen about 7 messages from users that have bought and are using this adapter. The challenge with this adapter right now is that Netgear decided to use a company VID/PID which is in the process of going into the kernel, but is not there yet. See:

https://github.com/morrownr/USB-WiFi/issues/218

Those that do have it working seem happy but most are only running managed mode so I'm not sure about AP mode and monitor mode. It is expensive.

 > D-Link AX1800 (DWA-X1850)

This adapter uses the rtl8832au chipset. The driver is called rtl8852au and supports both the rtl8852au and rtl8832au chipsets. They are really the same chipset as far as I can tell but rtl8852au has bluetooth turned on. It appears that production of these chipsets has ceased. I attempted, with help, to make the code that Realtek released available on this site but was unable. The driver is terrible. Linux users should avoid adapters with either of these chipsets.

> Edimax AX1800 (EW-7822UMX)

This adapter uses the rtl8832bu which is the 2nd generation Realtek WiFi 6 usb chipset. Myself and another individual are working on the driver in the background here currently. Don't hold you breathe, it could be months before it is ready. My opinion is that Linux users are taking a big chance with this chipset. The driver is an out-of-kernel driver and that is problematic on Linux due to the development model of Linux.

> In total there are four different chipsets

Indications are that there will be a mt7922u chipset as well. It will use the same mt7921u driver as the mt7921 chipsets are using. We are now seeing cards available with the mt7922 chipset. Cards are always available before usb adapters so it could be later this year before we see adapters. As far as I can tell, the only difference between the mt7921 and mt7922 is channel width. The mt7921 supports up to 80 MHz channel width while the mt7922 supports up to 160 MHz channel width.

I recently posted the results of a Speed Test:

https://github.com/morrownr/USB-WiFi/blob/main/home/Speed_Comparison_Test.md

It has several adapters with various chipsets, including the mt7921au and rtl8832bu. The mt7921au is faster.

Since we only have 2 chipset makers selling into the usb wifi market, we have to concentrate of what they bring to the table. Realtek is providing only out-of-kernel drivers which is not new. There driver are very big and complex making them hard to work on. The Realtek rtl8852au driver is terrible and since the chipset seems to have ended product, I see it as a likely candidate to be abandoned as far as Linux users are concerned. Possibly for other platforms also.

The rtl8832bu chipset looks like a better bet if you have to buy a Realtek chipset but there is currently not a lot of folks supporting the driver. I am working on it as I have time and it has problems so it may be a long time before I make it available here, So, right now, Linux users should expect a lack of support if they buy this chipset.

On the other hand, Mediatek's mt7921u driver was in the Linux kernel before adapters came on the market. The key here is that it is a fully Linux Wireless Standards compliant driver that is in-kernel which should result in long term support that users don't have to worry about. Also, while it is impossible for everyday users to report problems directly to Realtek, there are well documented ways to report problems to the kernel devs and Mediatek has several kernel devs working on their drivers. Is mt7921u a perfect driver? No but my experience and the reports of many users indicate it is pretty solid for most uses to include managed mode, AP mode and monitor mode. I am trying to help a couple users right now discover a problem they are having AP mode. I running AP mode with AX capability on my RasPi4B and I have yet to duplicate the problem but I'll keep trying.

@bjlockie 

> Does Realtek have a 6e chip yet? :-)

Not that I am aware of.

mendel5, 2023-03-20:
Thank you for your detailed and insightful answer, @morrownr !

morrownr, 2023-03-20:
@mendel5 

After looking at my response, I think there are some things I could add. The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi/blob/main/README.md

Menu item 2 gives information about and links to specific adapters. If you go down to the mt7612u section and look at the ALFA AWUS036ACM , it shows links to Varia and getic. They are among the European Alfa dealers. You could make contact with a dealer such as Varia to ask when they will have the ALFA AWUS036AXML in stock.

Right now you can see 4 adapters based on the mt7921au chipset in the Plug and Play list. More are expected this year. Based on the reports from users, many are having success with the CF-953AX, the Alfa ACML and the Netgear A8000. One thing to note about the Netgear A8000 is that Netgear used a company VIP/PID so it is not plug and play in current kernels yet. A patch has been sent in to the kernel devs to have it included but it is not included yet.

Your title says WiFi 6 so I'll keep the conversation to WiFi 6. Of the WiFi 6 chipsets, there is another major difference between the Mediatek and Realtek chipsets. The Realtek usb chipsets are WiFi 6 only, not WiFi 6e. WiFi 6e means that the chipset supports the 6 GHz band. Therefore, the Realtek chipsets do not support the 6 GHz band. On the other hand, the Mediatek chipsets are WiFi 6e and do support the 6 GHz band. Since the EU has it together better than the rest of the world regarding the 6 GHz band, you might want to lean toward the Mediatek chipsets so that you can use the 6 GHz band when the time comes.

mendel5, 2023-03-22:
> Your title says WiFi 6 so I'll keep the conversation to WiFi 6

Feel free to share information on WiFi 6E as well. The first post also includes the Netgear AXE3000 (A8000) which supports WiFi 6E.

> You could make contact with a dealer such as Varia to ask when they will have the ALFA AWUS036AXML in stock.

Maybe the following web page is helpful?: https://store.rokland.com/products/alfa-awus036axml-802-11ax-wifi-6-1800-mbps-tri-band-wifi-usb-adapter-w-bluetooth . I know Rokland is based in Florida, United States but on their web page it says that the ALFA AWUS036AXML will be in stock on April 4th 2023. I would expect the new ALFA to be available at Varia soon as well.

morrownr, 2023-03-23:
> Feel free to share information on WiFi 6E as well.

I will but I am not there yet. 

> I know Rokland is based in Florida, United States but on their web page it says that the ALFA AWUS036AXML will be in stock on April 4th 2023. 

I think Rokland has already had 2 shipments of the AXML. The first all went to pre-orders and the next one went to subsequent orders and they are out again. This is just based on me stopping by their site periodically to keep the list of plug and play adapters current so my observations could be off.

> I would expect the new ALFA to be available at Varia soon as well.

It might be a good idea to call or write to Varia to make sure they are aware of the demand. Rokland is very tuned into the Lixux community and does a good job of supporting Linux users. I don't have any experience with Varia so I don't know. Rokland even sells Raspberry Pi's these days so they are neck deep in Linux these days. They sell WiFi equipment that you won't see at most sites. They sell directional antennas and a lot of cool stuff.

morrownr, 2023-03-25:
@mendel5 

@bjlockie  started an issue recently about the following adapter:

https://www.alibaba.com/product-detail/EDUP-High-Speed-802-11AX-Dual_1600558720979.html

Some research seems to indicate that EDUP is starting to produce the above adapter but the only place it is listed currently is the above Alibaba site. Alibaba is not a site that I use and I'm not aware of anyone else that uses it either. The ads on Alibaba are more often than not wildly inaccurate. It is like they don't even care about posting correct information. Maybe the adapter will become available at another location soon.