morrownr/USB-WiFi #234: RT5370 Mesh Point alternatives

elockman, 2023-03-27:
@morrownr I really appreciate the info you have compiled here.

I've been using the RT5370 for a mesh project, but it does not seem to support mesh point anymore in Kernel 5.15.68. I'm building my own linux with yocto for the sama5d2 xplained board.
You mentioned that you have the RT5370. Did you see the mesh point mode disappear on recent kernels?
```
root@sama5d2-xplained:~# uname -a
Linux sama5d2-xplained 5.15.68-linux4microchip-2022.10 #1 Wed Nov 9 12:48:42 UTC 2022 armv7l armv7l armv7l GNU/Linux
root@sama5d2-xplained:~# lsusb
Bus 001 Device 002: ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
root@sama5d2-xplained:~# iw list | grep "Supported interface modes" -A 8
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
        Band 1:
                Capabilities: 0x17e
                        HT20/HT40
```
I've also had trouble finding the RT5370 in bulk for a wholesale price. What would you recommend that I look at for an alternative, which would support managed, AP, and mesh point modes?

I've purchased the following without much luck, but I may be configuring some of them incorrectly in the yocto build:
```
#**MEDIATEK**
# MT7610
# **RALINLK**
# MT7601
# RT5370
# **REALTEK**
# RTL8092 
# RTL8188 
# RTL8192 
# RTL8812 
# RTL8821  
# **TP LINK**
# RTL8811 
```
I've also tried building some of the custom drivers but ran into errors with asm includes.

elockman, 2023-03-27:
Some of the USB wifi dongles listed above are 802.11ac. I would prefer to stick with 802.11b/g/n. The RT5370 may still be available. If I could get mesh point working on kernel 5.15 (even with the RT5370), I would be thrilled.

morrownr, 2023-03-28:
Hi @elockman 

> I've been using the RT5370 for a mesh project, but it does not seem to support mesh point anymore in Kernel 5.15.68.

I just did a quick test:

6.2.7-060207-generic

ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter

	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point

Any other kernels that you want me to test?

@morrownr

morrownr, 2023-03-28:
@elockman 

I keep a lot of kernels installed on my main dev box for testing reasons:

5.15.0-69-generic

ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter

	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point

> Did you see the mesh point mode disappear on recent kernels?

No

I can see that you are using a 32 bit ARM kernel but I don't recognize the distro. Care to share details?

Before you ask, why I am seeing `mesh point` and you are not on kernel 5.15 is unknown to me right now. Old firmware version in your distro? 32 bit issue?

elockman, 2023-03-28:
@morrownr, thanks for the engagement.

If you are familiar with yocto, I am building the **microchip-headless-image** found [here](https://github.com/linux4sam/meta-atmel/tree/kirkstone/recipes-atmel/images).
I am building a "vanilla" Microchip distribution for the sama5d2 board. (Formerly Atmel, but purchased by Microchip.)
Details are [here](https://www.linux4sam.org/bin/view/Linux4SAM/Sama5d2XplainedMainPage).

If you scroll down to the kernel building section, you'll see there is a Microchip (mchp) [kernel](https://github.com/linux4microchip/linux).

You can also find the RALINK drivers in the normal [path](https://github.com/linux4microchip/linux/tree/linux-5.15-mchp/drivers/net/wireless/ralink). 

I'm not familiar enough with the drivers to know where to look for mesh point.

Working with kernel v5.4, I saw mesh point, but I do not now in 5.15. I had to upgrade for support on other fundamental packages. I do not see significant changes in the Ralink driver, so I'm not sure where mesh point would have been dropped.

I'd appreciate any insight you can provide.

elockman, 2023-03-28:
@morrownr, before you go looking, I may have found it.

It could likely be a config issue on my part. Now that I know yours is building in the mesh point I took another look.

The mesh point is enabled with an **#ifdef** [here](https://github.com/linux4microchip/linux/blob/linux-5.15-mchp/drivers/net/wireless/ralink/rt2x00/rt2x00dev.c#L1432).  

I'll rebuild and let you know what I find with the new image.

elockman, 2023-03-28:
@morrownr 
I've been very happy with the RT5370. It comes in a small package and historically has been inexpensive. This is a picture of the form factor I've been using. If I cannot find more, is there an alternative chipset you would recommend for managed/AP/mesh point?
[image]

morrownr, 2023-03-28:
Good to see you back in business.

> If I cannot find more, is there an alternative chipset you would recommend for managed/AP/mesh point?

Indications are that the rt5370 is still in production and may be for a considerable amount of time. The ar9271 and rt3070 made their final production runs in 2021. My opinion is that rt5370 is probably the single best solution for what you are doing. The rt5370 is a really solid chipset with a good driver.

The mt7601u does not support every capability you want. There are some Realtek alternatives but the status of the drivers, capability of the drivers, how well the drivers are maintained and how long production will continue, if production hasn't terminated are all issues that would have to be investigated.

Here is a link I use for Alienexpress to check the market:

https://www.aliexpress.com/w/wholesale-rt5370-usb-wifi-adapter.html?initiative_id=AS_20211217111156&origin=y

I use the little nano Panda as well as a little generic stick with the rt5370. If you contact Panda with questions, you might find a dependable source. If Panda is looking at quanity sales, they can probably get the price down to something you can work with. They might also give you an idea about how long production will continue on the chipset.

If you don't mind, I am going to reopen this issue as others may find this issue informative. If you want it closed, just reclose it and I will leave it alone.

Cheers,

@morrownr