morrownr/USB-WiFi #236: TP-Link T2U Plus (RTL8821AU)

tyt3ool, 2023-03-29:
Thank you for updating the information and bein the main source of USB adapters for Linux. The TP-Link T2U Plus is widely available in many countries with a very affordable price for beginners in pentesting. Though TP-Link products support for Linux is terrible and their chipset confusion is the second downside to it, I'm curious to know your thoughts about TP-Link T2U Plus. Being "Alfa AWUS036ACHM" 10 out of 10, how do you rate other alternative cheap/affordable products for pentesting, full/missing features, handshake catching, long range, etc? If managing challenging circumstances of compiling the proper driver is okay, what are the better options under $20 for pentesting?

ZerBea, 2023-03-29:
RTL8821AU driver is not part of the Linux stock kernel.
RTL8821AU driver does  not support active monitor mode (like Alfa AWUS036ACHM or other mt76 interfaces)
TP-Link T2U Plus antenna is fixed mounted

My rating is 0!

ZerBea, 2023-03-29:
An example: TP-Link T2UH
MT7610U is part of the Linux stock kernel
MT7610U provide full active monitor mode
TP-Link T2UH external antenna not fixed mounted

```
$ lsusb
ID 148f:761a Ralink Technology, Corp. MT7610U ("Archer T2U" 2.4G+5G WLAN Adapter
```

```
$ hcxlabtool -I wlp5s0f3u3

Requesting interface capabilities. This may take some time.
Please be patient...

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  2   5 503eaaa08f6f 503eaaa08f6f * wlp5s0f3u3       mt76x0u (NETLINK & WIRELESS EXTENSIONS)

available frequencies: frequency [channel] tx-power

  2412 [  1] 14.0 dBm	  2417 [  2] 14.0 dBm	  2422 [  3] 14.0 dBm	  2427 [  4] 14.0 dBm
  2432 [  5] 14.0 dBm	  2437 [  6] 14.0 dBm	  2442 [  7] 14.0 dBm	  2447 [  8] 14.0 dBm
  2452 [  9] 14.0 dBm	  2457 [ 10] 14.0 dBm	  2462 [ 11] 14.0 dBm	  2467 [ 12] disabled
  2472 [ 13] disabled	  2484 [ 14] disabled	  5180 [ 36] 17.0 dBm	  5200 [ 40] 17.0 dBm
  5220 [ 44] 17.0 dBm	  5240 [ 48] 17.0 dBm	  5260 [ 52] 17.0 dBm	  5280 [ 56] 17.0 dBm
  5300 [ 60] 17.0 dBm	  5320 [ 64] 17.0 dBm	  5500 [100] 17.0 dBm	  5520 [104] 17.0 dBm
  5540 [108] 17.0 dBm	  5560 [112] 17.0 dBm	  5580 [116] 17.0 dBm	  5600 [120] 17.0 dBm
  5620 [124] 17.0 dBm	  5640 [128] 17.0 dBm	  5660 [132] 17.0 dBm	  5680 [136] 17.0 dBm
  5700 [140] 17.0 dBm	  5720 [144] 17.0 dBm	  5745 [149] 17.0 dBm	  5765 [153] 17.0 dBm
  5785 [157] 17.0 dBm	  5805 [161] 17.0 dBm	  5825 [165] 17.0 dBm	  5845 [169] 17.0 dBm
  5865 [173] 17.0 dBm

bye-bye
```

tyt3ool, 2023-03-29:
Thank you for answering my question. Are any commands completely compromised by these alternative models?

tyt3ool, 2023-03-29:
> Do you have an Alfa AWUS036ACHM?

No, I want to purchase one but was thinking of trying with a cheaper one first.

> https://github.com/ZerBea/hcxdumptool/wiki/WiFi-Adapters

Unfortunately none of them are offered by local vendors, and the international delivery cost is nearly half or equal to the cost of the adapter. The market is full of Comfast and TP-Link adapters such as: 
CF-924AC (RTL8812BU)
CF-WU713N (MT7603)
CF-WU757F V2 (RTL8188GU)
CF-951AX (MT7921AU)
CF-939AC (RTL8814AU)
CF-WU783AC (RTL8814AU)
CF-WU782AC V2 (MT7612UN)
CF-WU781A (RTL8811CU)
CF-821AC (RTL8811CU)
CF-928AC (MT7612U)
CF-953AX (MT7921AU)
CF-WU785AC (MT7612u)
CF-759BF (RTL8821CU)
CF-781AC (RTL8811CU)
CF-782AC (RTL8811CU)
CF-927BF (RTL8822BU)
CF-955AX (RTL8832BU)
CF-WU711N (MT7601)
CF-926AC (MT7612U)
And a dozen of unbranded adapters based on AR9271, RT3070, MT7601U, RTL8811CU chipsets.

I might just go for the AWUS036ACHM because of its positive reputation but it would be nice to know other great products that can be purchased easily as needed.

ZerBea, 2023-03-29:
For the first steps a cheap adapter running a mt7601 should do it.

An AWUS036ACHM is a real beast. It is able t retrieve hundreds of PMKIDs and EAPOL M2 in a very short time.
```
$ hcxpcapngtool AWUS036ACHM.pcapng -o hash.hc22000
hcxpcapngtool 6.2.9 reading from AWUS036ACHM.pcapng...

summary capture file
--------------------
file name................................: AWUS036ACHM.pcapng
version (pcapng).........................: 1.0
operating system.........................: Linux 5.15.32+
application..............................: hcxlabtool 2.0.0
interface name...........................: wlan0
interface vendor.........................: 00c0ca
openSSL version..........................: 1.0
weak candidate...........................: 12345678
MAC ACCESS POINT.........................: 000e17d2dcfc (incremented on every new client)
MAC CLIENT...............................: acde48faba3c
REPLAYCOUNT..............................: 64129
ANONCE...................................: 35d62366994d0a53bdde658ca88274af144007133c9e9fdd45b89181b54076d4
SNONCE...................................: 219681af95fcbf17d35b5557f006722a0652a6a060c6715d3c2ac4ba84aaf08a
timestamp minimum (GMT)..................: 21.03.2023 19:15:28
timestamp maximum (GMT)..................: 21.03.2023 20:26:12
used capture interfaces..................: 1
link layer header type...................: DLT_IEEE802_11_RADIO (127)
endianess (capture system)...............: little endian
packets inside...........................: 27494
packets received on 2.4 GHz..............: 26777
packets received on 5 GHz................: 100
WIRELESS DISTRIBUTION SYSTEM.............: 242
ESSID (total unique).....................: 1076
BEACON (total)...........................: 2170
BEACON on 2.4 GHz channel (from IE_TAG)..: 1 2 3 4 5 6 7 8 9 10 11 13 
BEACON on 5/6 GHz channel (from IE-TAG)..: 36 100 
BEACON (SSID wildcard/unset).............: 204
BEACON (SSID zeroed).....................: 16
ACTION (total)...........................: 50
ACTION (containing ESSID)................: 49
PROBEREQUEST.............................: 424
PROBEREQUEST (directed)..................: 130
PROBERESPONSE (total)....................: 1370
PROBERESPONSE (SSID unset)...............: 1
AUTHENTICATION (total)...................: 2436
AUTHENTICATION (OPEN SYSTEM).............: 2436
ASSOCIATIONREQUEST (total)...............: 509
ASSOCIATIONREQUEST (PSK).................: 509
REASSOCIATIONREQUEST (total).............: 154
REASSOCIATIONREQUEST (PSK)...............: 151
REASSOCIATIONREQUEST (SAE SHA256)........: 1
EAPOL messages (total)...................: 20250
EAPOL RSN messages.......................: 20246
EAPOL WPA messages.......................: 4
EAPOLTIME gap (measured maximum msec)....: 48336
EAPOL ANONCE error corrections (NC)......: working
REPLAYCOUNT gap (suggested NC)...........: 1024
EAPOL M1 messages (total)................: 17143
EAPOL M2 messages (total)................: 1160
EAPOL M3 messages (total)................: 1578
EAPOL M4 messages (total)................: 369
EAPOL pairs (total)......................: 4209
EAPOL pairs (best).......................: 313
EAPOL ROGUE pairs........................: 172
EAPOL pairs written to 22000 hash file...: 313 (RC checked)
EAPOL M12E2 (challenge)..................: 201
EAPOL M32E2 (authorized).................: 111
EAPOL M34E4 (authorized).................: 1
RSN PMKID (useless)......................: 111
RSN PMKID (total)........................: 286
RSN PMKID (best).........................: 63
RSN PMKID ROGUE..........................: 53
RSN PMKID written to 22000 hash file.....: 63
malformed packets (total)................: 1
IE TAG length error (malformed packets)..: 1

frequency statistics from radiotap header (frequency: received packets)
-----------------------------------------------------------------------
 2412: 8151	 2417: 28	 2422: 12	 2427: 32	
 2432: 15	 2437: 8489	 2442: 164	 2447: 62	
 2452: 32	 2457: 11	 2462: 9752	 2467: 8	
 2472: 21	 5180: 58	 5500: 42	

session summary
---------------
processed pcapng files................: 1
```

ZerBea, 2023-03-29:
ALFA AC51 information:

```
$ hcxlabtool -L

Requesting interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  7  10 0c9d92b486ca a6658b06d347 * wlp5s0f3u2       mt76x0u (NETLINK & WIRELESS EXTENSIONS)

* active monitor mode available
+ monitor mode available
- no monitor mode available

bye-bye
```
```
$ hcxlabtool -I wlp5s0f3u2

Requesting interface capabilities. This may take some time.
Please be patient...

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  7  10 0c9d92b486ca a6658b06d347 * wlp5s0f3u2       mt76x0u (NETLINK & WIRELESS EXTENSIONS)

available frequencies: frequency [channel] tx-power

  2412 [  1] 16.0 dBm	  2417 [  2] 16.0 dBm	  2422 [  3] 16.0 dBm	  2427 [  4] 16.0 dBm
  2432 [  5] 16.0 dBm	  2437 [  6] 16.0 dBm	  2442 [  7] 16.0 dBm	  2447 [  8] 16.0 dBm
  2452 [  9] 16.0 dBm	  2457 [ 10] 16.0 dBm	  2462 [ 11] 16.0 dBm	  2467 [ 12] disabled
  2472 [ 13] disabled	  2484 [ 14] disabled	  5180 [ 36] 20.0 dBm	  5200 [ 40] 20.0 dBm
  5220 [ 44] 20.0 dBm	  5240 [ 48] 20.0 dBm	  5260 [ 52] 20.0 dBm	  5280 [ 56] 20.0 dBm
  5300 [ 60] 20.0 dBm	  5320 [ 64] 20.0 dBm	  5500 [100] 20.0 dBm	  5520 [104] 20.0 dBm
  5540 [108] 20.0 dBm	  5560 [112] 20.0 dBm	  5580 [116] 20.0 dBm	  5600 [120] 20.0 dBm
  5620 [124] 20.0 dBm	  5640 [128] 20.0 dBm	  5660 [132] 20.0 dBm	  5680 [136] 20.0 dBm
  5700 [140] 20.0 dBm	  5720 [144] 20.0 dBm	  5745 [149] 20.0 dBm	  5765 [153] 20.0 dBm
  5785 [157] 20.0 dBm	  5805 [161] 20.0 dBm	  5825 [165] 20.0 dBm	  5845 [169] 20.0 dBm
  5865 [173] 20.0 dBm

bye-bye
```

ZerBea, 2023-03-29:
Even the cheapest mt76 adapter support active monitor mode:
https://4.bp.blogspot.com/-bWDe3SYr0WE/VvCybE7ofzI/AAAAAAAAB0U/xQPQeTjkSy8fF3Rzuhd4OGCawy3YgSUfg/w858-h520-no/adaptador%2Bde%2Brede%2Bsem%2Bfio%2BRalink.jpg
```
$ lsusb
ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
```

```
$ hcxlabtool -L

Requesting interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  8  11 76013d100942 6ea417f7bc34 * wlp5s0f3u2       mt7601u (NETLINK & WIRELESS EXTENSIONS)

* active monitor mode available
+ monitor mode available
- no monitor mode available

bye-bye
```

```
$ hcxlabtool -I wlp5s0f3u2

Requesting interface capabilities. This may take some time.
Please be patient...

interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  8  11 76013d100942 6ea417f7bc34 * wlp5s0f3u2       mt7601u (NETLINK & WIRELESS EXTENSIONS)

available frequencies: frequency [channel] tx-power

  2412 [  1] 20.0 dBm	  2417 [  2] 20.0 dBm	  2422 [  3] 20.0 dBm	  2427 [  4] 20.0 dBm
  2432 [  5] 20.0 dBm	  2437 [  6] 20.0 dBm	  2442 [  7] 20.0 dBm	  2447 [  8] 20.0 dBm
  2452 [  9] 20.0 dBm	  2457 [ 10] 20.0 dBm	  2462 [ 11] 20.0 dBm	  2467 [ 12] 20.0 dBm
  2472 [ 13] 20.0 dBm	  2484 [ 14] disabled

bye-bye
```

ZerBea, 2023-03-29:
@morrownr how about adding an information about "active monitor mode" and its (huge) advantage to USB-WiFi?

tyt3ool, 2023-03-31:
It would be great to have a list of chipsets and drivers that are supporting active monitor mode, as there may be more options available beyond the ones mentioned in the #73  (mt7921, mt7612u, mt7610u, mt7601u).

Do these mentioned USB adapters work with Android phones?

ZerBea, 2023-03-31:
I agree, it would be great to have all this additional information on USB-WiFi.
Regarding Android drivers, this I suggest to ask here:
https://forum.xda-developers.com/

ZerBea, 2023-03-31:
Unfortunately public information about active monitor mode is very rare. As far as I know, only mt76 drivers provide this feature.
https://github.com/openwrt/mt76/issues/310
I took me several month to figure out how it exactly works and to discover its limitations (e.g. ACK only frames addressed to interface MAC - can be the virtual MAC).

morrownr, 2023-03-31:
> Unfortunately public information about active monitor mode is very rare.

Let's make it more public.

>  As far as I know, only mt76 drivers provide this feature.

My bet is that all in-kernel drivers support it because it is probably supported in the stack the in-kernel drivers use. I can test an Atheros adapter later... and maybe some Ralink adapters.

ZerBea, 2023-03-31:
Realtek drivers could be the next ones providing active monitor.
@kimocoder mentioned that.
https://github.com/kimocoder/realtek_rtwifi

ZerBea, 2023-03-31:
If you take a look at the injection radiotap header of hcxlabtool:
https://github.com/ZerBea/wifi_laboratory/blob/main/include/radiotap.h#L55
even the example from here:
https://www.kernel.org/doc/html/latest/networking/radiotap-headers.html
looks a little bit outdated.

ZerBea, 2023-04-11:
@morrownr , while hunting for a bug on rtl8xxxu, we discussed active monitor mode, too:
https://bugzilla.kernel.org/show_bug.cgi?id=217205
and decided to open a feature request:
https://bugzilla.kernel.org/show_bug.cgi?id=217319

morrownr, 2023-04-11:
It would be nice if a feature request went in for the rtw88 driver as well. I was testing the rtw88 on a rtl8812bu based adapter this morning with kernel 6.3. The performance and stability has increased a lot since 6.1. Heck, it is usable now.

ZerBea, 2023-04-12:
For sure, it will be and I can confirm your experience regarding rtw88.