morrownr/USB-WiFi #238: Intel AX201/AX411 & CF-953AX Wi-Fi Connection Issue

mazzz1y, 2023-04-09:
Hi all,

I've been experiencing an intermittent Wi-Fi issue with my Linux laptop, which has an Intel AX201 Wi-Fi card. Sometimes, my laptop can't find a specific Wi-Fi network. Other networks work perfectly, and other devices can connect to the problematic network without any issues. The problem occurs only with this laptop and this particular network.

The network is hosted by a Raspberry Pi, which is configured as a router and uses a CF-953AX Wi-Fi card. Both the server (Raspberry Pi) and client (laptop) are running kernel version 6.1.21. Sometimes the laptop connects to the network immediately, but most of the time, it doesn't.

After waiting for about 1-2 minutes, NetworkManager eventually detects the network and successfully connects to it. Interestingly, ~~if I mark the network as "hidden," the laptop can connect without any issues, which suggests there might be a problem with SSID broadcasting~~ not sure that it really helps or just self-hypnosis

I tried replacing the AX201 card with an AX411, but the issue persists. I've also tested both Wi-Fi 5 and Wi-Fi 6 modes with configs from this repo, but neither resolves the problem. I've checked the logs, but I don't see anything related to this issue.

Here's what I've tried so far:

1. Testing with a Terow ROW02FD (mt7612u) as a client, and it worked perfectly. Before that, I used the mt7612u as an access point, and it also worked perfectly. The issue seems to occur only when using the Intel AX201/AX411 and CF-953AX bundle.

2. I placed the router only 1 meter away from the laptop, with no walls in between. I still encountered the issue despite having a strong signal and 720mbit+ bandwidth when connected.

This is my main PC, and I haven't experienced the same issue with any other network.

Has anyone experienced a similar issue or have any suggestions on how to fix it? I'm trying to fix this issue and will update this post if I find something.

bjlockie, 2023-04-09:
Is there anything in dmesg on the AP?

morrownr, 2023-04-09:
Hi @mazzz1y 

I love a good mystery.

Where to start? Well, how about...

The cf-953ax uses the mt7921au chipset which uses the mt7921u driver that has been in the kernel with AP support since kernel 5.18. Since the RasPiOS recently decided to go ahead bump up to kernel 6.1, you are good with the mt7921u driver as in it seems to be stable, however, new supporting firmware files are coming fast and furious. I think I saw both wift firmware files for the mt7921u flow in new versions Friday. I saw them flow in at linux-wireless but let me check to see if they are posted in their home.

Not posted yet. The most recent showing is dated 2023-03-07:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/mediatek

The new files should be posted sometime this week. Then names are:

WIFI_MT7961_patch_mcu_1_2_hdr.bin
WIFI_RAM_CODE_MT7961_1.bin

The tree with the files, when they are updated is:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek

The link to the firmware installation guide here is:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

You will looking for section 2. You'll have to wait until the new files are posted.

For what it is worth, sometimes bluetooth is detected even though bluetooth is turned off on the 953. The easiest way to shut down the detection is to delete the bluetooth firmware file from your installation.

> Therefore, I believe the root of the problem lies in the CF-953AX.

It very well could be. I have a cf951ax and can do some testing but I don't have an intel card to test with.

FWIW: The sample hostpd.conf for WiFi 6 that I have up should be consider a beta as I am still working in it.

We could go over the hostapd.conf for WiFi 5 that you are using just to make sure if you want to post it.

One thing that I have been fighting for some time is that the RasPiOS is still based on the old Debian 11 so there are parts that may not be working well together. I'll be glad when RasPiOS rebases on the upcoming Debian 12.

@morrownr

lkraav, 2023-04-10:
> It very well could be. I have a cf951ax and can do some testing but I don't have an intel card to test with.

I bought an M2230 AX200 Desktop Kit, before I learned of Intel's ahem... "limitations" in 5 GHz AP mode.

Kit is new in box. Want me to mail it to you @morrownr?

mazzz1y, 2023-04-10:
> Is there anything in dmesg on the AP?

Nothing since boot:
```
...
[    9.428545] mt7921u 2-2:1.3: WM Firmware Version: ____010000, Build Time: 20230302150956
[    9.599945] bcmgenet fd580000.ethernet end0: Link is Up - 100Mbps/Full - flow control rx/tx
[    9.608420] IPv6: ADDRCONF(NETDEV_CHANGE): end0: link becomes ready
[   11.080610] mt7921u 2-2:1.3 wlp1s0u2i3: renamed from wlan1
[   11.103758] Bluetooth: hci0: Opcode 0x c03 failed: -110
[   13.124172] IPv6: ADDRCONF(NETDEV_CHANGE): wlp1s0u2i3: link becomes ready
```

> new supporting firmware files are coming fast and furious

Thank you, I updated the firmware to the latest available version, and I will keep an eye on future updates. I'll post here if anything changes with the firmware updates.

> One thing that I have been fighting for some time is that the RasPiOS is still based on the old Debian 11 so there are parts that may not be working well together. I'll be glad when RasPiOS rebases on the upcoming Debian 12.

I use Arch with kernel from Raspberry PI OS

> We could go over the hostapd.conf for WiFi 5 that you are using just to make sure if you want to post it.

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-5g.conf
https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi6.conf

I only changed ssid and password, both 5 and 6 have the same issue

bjlockie, 2023-04-10:
I don't use it for an AP.
Do you have the same HW/SW version?

[   11.650124] mt7921u 2-2.3:1.3: HW/SW Version: 0x8a108a10, Build Time: 
20220908210919a
[   11.666265] mt7921u 2-2.3:1.3: WM Firmware Version: ____010000, Build 
Time: 20220908211021

ZerBea, 2023-04-11:
Wireless regulatory domain set correctly?

mazzz1y, 2023-04-15:
> I don't use it for an AP.
> Do you have the same HW/SW version?
```
$ sudo dmesg | grep mt7
[    9.094050] usbcore: registered new interface driver mt7921u
[    9.106287] mt7921u 2-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230302150916a
[    9.399881] mt7921u 2-2:1.3: WM Firmware Version: ____010000, Build Time: 20230302150956
[   11.048281] mt7921u 2-2:1.3 wlp1s0u2i3: renamed from wlan1
```

> Wireless regulatory domain set correctly?
> https://github.com/aircrack-ng/aircrack-ng/discussions/2430#discussioncomment-5023204
> If not or not correct configured, the kernel prevent transmission and/or reception on several channels.
> Also it has an impact on max TX power.

I've thought about it. Intel cards have a feature called LAR, which sets the region based on nearby APs, and it cannot be manually handled on the latest kernels without patching. Usually, my region is `CL` or `00` when I check it using `iw reg get`.

I don't think it's related to the problem because if I set up two access points simultaneously, it works perfectly with `mt7612u` but not with `mt7921u`.

---
After updating the firmware and upgrading the Linux kernel to 6.1.23, I haven't experienced my issue for a few days. Maybe these updates helped me. I will monitor it for some time.

morrownr, 2023-04-15:
> After updating the firmware and upgrading the Linux kernel to 6.1.23, I haven't experienced my issue for a few days.

I have never seen firmware files upgraded as often as we are seeing with the mt7921 firmware files. I think we are averaging new files about once per month over the last 6 months. Some of the updates are to support new features but some are also to correct bugs. I can understand this as moving up to WiFi 6 and WiFi 6e capability makes these drivers big and complicated. I monitor lixux-wireless and usually see multiple patches per week. Work is ongoing at a high rate for both the mt7921 and mt7922 chipset drivers and firmware. This should calm down over time.

> it works perfectly with mt7612u but not with mt7921u.

The mt7612u and mt7610u chipsets have drivers and firmware that have had the benefit of time. They are very mature and do what they do in a very stable manner. This will come for these new chipsets. I am working on a rtl8852/32bu driver in the background here with another person. There is a BIG difference between the mt7921u driver and the rtl8852bu driver. The mt7921u is by far more stable and faster. It also handles AP and monitor modes much better.

@morrownr

mazzz1y, 2023-04-17:
Unfortunately, I encountered this problem again. It may be happening less often, but I'm not certain.

Continuing the investigation