morrownr/USB-WiFi #240: News: Debian 12 supports the mt7921au chipset in the default download...

morrownr, 2023-04-10:
Debian has long made it challenging for new users and experienced users that need to install and go, much like me. They have refused to include firmware in their flagship distro forever. This has now changed. I just took a trip over to:

https://cdimage.debian.org/cdimage/weekly-builds/

I downloaded the amd64 weekly test iso, threw it on a flash drive and booted. The installation was easy. I installed the xfce desktop and after a reboot and login, I had to put my cf-951ax adapter (mt7921au chipset) in a usb3 port to see what would happen. It came to life immediately and I was in business.

The world of wifi has seen some challenges over the last couple of years as there have been so many new standards and products coming online. It started with WPA3 and the question was "does my adapter support WPA3?" I have old WiFi 4 adapters that date back to 2009 and they work well with WPA3 so the physical adapter is not the issue. There were many things that had to be updated: wpa_supplicant, NetworkManager, drivers and various other bits of code but we seem to be nearing the point where WPA3 is going to Just Work (t) in all modern distros. Then we have WiFi 6 and don't forget, WiFi 6e is another separate issue. I have posted a sample WiFi 6 hostapd.conf on the Main Menu for those running your own AP. I continue to refine it and if anyone sees anything that can be improved, please speak up. WiFi 6e (6 GHz) is another can of worms. I have not posted a sample for WiFi 6e for hostapd yet. If anyone has a good example hostapd.conf that they would like to submit, please do.

WiFi 6e is complicated by the fact that countries around the world have not passed laws making parts or all of the new 6 GHz channels available for use. The EU seems to be ahead of most of the world. I was reading recently that Taiwan has not passed their new law yet and I immediately thought, how are they going to legally test new 6e products? Maybe there is allowance for that or maybe the article I read was wrong. One way or another, 6 GHz is not necessarily going to work out of the box in many situations yet. This is not just a Linux issue. Windows did not support 6 GHz at all until an upgrade to Windows 11. So, there are challenges that need to be overcome.

Back to the topic: with the new Debian 12 released, we will see many Debian based distros rebasing which will improve support for all of the challenges I have mentioned. Oh, did I mention that Intel, Qualcomm and Mediatek are already merging their WiFi 7 drivers in the kernel?

As you contemplate upgrading your wifi routers and APs to WiFi 6e, keep in mind that the very few reasonably priced 6e capable routers seem to all be using chipsets from the same company that provides the Raspberry Pi internal wifi. Said company has never done a good job of supporting Linux so I will wait until a reasonably prices router is available that uses a Qualcomm or Mediatek chipset. In the meantime, I am hoping the next version of OpenWRT supports 6e.

Regards,

@morrownr

fhteagle, 2023-04-14:
I have a Rpi4 running RaspiOS naturally updated from Buster to Bullseye (kernel 6.1.y) that worked with a CF-953AX (MT7921AU?) on first insert. The upgrade from Buster to Bullseye wasn't pretty, and I needed to compile hostapd from source to get all the latest AP options goodness. But I think for just basic use as an STA, or for relatively basic AP creation, I could recommend.

morrownr, 2023-04-15:
>  I needed to compile hostapd from source to get all the latest AP options goodness.

Yes, compiling hostapd, wpa_supplicant and until now, the kernel. Can't wait unit the RasPiOS is rebased on Debian 12.