morrownr/USB-WiFi #244: x86 Openwrt 22.03.3 with mt7921e mt7921au

shir474, 2023-04-14:
I'm seeing few minor issues with mt7921,.

1) I cannot connect to a hidden AP as a client.  This is true for Openwrt, Ubuntu, or Arch.
2) For mt7921e, the device's initial state is down.
3a) mt7921u as a client connects to an AP, but then disconnects.
3b) #3a works if I have mt7921e enabled on the same device.
4) mt7921u as client and mt7921e as AP, transfer rate is slow, but mt7921u as AP and mt7921 as client, transfer rate is high.

For mt7921au, I'm using Comfast CF-953AX.

Any help would be appreciated.

morrownr, 2023-04-15:
Hi @shir474 

I still don't have my lab set up due to a recent move but I can start looking at one thing at a time slowly.

> mt7921u as client and mt7921e as AP, transfer rate is slow, but mt7921u as AP and mt7921 as client, transfer rate is high.

I have a CF-951AX which uses the mt7921au chipset but does not have the external antennas like you 953. I also have a mt7921e chipset in my laptop but have never used it for AP mode. Are you using hostapd for AP mode? To test this, I really need more details.

shir474, 2023-04-16:
Yes, hostapd-openssl | 2022-01-16-cff80b4f-16.2 and wpa-supplicant-openssl | 2022-01-16-cff80b4f-16.2.

for radio
        option band '5g'
        option channel 'auto'
        option cell_density '0'
        option htmode 'HE80'

for AP/client
        option encryption 'sae'

morrownr, 2023-04-17:
I can confirm issue 1. No hidden SSIDs.

jclark, 2023-05-04:
I tried a  Comfast CF-953AX as an AP on OpenWrt 22.03.5 (on an x86 machine), and it wasn't reliable for me.  I used the wpad-openssl package, and didn't change any defaults other than to enable WPA2-PSK. Running a speed test from an iPhone 14 hung almost immediately with some kernel messages:

```
Thu May  4 05:39:03 2023 kern.err kernel: [  450.578724] mt7921u 2-2:1.3: Message 00020002 (seq 10) timeout
Thu May  4 05:39:03 2023 kern.err kernel: [  450.808726] mt7921u 2-2:1.3: timed out waiting for pending tx
Thu May  4 05:39:03 2023 kern.warn kernel: [  450.837648] ------------[ cut here ]------------
Thu May  4 05:39:03 2023 kern.warn kernel: [  450.846659] WARNING: CPU: 3 PID: 38 at 0xffffffff810d8ad5
Thu May  4 05:39:03 2023 kern.warn kernel: [  450.856529] Modules linked in: pppoe ppp_async option nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet usb_wwan qmi_wwan pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_
reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt792
1u mt7921_common mt76x2u mt76x2e mt76x2_common mt76x02_usb mt76x02_lib mt7603e mt76_usb mt76_connac_lib mt76 mac80211 cfg80211 usbserial usbnet slhc r8169 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag
_ipv6 nf_defrag_ipv4 libcrc32c igc forcedeth e1000e crc_ccitt compat cdc_wdm bnx2 i2c_dev ixgbe e1000 amd_xgbe mdio nls_utf8 ena sha256_ssse3 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac nls_iso8859_1 nls_cp437 vfat fat igb
 button_hotplug tg3 ptp realtek pps_core mii
Thu May  4 05:39:03 2023 kern.warn kernel: [  450.978211] CPU: 3 PID: 38 Comm: kworker/u8:3 Not tainted 5.10.176 #0
Thu May  4 05:39:03 2023 kern.warn kernel: [  450.986818] Hardware name: Default string Default string/Default string, BIOS 5.13 09/13/2022
Thu May  4 05:39:03 2023 kern.warn kernel: [  450.997567] Workqueue: mt76 0xffffffffa0324560 [mt7921_common@00000000d115bbdf+0xc000]
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.007554] RIP: 0010:0xffffffff810d8ad5
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.013220] Code: 32 31 c0 5b 41 5c 5d c3 cc cc cc cc 0f 0b 49 8b 9c 24 20 05 00 00 a8 04 74 ab 0f 0b b8 da ff ff ff 5b 41 5c 5d c3 cc cc cc cc <0f> 0b b8 f0 ff ff ff eb d0 0f 0b eb cc 66 66 2e 0f
 1f 84 00 00 00
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.036016] RSP: 0018:ffffc9000014fd88 EFLAGS: 00010202
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.043003] RAX: 0000000000000004 RBX: ffff8881014e4d00 RCX: 0000000000000002
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.052048] RDX: ffffffff824f0748 RSI: 0000000000000297 RDI: ffff88810156cf80
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.061089] RBP: ffffc9000014fd98 R08: 0000000000000400 R09: 0000000000000000
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.070046] R10: 0000000000000001 R11: 0000000000000000 R12: ffff88810156cf80
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.079157] R13: ffffc9000014fdb0 R14: 0000000000000100 R15: ffff888100a99f98
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.088213] FS:  0000000000000000(0000) GS:ffff888277d80000(0000) knlGS:0000000000000000
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.098283] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.105892] CR2: 00007fb4199c2000 CR3: 000000000340a000 CR4: 0000000000350ee0
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.115009] Call Trace:
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.119166]  0xffffffffa02f3436 [mt76_usb@00000000f6400edc+0x5000]
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.127262]  ? 0xffffffff810f60c0
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.132264]  0xffffffffa0330e99 [mt7921u@00000000d4dc2d3c+0x4000]
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.140198]  0xffffffffa03245eb [mt7921_common@00000000d115bbdf+0xc000]
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.148603]  0xffffffff810d2644
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.153448]  0xffffffff810d27ee
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.158203]  0xffffffff810d8403
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.162989]  ? 0xffffffff810d27a0
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.167881]  ? 0xffffffff810d82e0
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.172805]  0xffffffff81003c1f
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.177505] ---[ end trace 2b75a9992dca1424 ]---
Thu May  4 05:39:03 2023 kern.info kernel: [  451.322002] mt7921u 2-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20220311230842a
Thu May  4 05:39:03 2023 kern.info kernel: [  451.322002]
Thu May  4 05:39:03 2023 kern.info kernel: [  451.351824] mt7921u 2-2:1.3: WM Firmware Version: ____010000, Build Time: 20220311230931
Thu May  4 05:39:10 2023 kern.err kernel: [  458.178722] mt7921u 2-2:1.3: Message 00020003 (seq 9) timeout
Thu May  4 05:39:13 2023 kern.err kernel: [  461.218721] mt7921u 2-2:1.3: Message 00020003 (seq 10) timeout
Thu May  4 05:39:13 2023 kern.err kernel: [  461.448722] mt7921u 2-2:1.3: timed out waiting for pending tx
Thu May  4 05:39:14 2023 kern.info kernel: [  461.613176] 7mt7921u 2-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20220311230842a
Thu May  4 05:39:14 2023 kern.info kernel: [  461.613176]
Thu May  4 05:39:14 2023 kern.info kernel: [  461.643854] mt7921u 2-2:1.3: WM Firmware Version: ____010000, Build Time: 20220311230931
```

I switched over to a MT7912u (a TEROW ROW02CD) and everything seemed to work fine in the brief test I did.

morrownr, 2023-05-04:
Hi @jclark 

> I tried a Comfast CF-953AX as an AP on OpenWrt 22.03.5 (on an x86 machine), and it wasn't reliable for me. 

My Comfast CF-951AX (same chipset as yours) runs as an AP on OpenWRT 22.04 (ZyXEL router, ARM) and it works really well up to and including WiFi 6. It does not do WiFi 6e (6 GHz) yet but OpenWRT is simply not ready for that yet.

The above is just to let you know that it is working well on some systems.

If I had to guess what the problem may be... check the firmware. The mt7921au requires 2 wifi firmware files.  If you go to the Main Menu:

https://github.com/morrownr/USB-WiFi

...then go down to menu item 8 and select it.

You will see that section 3 is specific to OpenWRT.

I use ethtool to check firmware versions on my x86_64 systems. I don't see an ethtool package in OpenWRT so maybe just following the instructions to update the files from from kernel.org will help.

You might also touch base with the maintainer of the x86 distro of OpenWRT as this may be x86 specific.

shir474, 2023-05-06:
> I tried a Comfast CF-953AX as an AP on OpenWrt 22.03.5 (on an x86 machine), and it wasn't reliable for me. I used the wpad-openssl package, and didn't change any defaults other than to enable WPA2-PSK. Running a speed test from an iPhone 14 hung almost immediately with some kernel messages:

I'm not having any issues like that on 22.03.3 with CF-953AX.  I'm using both m.2 and usb version at the same time with iPhone, iPad and Apple Watch.  Maybe the driver is behaving differently between 5.4 and 5.10 kernels.

shir474, 2023-05-06:
> My Comfast CF-951AX (same chipset as yours) runs as an AP on OpenWRT 22.04 (ZyXEL router, ARM) and it works really well up to and including WiFi 6. It does not do WiFi 6e (6 GHz) yet but OpenWRT is simply not ready for that yet.

22.04?  Is this trunk/master?

morrownr, 2023-05-06:
> 22.04? Is this trunk/master?

That is what is commonly known as a typo.

22.03.4, it is fixed.

shir474, 2023-05-06:
> 22.03.4, it is fixed.

So, something new in 22.03.05 maybe the problem?  I'm running 22.03.3 on x86 and don't have any issues with the driver with the firmware shipped with OpeWRT or the latest from the kernel tree.

morrownr, 2023-05-06:
So, let me see if I have this straight:

The ugliness in the log appeared sometime after 22.03.3?
If it an x86 box running OpenWRT with the Comfast usb adapter as an AP?

I can tell you what I don't know: I have no idea if the OpenWRT backported mt7921u driver is getting all or any of the patches that are being submitted. Probably only Felix has a handle on that.

Have you tried the latest snapshot? It runs kernel 5.15.

shir474, 2023-05-06:
> So, let me see if I have this straight:
> 
> The ugliness in the log appeared sometime after 22.03.3? If it an x86 box running OpenWRT with the Comfast usb adapter as an AP?

@jclark is the one with the driver kernel errors running 22.03.5 on x86.  I was just sharing my experience with 22.03.3 on x86.  But yes, that seems to be the case.

jclark, 2023-05-07:
The x86 box I tried it on is just a temporary thing I set up for testing something else. I didn't do anything to set up the firmware.  I'm not completely confident about the nvme drive I'm using: it's an unusual small size (2252) and the only one I could find locally was not from a reputable brand; I also got an fsck error at one point. @shir474's positive experience with a very similar setup makes me wonder if that's the problem. I'll try again on another x86 box.  I've also got an ARM box (a NanoPi) on order specifically to use with OpenWrt; I'll give it a try on that when it arrives.

morrownr, 2023-05-07:
@shir474 

Thanks for lining me out. I had read the threat again before answering but was a little confused. I failed to see that it is 2 of you with more or less the same issue.

@jclark 

Thanks for the additional info.

Both,

Let me suggest something: Since both of you are OpenWRT users, there is an excellent downstream testing location for mt76:

https://github.com/openwrt/mt76

You can read through reported bugs and see what has been patched. You can report bugs. The drivers such as the mt7921u driver may have different source than Linus' kernel at any specific time because patches going into mainline may not be pulled into this driver on a specific schedule and patches merged here may be given time to bake before being pushed to mainline.

I'll keep an eye open but if you guys can narrow this down and report it to the OpenWRT mt76 repo, that is probably the best place. If you do find the answer, please report it here.

shir474, 2023-05-11:
Regarding #3, I used an USB3 extension cable to move the adapter away from each other and the performance is fine.  Since  one was acting as AP and other as client, it was interfering with each other.

shir474, 2023-05-11:
I logged an issue for the hidden AP in Openwrt/mt76.  Will close this issue.

closed 2023-05-11 by shir474

morrownr, 2023-05-11:
> I logged an issue for the hidden AP in Openwrt/mt76. 

Thank you for doing this. It was on my list of things to do but I had not accomplished it yet.