morrownr/USB-WiFi #246: COMFAST CF-953AX

nicobellic1, 2023-04-17:
Hi, I bought a COMFAST CF-953AX wifi dongle from the suggested link, but it's not recognized by the os(Kali kernel version 6.1.0).
I also tried installing the firmware for the chipset following the guide but i get "...and .... are the same file".
Any idea on how to solve this problem?
Thanks for your assistance

nicobellic1, 2023-04-17:
This is the output of lsub if it can be helpful:
Bus 004 Device 002: ID 8087:8000 Intel Corp. Integrated Rate Matching Hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 8087:8008 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 006: ID 3574:6211 MediaTek Inc. Wireless_Device
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04f2:b3fd Chicony Electronics Co., Ltd HD WebCam (Asus N-series)
Bus 001 Device 002: ID 13d3:3402 IMC Networks Bluetooth USB Host Controller
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
 
iw list doesn't show the usb adapter

bjlockie, 2023-04-17:
See https://github.com/morrownr/USB-WiFi/discussions/88#discussioncomment-5088719

nicobellic1, 2023-04-17:
Thank you for pointing to the right direction. I tried doing
`su
modprobe mt7921u
echo 3574 6211 > /sys/bus/usb/drivers/mt7921u/new_id`
but I was unsuccesful.
Is there anything else I could try or I have to waith for the patch morrownr was talking about?

morrownr, 2023-04-17:
Hi @nicobellic1 

As you can see from @bjlockie  's post, we have been down this road before. Sometimes, some makers do this. I think the VID/PID you are seeing went into kernel 6.2 and no request was made to have it added to 6.1.

I haven't had to try to this temporary method but am not sure if you will get good results with everything on one line. Maybe you should try one line at a time
```
su
modprobe mt7921u
echo 3574 6211 > /sys/bus/usb/drivers/mt7921u/new_id
```

> I also tried installing the firmware for the chipset following the guide but i get "...and .... are the same file".

The secret to firmware files is that they don't change name so you just have to copy over what is there:

Go to the following site:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek

Click on `WIFI_MT7961_patch_mcu_1_2_hdr.bin`

Click on `plain`

Save file

Click on `WIFI_RAM_CODE_MT7961_1.bin`

Click on `plain`

Save file

Create the needed directory (if necessary):

```
sudo mkdir /lib/firmware/mediatek
```

Copy the files to the following locations:

```
sudo cp WIFI_MT7961_patch_mcu_1_2_hdr.bin /lib/firmware/mediatek
```

```
sudo cp WIFI_RAM_CODE_MT7961_1.bin /lib/firmware/mediatek
```

Reboot:

```
sudo reboot

Be careful with the firmware guide as section 1 is for the mt7922 chipsets. Section 2 is for the mt7921 chipsets. Both chipsets use the mt7921u driver but have different firmware files so this can be confusing.

ea4eoz, 2023-04-18:
Hello!

I received my Comfast CF-953AX this morning and the device comes with the 3574:6211 USB ID.

It didn't work at first try, because currently I'm running kernel 6.1.0 (Debian unstable) but once I echoed the USB ID into /sys/bus/usb/drivers/mt7921u/new_id worked like a charm in my 802.11ac network. Zero issues.

I hope this helps.

Zen3515, 2023-04-19:
Can anyone confirm my findings, CF-953AX in Access point mode using [config from this repo](https://github.com/morrownr/USB-WiFi/blob/dfb1feb1f64dfc111110f728f4f5d7861aa0b7e3/home/AP_Mode/hostapd-WiFi6.conf)

I can only get maximum of around 240 Mbits/sec with `iperf -c 192.168.1.1`
Is it just me or this card doesn't working very well in AP mode.

nicobellic1, 2023-04-21:
Thank you, now everything work

closed 2023-04-21 by nicobellic1