morrownr/USB-WiFi #247: Unable to use stick Netgear AXE3000 (A8000) with Ubuntu 22.10 LTS

rdpdo, 2023-04-18:
Hello,

I am trying to use the Netgear AXE3000 (A8000) with Ubuntu 22.10 LTS.

I done this : 

```
modprobe mt7921u
sudo vi /sys/bus/usb/drivers/mt7921u/new_id => added 0846 9060
```
But after restart my adapter is not showed in ifconfig.

Here are some command I already tested. Thanks in advance for your help :-)

```
rdpdo@plexsrv:~$ lsusb
Bus 004 Device 002: ID 04e8:4001 Samsung Electronics Co., Ltd PSSD T7
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 0846:9060 NetGear, Inc. Wireless_Device
Bus 002 Device 003: ID 0bc2:331a Seagate RSS LLC Desktop HDD 5TB (ST5000DM000)
Bus 002 Device 002: ID 174c:1153 ASMedia Technology Inc. ASM1153 SATA 3Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. Bluetooth wireless interface
Bus 001 Device 002: ID 1c4f:0027 SiGma Micro USB Keyboard
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
rdpdo@plexsrv:~$
```
The one I want is : Bus 002 Device 004: ID 0846:9060 NetGear, Inc. Wireless_Device

```
rdpdo@plexsrv:~$ uname -a
Linux plexsrv 5.19.0-38-generic #39-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 17 17:33:16 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
rdpdo@plexsrv:~$
```

```
rdpdo@plexsrv:~$ lsmod | grep cfg80211
cfg80211             1040384  3 iwlmvm,iwlwifi,mac80211
rdpdo@plexsrv:$
```

```
rdpdo@plexsrv:~$ lsusb -v
Bus 002 Device 002: ID 0846:9060 NetGear, Inc. Wireless_Device
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         9
  idVendor           0x0846 NetGear, Inc.
  idProduct          0x9060
  bcdDevice            1.00
  iManufacturer           2 MediaTek Inc.
  iProduct                3 Wireless_Device
  iSerial                 4 000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0087
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          5
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              160mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           9
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x07  EP 7 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x09  EP 9 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1
        bMaxBurst               0
```

rdpdo, 2023-04-18:
In fact the file /sys/bus/usb/drivers/mt7921u/new_id is not completly saved after rebooting...
Do you know how I can save this file so it will not be erased after reboot ?

morrownr, 2023-04-18:
Hi @rdpdo 

> Do you know how I can save this file so it will not be erased after reboot ?

Does that mean that the adapter is working as long as you do not reboot?

The only way to have this work automatically is to create a script that runs automatically during boot. I guess maybe I could write one but you would have to test it. This method is meant to be a temp solution and... see below...

The good news is that I saw a message in linux-wireless this morning that confirms that the VID/PID is going into kernel 6.4 so this will work out good at some point. @russeree  - did you see the message? Thanks again for submitting the patch. Maybe, once 6.4 rcx is released and confirmed to be working, you can put in requests for kernels 6.1 and 6.1... remember it requires a different method.

@morrownr

russeree, 2023-04-18:
I did not see the message and have got no replies to my patch submission; none the less that is great to hear. Once tested and verified in release I will gladly work on backporting the patch.

rdpdo, 2023-04-18:
Hi @morrownr
Yes it work as long I do not reboot ... I will try to make a script thanks you.

russeree, 2023-04-18:
@rdpdo if you don't mind compiling the Kinetic kernel with a patch you will certainly avoid the reboot issue. 

This patch will do the job quite nicely. https://patchwork.kernel.org/project/linux-wireless/patch/20230404073516.33266-1-git@qrsnap.io/

rdpdo, 2023-04-18:
Hi @russeree
Thanks but I do not know how to compile a kernel do you have a link wich explain this ?

russeree, 2023-04-18:
This link will do. I would recommend practicing inside of a VM first since doing this on a production machine can cause hiccups.

 
https://wiki.ubuntu.com/Kernel/BuildYourOwnKernel

rdpdo, 2023-04-18:
Great thanks !

morrownr, 2023-04-18:
> I did not see the message and have got no replies to my patch submission; none the less that is great to hear. Once tested and verified in release I will gladly work on backporting the patch.

I think you are expected to track progress on your own. So, I guess this was a joint effort. You did the work and I tracked the progress. He he. Now we need a guide that steps newbies through the process.

See #248 for a posting of most of the first PR for kernel 6.4.

rdpdo, 2023-04-19:
I am unable to find the proper deb-src to include in my apt ... can you give me please so I can download depenpendancies packages ? (sudo apt-get build-dep linux linux-image-$(uname -r))

morrownr, 2023-04-19:
@rdpdo 

> Do you know what's going wrong ?

To tell you the truth, I did not look at the guide you are using. Since you are using Ubuntu, I have a guide on the Main Menu here. Look at menu item 12.

rdpdo, 2023-04-20:
Ok thanks you for your help :-)

closed 2023-04-20 by rdpdo