morrownr/USB-WiFi #248: News: mt76 patches for 6.4 (new VID/PID for Netgear A8000 going in... thanks Reece

morrownr, 2023-04-18:
```
mt76 patches for 6.4

- fixes
- connac code unification
- mt7921 p2p support
- mt7996 mesh a-msdu support
- mt7996 eht support
- mt7996 coredump support

----------------------------------------------------------------
Bo Jiao (1):
       wifi: mt76: mt7996: enable full system reset support

Dan Carpenter (1):
       wifi: mt76: mt7915: unlock on error in mt7915_thermal_temp_store()

Deren Wu (3):
       wifi: mt76: remove redundent MCU_UNI_CMD_* definitions
       wifi: mt76: mt7921: fix wrong command to set STA channel
       wifi: mt76: mt7921: fix PCI DMA hang after reboot

Felix Fietkau (4):
       wifi: mt76: add missing locking to protect against concurrent rx/status calls
       wifi: mt76: mt7615: increase eeprom size for mt7663
       wifi: mt76: dma: use napi_build_skb
       wifi: mt76: set NL80211_EXT_FEATURE_CAN_REPLACE_PTK0 on supported drivers

Frank Wunderlich (1):
       dt-bindings: mt76: add active-low property for led

Gustavo A. R. Silva (2):
       wifi: mt76: Replace zero-length array with flexible-array member
       wifi: mt76: mt7921: Replace fake flex-arrays with flexible-array members

Howard Hsu (1):
       wifi: mt76: mt7915: rework init flow in mt7915_thermal_init()

Jiefeng Li (1):
       wifi: mt76: mt7921: fix missing unwind goto in `mt7921u_probe`

Kang Chen (1):
       wifi: mt76: handle failure of vzalloc in mt7615_coredump_work

Lorenz Brun (1):
       wifi: mt76: mt7915: expose device tree match table

Lorenzo Bianconi (8):
       wifi: mt76: mt7921: introduce mt7921_get_mac80211_ops utility routine
       wifi: mt76: move irq_tasklet in mt76_dev struct
       wifi: mt76: add mt76_connac_irq_enable utility routine
       wifi: mt76: get rid of unused sta_ps callbacks
       wifi: mt76: add mt76_connac_gen_ppe_thresh utility routine
       wifi: mt76: mt7921: get rid of eeprom.h
       wifi: mt76: move shared mac definitions in mt76_connac2_mac.h
       wifi: mt76: move mcu_uni_event and mcu_reg_event in common code

Mario Limonciello (1):
       wifi: mt76: mt7921e: Set memory space enable in PCI_COMMAND if unset

Ming Yen Hsieh (1):
       wifi: mt76: fix 6GHz high channel not be scanned

Neil Chen (1):
       wifi: mt76: mt7921: use driver flags rather than mac80211 flags to mcu

Peter Chiu (4):
       wifi: mt76: drop the incorrect scatter and gather frame
       wifi: mt76: mt7996: fix pointer calculation in ie countdown event
       wifi: mt76: mt7996: init mpdu density cap
       wifi: mt76: mt7996: remove mt7996_mcu_set_pm()

Quan Zhou (3):
       wifi: mt76: mt7921e: fix probe timeout after reboot
       wifi: mt76: mt7921e: improve reliability of dma reset
       wifi: mt76: mt7921e: stop chip reset worker in unregister hook

Reese Russell (1):
       wifi: mt76: mt7921: add Netgear AXE3000 (A8000) support

Ryder Lee (14):
       wifi: mt76: mt7996: fix radiotap bitfield
       wifi: mt76: dynamic channel bandwidth changes in AP mode
       wifi: mt76: connac: refresh tx session timer for WED device
       wifi: mt76: mt7915: remove mt7915_mcu_beacon_check_caps()
       wifi: mt76: mt7996: remove mt7996_mcu_beacon_check_caps()
       wifi: mt76: mt7915: drop redundant prefix of mt7915_txpower_puts()
       wifi: mt76: mt7996: add full system reset knobs into debugfs
       wifi: mt76: mt7996: enable coredump support
       wifi: mt76: connac: fix txd multicast rate setting
       wifi: mt76: connac: add nss calculation into mt76_connac2_mac_tx_rate_val()
       wifi: mt76: mt7996: enable BSS_CHANGED_BASIC_RATES support
       wifi: mt76: mt7996: enable BSS_CHANGED_MCAST_RATE support
       wifi: mt76: mt7996: enable configured beacon tx rate
       wifi: mt76: mt7996: enable mesh HW amsdu/de-amsdu support

Sean Wang (2):
       wifi: mt76: mt7921: enable p2p support
       mt76: mt7921: fix kernel panic by accessing unallocated eeprom.data

Shayne Chen (3):
       wifi: mt76: mt7996: add eht rx rate support
       wifi: mt76: mt7996: let non-bufferable MMPDUs use correct hw queue
       wifi: mt76: mt7996: remove unused eeprom band selection

StanleyYP Wang (1):
       wifi: mt76: mt7996: fix eeprom tx path bitfields

Sujuan Chen (1):
       wifi: mt76: mt7915: add dev->hif2 support for mt7916 WED device

Yang Li (1):
       wifi: mt76: mt7996: Remove unneeded semicolon

```

russeree, 2023-04-18:
This is great to see that such a great adapter is going to actually be plug and play. I will work on a guide for others on how to patch/submit patches as well.

morrownr, 2023-04-20:
> This is great to see that such a great adapter is going to actually be plug and play. 

Now that you have used the A8000 for a while, does your review in the Plug and Play list need to be updated?

> I will work on a guide for others on how to patch/submit patches as well.

I think a lot of Linux users would appreciate that. The need to patch in new VID/PID's is going to happen as more adapters come on the market so having a good guide would be wonderful.

Thanks again.

morrownr, 2023-05-04:
In kernel 6.4 it goes...

```
From: Reese Russell <git@qrsnap.io>

[ Upstream commit 03eb52dd78cab08f13925aeec8315fbdbcba3253 ]

Issue: Though the Netgear AXE3000 (A8000) is based on the mt7921
chipset because of the unique USB VID:PID combination this device
does not initialize/register. Thus making it not plug and play.

Fix: Adds support for the Netgear AXE3000 (A8000) based on the Mediatek
mt7921au chipset. The method of action is adding the USD VID/PID
pair to the mt7921u_device_table[] array.

Notes: A retail sample of the Netgear AXE3000 (A8000) yeilds the following
from lsusb D 0846:9060 NetGear, Inc. Wireless_Device. This pair
0846:9060 VID:PID has been reported by other users on Github.

---
 drivers/net/wireless/mediatek/mt76/mt7921/usb.c | 3 +++
 1 file changed, 3 insertions(+)

diff --git a/drivers/net/wireless/mediatek/mt76/mt7921/usb.c b/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
index 8fef09ed29c91..a95ec388653c5 100644
--- a/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
+++ b/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
@@ -18,6 +18,9 @@ static const struct usb_device_id mt7921u_device_table[] = {
        /* Comfast CF-952AX */
        { USB_DEVICE_AND_INTERFACE_INFO(0x3574, 0x6211, 0xff, 0xff, 0xff),
                .driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
+       /* Netgear, Inc. [A8000,AXE3000] */
+       { USB_DEVICE_AND_INTERFACE_INFO(0x0846, 0x9060, 0xff, 0xff, 0xff),
+               .driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
        { },
 };
```

Reece,

Any possibility of you testing kernel 6.4 after rc1 is released this Sunday?