morrownr/USB-WiFi #249: A word of caution about TP-Link Archer T1U

ea4eoz, 2023-04-18:
Hello!

I've seen the TP-Link Archer T1U in the list of supported devices with Linux in-kernel drivers. I brought my TP-Link Archer T1U some years ago and I have never been able to use it in a 802.11ac network. Only in 5 GHz 802.11n ones.

The device works, scans, see networks and connect flawlessly to 802.11n networks, but if you try to connect to a 802.11ac network it does not connect, falling in a endless loop of authentication timed outs. Tested on many many kernels through these years, being the last one the kernel 6.1.0 I'm using now (Debian Sid).

I wrote a message in the linux-wireless mailing list but it was ignored:

https://www.spinics.net/lists/linux-wireless/msg192785.html

Messages in syslog are always like these:

send auth to AP (try 1/3)
send auth to AP (try 2/3)
send auth to AP (try 3/3)
authentication with AP timed out

So be careful using this adapter with Linux.

Greetings

morrownr, 2023-04-19:
Hi @ea4eoz 

I have removed this item from the Plug and Play list.

A user requested that I add this adapter and I resisted but finally gave as I was unable to find any negative info. Now the list is TP-Link free again. I feel much better.

Thanks for the info.

@morrownr

ea4eoz, 2023-04-19:
Good decision. I also wouldn't recommend it now that 802.11n networks in the 5GHz band are residual.

@ea4eoz

closed 2023-04-19 by ea4eoz

Anthony96922, 2023-05-20:
I'm the one who requested the T1U be added to the list. TBH I don't mind having it removed. I think the problem you were facing is due to it having an extremely low gain antenna. The adapter can hear the AP but the AP can't hear the adapter. However it seems to work well as a short range AP which is what I use it for. I never used it as a client station so I never saw that issue.

morrownr, 2023-05-21:
@Anthony96922  @ea4eoz 

I generally try to only list only new products and I generally don't list TP-Link products because some of their business practices make things hard on Linux users but Anthony made a good case and there is demand for a nano dual band adapter that uses an in-kernel driver so after some internet searching, I added the adapter to the list with a warning that it is only available as a used product.

I will add the adapter back into the list if I see that you two come to a meeting of the minds that it should be added back.

morrownr, 2023-05-21:
For what it is worth, I just sent a message to a vendor and a maker requesting a new nano adapter that works with Linux in-kernel drivers. This is not me pulling something out of my backside. I actually have a relationship with both so the request will be taken seriously. I can't say that it will be approved but even if it is, concept to shipping takes from 6 months to over a year so don't hold your breathe.

Anthony96922, 2023-05-22:
I don't mind having it added back. Maybe add a disclaimer stating it's a short range adapter.

ea4eoz, 2023-06-12:
I don't mind if this adapter returns to the list: It is very old and is no longer for sale, so it will probably do not hurt anyone. 

I made several tests with it, using two different AP, in the same room as the adapter and the ONLY way to make it work was configuring the APs to 802.11n mode only. Modes like 802.11n+ac or 802.11ac only produced the infinite connection loop I mentioned earlier in this thread. I didn't test it against a 802.11a access point.

About its range it is similar to a handheld phone. It has no problem to connect to my AP (in N mode) at the other side of the house (with several walls in between) with reasonable speed. Range is not the problem for this adapter.

morrownr, 2023-06-15:
I am not ignoring the above post. I am simply very busy. 

For now, I think I will leave this adapter out of the list as I am getting some indications that we may get a new nano plug and play adapter based on more modern technology later this year you. I can readdress this issue if need be down the line.