morrownr/USB-WiFi #251: Atheros AR9271 Firmware Transfer Pipe1 Failure

elockman, 2023-04-19:
I have enabled AR9271 support for embedded Linux v5.15.68. 
```
root@sama5d2-xplained:~# uname -r
5.15.68-linux4microchip-2022.10
```
When I plug the device in, it detects it but seems to fail during the frimware transfer.
The blocks of debug continually loop until the device is unplugged.
Has anyone seen this, or have any suggestions to resolve the issue?
```
sama5d2-xplained login: root
root@sama5d2-xplained:~# usb 1-2: new full-speed USB device number 2 using at91_ohci
usb 1-2: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
usb 1-2: New USB device strings: Mfr=16, Product=32, SerialNumber=48
usb 1-2: Product: UB93
usb 1-2: Manufacturer: ATHEROS
usb 1-2: SerialNumber: 12345
usb 1-2: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
usbcore: registered new interface driver ath9k_htc
usb 1-2: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
------------[ cut here ]------------
WARNING: CPU: 0 PID: 61 at drivers/usb/core/urb.c:502 0xc05c3eac
usb 1-2: BOGUS urb xfer, pipe 1 != type 3
Modules linked in: ath9k_htc ath9k_common ath9k_hw ath rt2800usb rt2800lib crc_ccitt rt2x00usb rt2x00lib mac80211 cfg80211 8021q g_serial libcomposite
CPU: 0 PID: 61 Comm: kworker/0:2 Not tainted 5.15.68-linux4microchip-2022.10 #1
Hardware name: Atmel SAMA5
Workqueue: events 0xc0520ec0
Function entered at [<c010c218>] from [<c01096dc>]
Function entered at [<c01096dc>] from [<c08fd950>]
Function entered at [<c08fd950>] from [<c08fda24>]
Function entered at [<c08fda24>] from [<c05c3eac>]
Function entered at [<c05c3eac>] from [<bf158a2c>]
Function entered at [<bf158a2c>] from [<bf157340>]
Function entered at [<bf157340>] from [<bf159a94>]
Function entered at [<bf159a94>] from [<bf15fba8>]
Function entered at [<bf15fba8>] from [<bf15fe94>]
Function entered at [<bf15fe94>] from [<bf1577ec>]
Function entered at [<bf1577ec>] from [<bf159264>]
Function entered at [<bf159264>] from [<c0520f10>]
Function entered at [<c0520f10>] from [<c012e75c>]
Function entered at [<c012e75c>] from [<c012edf4>]
Function entered at [<c012edf4>] from [<c0134e94>]
Function entered at [<c0134e94>] from [<c0100130>]
Exception stack(0xc1275fb0 to 0xc1275ff8)
5fa0:                                     00000000 00000000 00000000 00000000
5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
---[ end trace 14fef60cbc8027d6 ]---
------------[ cut here ]------------
WARNING: CPU: 0 PID: 61 at drivers/usb/core/urb.c:502 0xc05c3eac
usb 1-2: BOGUS urb xfer, pipe 1 != type 3
Modules linked in: ath9k_htc ath9k_common ath9k_hw ath rt2800usb rt2800lib crc_ccitt rt2x00usb rt2x00lib mac80211 cfg80211 8021q g_serial libcomposite
CPU: 0 PID: 61 Comm: kworker/0:2 Tainted: G        W         5.15.68-linux4microchip-2022.10 #1
Hardware name: Atmel SAMA5
Workqueue: events 0xc0520ec0
Function entered at [<c010c218>] from [<c01096dc>]
Function entered at [<c01096dc>] from [<c08fd950>]
Function entered at [<c08fd950>] from [<c08fda24>]
Function entered at [<c08fda24>] from [<c05c3eac>]
Function entered at [<c05c3eac>] from [<bf158a2c>]
Function entered at [<bf158a2c>] from [<bf157340>]
Function entered at [<bf157340>] from [<bf15fbec>]
Function entered at [<bf15fbec>] from [<bf15fe94>]
Function entered at [<bf15fe94>] from [<bf1577ec>]
Function entered at [<bf1577ec>] from [<bf159264>]
Function entered at [<bf159264>] from [<c0520f10>]
Function entered at [<c0520f10>] from [<c012e75c>]
Function entered at [<c012e75c>] from [<c012edf4>]
Function entered at [<c012edf4>] from [<c0134e94>]
Function entered at [<c0134e94>] from [<c0100130>]
Exception stack(0xc1275fb0 to 0xc1275ff8)
5fa0:                                     00000000 00000000 00000000 00000000
5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
---[ end trace 14fef60cbc8027d7 ]---
------------[ cut here ]------------
WARNING: CPU: 0 PID: 61 at drivers/usb/core/urb.c:502 0xc05c3eac
usb 1-2: BOGUS urb xfer, pipe 1 != type 3
Modules linked in: ath9k_htc ath9k_common ath9k_hw ath rt2800usb rt2800lib crc_ccitt rt2x00usb rt2x00lib mac80211 cfg80211 8021q g_serial libcomposite
CPU: 0 PID: 61 Comm: kworker/0:2 Tainted: G        W         5.15.68-linux4microchip-2022.10 #1
Hardware name: Atmel SAMA5
Workqueue: events 0xc0520ec0
Function entered at [<c010c218>] from [<c01096dc>]
Function entered at [<c01096dc>] from [<c08fd950>]
Function entered at [<c08fd950>] from [<c08fda24>]
Function entered at [<c08fda24>] from [<c05c3eac>]
Function entered at [<c05c3eac>] from [<bf158a2c>]
Function entered at [<bf158a2c>] from [<bf157340>]
Function entered at [<bf157340>] from [<bf15fc28>]
Function entered at [<bf15fc28>] from [<bf15fe94>]
Function entered at [<bf15fe94>] from [<bf1577ec>]
Function entered at [<bf1577ec>] from [<bf159264>]
Function entered at [<bf159264>] from [<c0520f10>]
Function entered at [<c0520f10>] from [<c012e75c>]
Function entered at [<c012e75c>] from [<c012edf4>]
Function entered at [<c012edf4>] from [<c0134e94>]
Function entered at [<c0134e94>] from [<c0100130>]
Exception stack(0xc1275fb0 to 0xc1275ff8)
5fa0:                                     00000000 00000000 00000000 00000000
5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
---[ end trace 14fef60cbc8027d8 ]---
------------[ cut here ]------------

```

morrownr, 2023-04-19:
See if this link helps:

https://wireless.wiki.kernel.org/en/users/drivers/ath9k_htc

elockman, 2023-04-20:
Moving on to WiFi 6. Closing issue.