morrownr/USB-WiFi #252: (Question) 802.11ax WiFi 6 mesh implementation

elockman, 2023-04-20:
Hello all,

I've been running a mesh network with the RT5370 WiFi-4 usb adapters, and have found availability is becoming very difficult. I'm venturing on into WiFi 6 and have some AX1800 adapters on order. Unfortunately, I'm not sure of the chipsets yet, which may prove disastrous after hearing about driver issues.

My question is whether 802.11ax WiFi Mesh requires a router, or if ad hoc mesh networks can be formed between multiple laptops with WiFi 6 adapters. Internet connectivity is not a concern, so I'm just looking to connect devices with mesh capabilities.

If anyone has experience or can point me to examples or tutorials, I'd appreciate it.

Thanks.

elockman, 2023-04-21:
Does anyone have a source in the USA for a mt7921au usb wifi module? I'm not finding good solutions for 802.11ax.

elockman, 2023-04-21:
@morrownr if you have the mt7921au, can you tell me if it supports mesh point mode?

bjlockie, 2023-04-22:
Do you mean the chips or the adapter?
I think all the chips are made in Taiwan.
I'd contact Mediatek for the closest reseller.
Rockland (Florida I think) sells the Alfa.

elockman, 2023-04-22:
Thank you, I'll contact them. I'm really wondering if the USB adapters support mesh with the built-in Linux drivers. I had to to update my kernel from 5.15 to 6.1 to get over the 5.18 minimum, but I'm waiting on hardware.

morrownr, 2023-04-22:
@elockman 

The follow is the link to the AXML at Rokland (US):

https://store.rokland.com/collections/wifi-6-6e/products/alfa-awus036axml-802-11ax-wifi-6-1800-mbps-tri-band-wifi-usb-adapter-w-bluetooth

Rokland's customer support and public relations is excellent. They support Linux users and they actually know what they are talking about. One of the very rare reselers that is capable of and will take care of Linux users.

Regarding your question about whether the mt7921au support mesh for your project: I'm willing to test and provide results but mesh is not an area that I know a lot about so you would need to tell me what to do and what to look for.

@morrownr

elockman, 2023-04-23:
I was just looking for **mesh point** listed as Supported interface modes, but I ordered a [Netgear AXE3000](https://www.amazon.com/gp/product/B0B94R78N7). It should arrive Monday night, so I should be able to check it out. 

If it does support mesh, I'll get more to test an ad hoc mesh network with and run some iperf tests and healing tests. I will report what I find.

morrownr, 2023-04-23:
> mesh point listed as Supported interface modes

Okay.

Alfa ACHM - mt7610u chipset - AC600, basically any adapter that uses the mt7610u chipset should show the same.

Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO

Alfa ACM - mt7612u chipset - AC1200, basically any adapter that uses the mt7612u chipset should show the same.

	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO

Netgear A8000 - mt7921au chipset - AXE3000, basically any adapter that uses the mt7921au chipset should show the same right now but the mt7921u driver is still under development. P2P is going into the mt7921u driver with kernel 6.4 which is about to begin merging new features. There is no doubt in my mind that mesh point support will be added, probably this year. I have noticed that with WiFi6/6e the size and complexity of the drivers has grown a lot which is probably why Mediatek has taken the approach of getting the driver in the kernel and then add features as able. The mt7921u driver originally showed in kernel 5.18 but only supported managed and monitor modes. Then AP and AP/VLAN was added. Now P2P is being added. I would estimate the mesh point and IBSS will be here soon enough.

	Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor

Regarding the Netgear A8000: I don't have one but the users that have bought one speak highly of it so I hope you enjoy the adapter. The Device ID will need to be added for now for the driver to be able to ID the adapter. In menu item 2, in the section that has the Netgear A8000, there are 2 methods shown to help you add the ID until such time as your distro is using kernel 6.4 or later. If you need help, ask.

@morrownr

elockman, 2023-04-24:
@morrownr 

Thank you, that helps. Do you know the difference between MT7610U and MT7610UN? I'm looking for a smaller form factor.

morrownr, 2023-04-24:
> Do you know the difference between MT7610U and MT7610UN?

I do not but will see if I can find out.

>  I'm looking for a smaller form factor.

Go to the Main Menu:

https://github.com/morrownr/USB-WiFi

Open menu item 2 and go to the section about the mt7610u.

I see 7 adapters listed and some are small. It has been about a month since I went over the list so if something is out of stock, blame me.

elockman, 2023-04-25:
@morrownr 

When I plug the ALFA AWUS036ACHM into my development board, the entire board resets. I'm running kernel 6.1. 

I looked at power, it is pulling about twice the power of an RT5370 or Rt3070. Power draw still seems low, however. Max around 92mA instead of 55mA. I don't want power to be a red herring, just an observation.

Is this something you've seen with the ALFA AWUS036ACHM, or other WiFi adapters?

morrownr, 2023-04-25:
Do you have the ability to use any of the following kernels?

5.15, 5.19 or 6.2?

The reason I ask is that there was a patch submitted by an Intel dev at the beginning of kernel 6.0 testing and it messed up certain wifi drivers. It was a change in the stack... I think it was mac80211. One of the Mediatek devs jumped in and cleaned up the mess but major changes to the stack take some time and it was until sometime during kernel 6.1 that it was fixed. The only guidance we could give while it was being fixed was to use a previous kernel while the fix is being tested. It could be that you are using one of the bad kernels.

elockman, 2023-04-26:
Thanks, you are a life-saver. I should be able to revert to v5.15.

elockman, 2023-04-26:
@morrownr thanks again for the guidance.

Follow up:
I reverted to kernel 5.15 and had the same issues. I'm still not sure if it is power related (considering the ALFA AWUS036ACHM is a long range adapter)

However, the Panda PAU0A works as expected. It uses the mt7610un (per chip markings).

lsusb yields:
ID 0e8d:7610 MediaTek Inc. WiFi

The adapter is available on Amazon [here](https://www.amazon.com/Panda-Wireless-PAU0A-AC600-Adapter/dp/B07C9TYDR4).

morrownr, 2023-04-26:
> I reverted to kernel 5.15 and had the same issues. I'm still not sure if it is power related (considering the ALFA AWUS036ACHM is a long range adapter)

Hmmm... manufacturing problems happen. If the Panda with the same chipset is working, the Alfa should work. You may need to call your seller.