morrownr/USB-WiFi #253: RT5370 doesn't support simultaneous STA + AP?

tve, 2023-04-22:
I just got an RT5370 adapter (panda nano) and was hoping to use it for simultaneous STA + AP, but that doesn't seem to work. Is anyone else able to do that?

I'm using the 2023-02-22 Raspberry Pi OS image in a Compute Module 3, kernel 5.15. The driver used is rt2800usb included in the distro.

The symptoms I'm seeing are that I can start wlan0 or ap0 individually but any attempt to combine them results in a `Device or resource busy` error. I can't even `ip link set <interface> up` both of them simultaneously.

dmesg discovery:
```
[    6.270985] usb 1-1.2.4: new high-speed USB device number 9 using dwc_otg                   
[    6.448631] usb 1-1.2.4: New USB device found, idVendor=148f, idProduct=5370, bcdDevice= 1.0
1                                                                                              
[    6.464512] usb 1-1.2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3           
[    6.475754] usb 1-1.2.4: Product: 802.11 n WLAN                                             
[    6.483885] usb 1-1.2.4: Manufacturer: Ralink                                               
[    6.491709] usb 1-1.2.4: SerialNumber: 1.0                                                  
```

Trying to `ip up` both interfaces:
```
> ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 00:50:b6:0d:31:8b brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.137/24 brd 192.168.0.255 scope global dynamic noprefixroute eth0
       valid_lft 2519sec preferred_lft 2069sec
    inet6 fe80::aa9b:b7f4:c3f5:dd46/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 9c:ef:d5:f9:83:cf brd ff:ff:ff:ff:ff:ff
4: ap0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 9c:ef:d5:f9:83:c8 brd ff:ff:ff:ff:ff:ff
> sudo ip link set wlan0 up
RTNETLINK answers: Device or resource busy
> sudo ip link set ap0 down
> sudo ip link set wlan0 up
> sudo ip link set ap0 up
RTNETLINK answers: Device or resource busy
```

Trying to start hostapd manually when wlan0 is up results in the same error:
```
gnome@SG-FC32RPS1EE0F ~> sudo hostapd -d /etc/hostapd/hostapd.conf
random: getrandom() support available
Configuration file: /etc/hostapd/hostapd.conf
nl80211: Supported cipher 00-0f-ac:1
...
nl80211: Using driver-based off-channel TX
nl80211: Driver-advertised extended capabilities (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: Driver-advertised extended capabilities mask (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: interface ap0 in phy phy0
nl80211: Set mode ifindex 4 iftype 3 (AP)
nl80211: Setup AP(ap0) - device_ap_sme=0 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x7ab010
nl80211: Register frame type=0xb0 (WLAN_FC_STYPE_AUTH) nl_handle=0x7ab010 match=
nl80211: Register frame type=0x0 (WLAN_FC_STYPE_ASSOC_REQ) nl_handle=0x7ab010 match=
nl80211: Register frame type=0x20 (WLAN_FC_STYPE_REASSOC_REQ) nl_handle=0x7ab010 match=
nl80211: Register frame type=0xa0 (WLAN_FC_STYPE_DISASSOC) nl_handle=0x7ab010 match=
nl80211: Register frame type=0xc0 (WLAN_FC_STYPE_DEAUTH) nl_handle=0x7ab010 match=
nl80211: Register frame type=0x40 (WLAN_FC_STYPE_PROBE_REQ) nl_handle=0x7ab010 match=
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x7ab010 match=04
...
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
Could not set interface ap0 flags (UP): Device or resource busy
nl80211: Could not set interface 'ap0' UP
```

iw seems to claim that interface combinations are supported:
```
        valid interface combinations:
                 * #{ AP, mesh point } <= 8,
                   total <= 8, #channels <= 1
```

morrownr, 2023-04-22:
Hi @tve 

> iw seems to claim that interface combinations are supported:

Yes, interface combinations are supported on most in-kernel drivers. However, what combinations are supported depends on the driver.

```
        valid interface combinations:
                 * #{ AP, mesh point } <= 8,
                   total <= 8, #channels <= 1
```

What that means is that you can have up to 8 interfaces in any combination of AP or mesh point. You can have 8 AP interfaces or 4 AP and 4 mesh point, etc. STA is not listed so it is not a supported interface for combinations with this driver.

There are several in-kernel drivers that will support the combination you want.

morrownr, 2023-04-22:
More info:

```
valid interface combinations:
		 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 2,
		   total <= 2, #channels <= 1, STA/AP BI must match
```

That is from the mt7612u driver on the Panda PAU0D but will be the same with any adapter that uses the mt7612u driver.. You can have up to 2 interfaces and the combination can be any combination of managed, AP, mesh point, P2P-client, P2P-GO.

Does this make sense?

morrownr, 2023-04-23:
I am on my laptop now. It uses a card based on the mt7921 chipset and uses the mt7921e driver.

```
valid interface combinations:
		 * #{ managed } <= 2, #{ AP } <= 1,
		   total <= 2, #channels <= 2
```

Here is a question to see if you are picking up on this:

Can you set up what you were looking for with this driver?

Now I am checking a usb wifi adapter with a mt7921au chipset:

```
valid interface combinations:
		 * #{ managed } <= 4, #{ AP } <= 1,
		   total <= 4, #channels <= 1, STA/AP BI must match
```

Will that do what you wanted?

tve, 2023-04-23:
I see, I misinterpreted the interface combinations output. It's pretty cryptic. Disappointing...

morrownr, 2023-04-23:
> Do the mt7610u-based adapters support STA+AP mode?

Yes

```
valid interface combinations:
		 * #{ IBSS } <= 1, #{ managed, AP, mesh point, P2P-client, P2P-GO } <= 2,
		   total <= 2, #channels <= 1, STA/AP BI must match
```

Main Menu item 4 contains many files with technical information about many adapters and chipsets. I need to work on it so as to make sure each filename shows the chipset but the work around here never seems to finish.

tve, 2023-04-23:
thanks for maintaining all this info!!!