morrownr/USB-WiFi #254: mt7921au Comfast CF-953AX in NUC-PC Manjaro Kernel 6.2 hangs

sektorhybrid, 2023-04-23:
Hello

I have the following setup 
- Intel NUC KIT Nuc6cayh Celeron J3455 Mini PC 
- Comfast CF-953AX (0e8d:796)
- hostapd v2.10
- Kernel 6.2.12-1-MANJARO

The box freezes always while receiving the first 4-Way handshake from the first STA trying to connect. 
I tested with 4G, 5G and 6G with exact the same result. 
I tested directly in USB2 port, USB3 port, externally powered USB2 and USB3 hubs, USB2 and USB3 extension cord etc

In 4G, the same settings are working flawlessly for all the adapters except the CF953AX, also sharing my udev rules

```
# Num1
SUBSYSTEM=="net",ACTION=="add",ATTRS{idVendor}=="0cf3",ATTRS{idProduct}=="9271",NAME="wlanap_WN722N"
#Num2
SUBSYSTEM=="net",ACTION=="add",ATTRS{idVendor}=="2357",ATTRS{idProduct}=="0106",NAME="wlanap_T9UH"
#Num3
SUBSYSTEM=="net",ACTION=="add",ATTRS{idVendor}=="0b05",ATTRS{idProduct}=="1841",NAME="wlanap_AC54"
#Num 4
SUBSYSTEM=="net",ACTION=="add",ATTRS{idVendor}=="2001",ATTRS{idProduct}=="3315",NAME="wlanap_DWA182"
#Num5
SUBSYSTEM=="net",ACTION=="add",ATTRS{idVendor}=="0e8d",ATTRS{idProduct}=="7961",NAME="wlanap_CF953AX"
```

Logs from previous boot that ultimately hanged do not show a Kernel Panic

```
Απρ 22 02:48:42.215565 kitana kernel: Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
Απρ 22 02:48:42.242930 kitana bluetoothd[2081]: Bluetooth daemon 5.66
Απρ 22 02:48:42.249272 kitana systemd[1]: Started Bluetooth service.
Απρ 22 02:48:42.249910 kitana systemd[1]: Reached target Bluetooth Support.
Απρ 22 02:48:42.250864 kitana bluetoothd[2081]: Starting SDP server
Απρ 22 02:48:42.255475 kitana bluetoothd[2081]: profiles/audio/vcp.c:vcp_init() D-Bus experimental not enabled
Απρ 22 02:48:42.255500 kitana bluetoothd[2081]: src/plugin.c:plugin_init() Failed to init vcp plugin
Απρ 22 02:48:42.255513 kitana bluetoothd[2081]: profiles/audio/mcp.c:mcp_init() D-Bus experimental not enabled
Απρ 22 02:48:42.255523 kitana bluetoothd[2081]: src/plugin.c:plugin_init() Failed to init mcp plugin
Απρ 22 02:48:42.255533 kitana bluetoothd[2081]: profiles/audio/bap.c:bap_init() D-Bus experimental not enabled
Απρ 22 02:48:42.255543 kitana bluetoothd[2081]: src/plugin.c:plugin_init() Failed to init bap plugin
Απρ 22 02:48:42.275340 kitana kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Απρ 22 02:48:42.275531 kitana kernel: Bluetooth: BNEP filters: protocol multicast
Απρ 22 02:48:42.275634 kitana kernel: Bluetooth: BNEP socket layer initialized
Απρ 22 02:48:42.275949 kitana dbus-daemon[312]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.13' (uid=0 pid=2081 comm="/usr/lib/bluetooth/bluetoothd")
Απρ 22 02:48:42.276795 kitana bluetoothd[2081]: Bluetooth management interface 1.22 initialized
Απρ 22 02:48:42.285378 kitana kernel: Bluetooth: hci0: AOSP extensions version v1.00
Απρ 22 02:48:42.285556 kitana kernel: Bluetooth: hci0: AOSP quality report is supported
Απρ 22 02:48:42.285618 kitana kernel: Bluetooth: MGMT ver 1.22
Απρ 22 02:48:42.298724 kitana kernel: NET: Registered PF_ALG protocol family
Απρ 22 02:48:42.312821 kitana systemd[1]: Starting Hostname Service...
Απρ 22 02:48:42.368685 kitana kernel: mt7921u 2-2:1.3: WM Firmware Version: ____010000, Build Time: 20230302150956
Απρ 22 02:48:42.395365 kitana dbus-daemon[312]: [system] Successfully activated service 'org.freedesktop.hostname1'
Απρ 22 02:48:42.395618 kitana systemd[1]: Started Hostname Service.
Απρ 22 02:48:43.995371 kitana kernel: mt7921u 2-2:1.3 wlanap_CF953AX: renamed from wlan0
Απρ 22 02:48:44.028960 kitana systemd-networkd[238]: wlan0: Interface name change detected, renamed to wlanap_CF953AX.
Απρ 22 02:48:46.544483 kitana ModemManager[361]: <info>  [base-manager] couldn't check support for device '/sys/devices/pci0000:00/0000:00:15.0/usb2/2-2': not supported by any plugin
Απρ 22 02:48:48.992149 kitana systemd[1]: systemd-rfkill.service: Deactivated successfully.
Απρ 22 02:49:03.012952 kitana systemd[1]: Started Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
Απρ 22 02:49:03.568602 kitana systemd-networkd[238]: wlanap_CF953AX: Link UP
Απρ 22 02:49:03.569598 kitana hostapd[2114]: wlanap_CF953AX: interface state UNINITIALIZED->COUNTRY_UPDATE
Απρ 22 02:49:03.572020 kitana kernel: br0: port 2(wlanap_CF953AX) entered blocking state
Απρ 22 02:49:03.572263 kitana kernel: br0: port 2(wlanap_CF953AX) entered disabled state
Απρ 22 02:49:03.572729 kitana kernel: device wlanap_CF953AX entered promiscuous mode
Απρ 22 02:49:12.429326 kitana systemd[1]: systemd-hostnamed.service: Deactivated successfully.
Απρ 22 02:49:14.582570 kitana hostapd[2114]: wlanap_CF953AX: interface state COUNTRY_UPDATE->HT_SCAN
Απρ 22 02:49:14.608260 kitana systemd-networkd[238]: wlanap_CF953AX: Gained carrier
Απρ 22 02:49:14.609044 kitana kernel: IPv6: ADDRCONF(NETDEV_CHANGE): wlanap_CF953AX: link becomes ready
Απρ 22 02:49:14.609105 kitana kernel: br0: port 2(wlanap_CF953AX) entered blocking state
Απρ 22 02:49:14.609148 kitana kernel: br0: port 2(wlanap_CF953AX) entered forwarding state
[... 759 lines cut ...]
driver=nl80211
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
ssid=killing_technology
bssid=d0:37:45:96:50:51
wpa_passphrase=XXXXXXXXXXXX
hw_mode=g
channel=6
ieee80211n=1
ieee80211d=1
wmm_enabled=1
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
logger_syslog=-1
logger_syslog_level=1
logger_stdout=-1
logger_stdout_level=1
ctrl_interface=/var/run/hostapd_CF953AX
```

Main requirement is to be able to run in 4G and then in 5G AP.
When the box hangs, the cursor in terminal screen stops blinking and nothing works except hard shutdown using the box shutdown button.

morrownr, 2023-04-24:
I'm pondering whether the following patch is applicable to your problem:

https://patchwork.kernel.org/project/linux-wireless/patch/20230420192451.28814-1-kernel@mattwhitlock.name/

The author, @whitslack , is a user here.

sektorhybrid, 2023-04-24:
@morrownr recompiled the module in my NUC box with https://github.com/whitslack 's patch and I am up and running without issue in 4G, the culprit seems to be in that direction. I will leave it for a while to test its stability and report back after a while. 

If you need extra testing from my side with various configurations feel free to ask and thanks for your support.

morrownr, 2023-04-25:
> I will leave it as is to test its stability and report back after a while.

Interesting. I look forward to your report. @whitslack is the one who needs the report as he could add additional info to his patch if a version 2 is necessary. It will be interesting what Felix has to say about the patch.

I was unable to duplicate the problem when @whitslack brought it up. I am of the opinion that there are differences in how I set up AP mode and that is the reason I am not seeing the issue. If you can take a look and isolate the differences in your setup and mind, that could provide some clues. The guide I made and use is located:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md

sektorhybrid, 2023-04-25:
@morrownr  I am leaving here my current configuration notes until I read your instructions page and compare them with my setup. There might be other configs that might be missing from my side, I 'll report them after my reading session. It is indeed interesting, and from my side an opportunity to learn a few things more.

```
[kitana hostapd]# ll /etc/systemd/network/
total 24K
drwxr-xr-x 3 root root 4,0K Απρ  12 00:30 .
drwxr-xr-x 6 root root 4,0K Απρ   8 21:01 ..
-rw-r--r-- 1 root root   30 Απρ  10 00:51 10-br0.netdev
-rw-r--r-- 1 root root  125 Απρ  10 01:57 13-eth0-bridged.network
-rw-r--r-- 1 root root  271 Απρ  10 02:16 15-br0.network

##################################### [/etc/systemd/network/10-br0.netdev]
[NetDev]
Name=br0
Kind=bridge

##################################### [/etc/systemd/network/13-eth0-bridged.network]
[Match]
Name=eth0
[Link]
RequiredForOnline=no
[Network]
Bridge=br0

##################################### [/etc/systemd/network/15-br0.network]
[Match]
Name=br0
[Network]
Address=192.168.1.10/24
Gateway=192.168.1.1
DNS=1.1.1.1
LinkLocalAddressing=ipv4
[Route]
Gateway=192.168.1.1
Destination=192.168.1.0/24
GatewayOnlink=yes

##################################### [/etc/hostapd/hostapd_wlanap_CF953AX.conf]
country_code=FM
interface=wlanap_CF953AX
bridge=br0
driver=nl80211
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
ssid=killing_technology_4G
bssid=XX:XX:XX:XX:XX:XX
wpa_passphrase=XXXXXXXXXXXXXXXXX
hw_mode=g
channel=6
ieee80211n=1
ieee80211d=1
wmm_enabled=1
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
logger_syslog=-1
logger_syslog_level=1
logger_stdout=-1
logger_stdout_level=1
ctrl_interface=/var/run/hostapd_CF953AX

##################################### [/usr/lib/systemd/system/hostapd@.service]
[Unit]
Description=Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
After=network.target
[Service]
ExecStart=/usr/bin/hostapd /etc/hostapd/hostapd_%i.conf
ExecReload=/bin/kill -HUP $MAINPID
Restart=always
RestartSec=4
[Install]
WantedBy=multi-user.target
```

Everything is still up and running without issue and seems stable. I have a minidlna server in an Rpi2, the first graph's STA is my current STA's dbms (one wall between CF953AX AP and 00:51:ed:00:16:82, the rest of STAs are closer to AP). The other two graphs show utilization from a yesterday's session, minidlna streaming 1080p movie through AP to the STA for a couple of hours. 

[image]

morrownr, 2023-04-25:
> I am leaving here my current configuration notes until I read your instructions page and compare them with my setup.

That is fine. I need to do a fresh round of testing on my guide soon since there was a recent RasPiOS release. I can look at this while doing the new setup.

I suspect the issue may involve my use of systemd-network. My guide is the result of long term testing to get the most stable AP setup possible. I don't care what it is based on as long as it is ultra stable  and fast.

Hopefully the patch is what was needed and can proceed on into the kernel.