morrownr/USB-WiFi #256: Getting worse performance with in-kernel rtw_8822bu driver

sjtmbg, 2023-04-26:
I've been manually compiling and using the 88x2bu drivers from github for about a year now and getting the max dl speeds that I can with my ISP (~200 Mbps).  I recently upgraded to xubuntu 23.04 with the 6.2.0 kernel support for the rtw_8822bu driver and am now getting dl speeds of about 25 Mbps when using the in-kernel driver.

For the comparison I'm using same computer, same USB port (it's a 2.0 but that shouldn't matter in side-by-side comparison, should it?) and same network configuration.  The desktop is a Lenovo M91p (4518W7Y) - I can provide other specs upon request.

Anyone else seeing similar performance issues using realtek/rtw88/rtw88_8822bu.ko ?

sjtmbg, 2023-04-26:
Excellent info, thank you so much!  I will muddle through with the 6.2 kernel then and look forward to 6.3 and 6.4.

Regarding the 88x2bu-20210702 out-of-kernel driver - I had been using that up until it wouldn't compile for me starting with 5.19 kernels on arch, fedora, and xubuntu.  That's when I switched over to RinCat's.  

The compile error I get when I try now, on 6.2.0, is "implicit declaration of function 'prandom_u32' in os_dep/osdep_service.c.

Please don't make any big effort to correct that, it has been overtaken by more recent developments, so it's just an FYI.

morrownr, 2023-05-12:
> The compile error I get when I try now, on 6.2.0, is "implicit declaration of function 'prandom_u32' in os_dep/osdep_service.c.

Remember, work on the out-of-kernel drivers is continuous because of the development model of Linux. I can't update the code sitting on your hardware but I have made it easy for users to update:

$ git pull
$ sudo sh install-driver.sh

There is no need to remove the old driver as my script, install-driver.sh, detects the old installation and removes it prior to the new installation.

ANYTIME you are going to move to a new kernel such as you did in your reply, you need to run the two commands above before you install the new kernel.

Cheers