morrownr/USB-WiFi #263: News: mt7921au based adapters now plug and play on newly released Raspberry Pi OS (2023-05-03)

morrownr, 2023-05-06:
A new release of the RasPiOS showed up on 2023-05-03 so I decided to burn a copy and test.

I used the 64 bit version and I run headless. 

 Result: My mt7921au based adapter worked immediately. No problems noted. I did not have to do anything.

This new release uses the 6.1 kernel and the mt7921 firmware was up to date.

I decided to test my AP guide (see Main Menu item 9). The guide worked well . There is only one thing that I had to update for WiFi 6 to work:

hostap

For anyone that likes to build AP's, WiFi 6 only works with hostapd v2.10 or later. RasPiOS still has version 2.9 for now. I have added two guides/examples to help those that want add WiFi 6 support to your homemade AP:

Under Main Menu item 9, I have added two items:

- Example - hostapd.conf for WiFi6 with mt7921au chipset based adapters
- Upgrade hostapd to support WiFi 6

The hostapd.conf example is considered a beta. It is really stable here and maybe it should be considered rc1. Try it and let me know what needs to be fixed/improved. The guide to compiling hostapd for WiFi 6 works for me but please suggest improvements if you see any.

FYI: I have been testing other distros to see which are now plug and play for use with the mt7921au based adapters: Results so far:

Ubuntu 23.04 and all of the other distros based on Ubuntu 23.04
Debian 12 weekly. It should be released June 10.
RasPiOS 2023-05-03

If you have other distros that should be added to the list, please reply.

Enjoy.

@morrownr

fhteagle, 2023-05-07:
By "update", did you mean compile from git source from w1.fi? Or did you use a packaged version of the hostapd binary from somewhere? 

I have found compilation from w1.fi source to be very easy, no wrinkles in the rug on x86_64 or armv7l environments.

morrownr, 2023-05-07:
> By "update", did you mean compile from git source from w1.fi? 

Yes

Here is the direct link to the guide. If you see anything that can be improved, let me know.

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Upgrade_hostapd.md

> I have found compilation from w1.fi source to be very easy, no wrinkles in the rug on x86_64 or armv7l environments.

Same here.

dellagd, 2023-05-09:
I was able to successfully set up a 6GHz AP using this USB adapter on an Ubuntu 22.04 VM following your [hostapd upgrade guide](https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Upgrade_hostapd.md) and the sections of the [bridged AP guide](https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md) that were not raspi specific. I made a couple small modifications:
* Upgrade 22.04 to the latest non-RC mainline kernel (https://kernel.ubuntu.com/~kernel-ppa/mainline/v6.3.1/)
* Add `op_class` to the [wifi6e conf example](https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi6e.conf) you have here. I specified `op_class=133` for the 80MHz channel I was trying to use. Without doing this, I was given a `could not determine operating frequency` error from hostapd when trying to set a 6GHz channel.
* Update the installed firmware files to the latest available from MediaTek

With this, I was able to connect from an OpenWRT client running a MT7922 PCIe M.2 module without issues (apart from VM-induced ones).

morrownr, 2023-05-11:
@dellagd 

That is great. Is it time for me to post a testing version of `hostapd-WiFi6e.conf` ?

Would you mind posting the entire .conf ?

> Without doing this, I was given a could not determine operating frequency error from hostapd when trying to set a 6GHz channel. I used this patch as reference for selecting what class I needed.

Interesting.

> apart from VM-induced ones

vm induced wifi issues? Surely not. (sarcasm)

Great post.