morrownr/USB-WiFi #268: Sale: Now through Sunday, 2023-05-14... ALFA WiFi Adapters...

morrownr, 2023-05-12:
I am on the Rokland mailing list and this showed up this morning.

Coupons:

https://us2.campaign-archive.com/?e=fff790ce9f&u=d4c872c7ace9d07bc55457321&id=2c3409c08e

ALFA AWUS036AXML:

https://store.rokland.com/products/alfa-awus036axml-802-11ax-wifi-6-1800-mbps-tri-band-wifi-usb-adapter-w-bluetooth?mc_cid=2c3409c08e&mc_eid=fff790ce9f

ALFA ARS-WiFi6E-M2 2.4 GHz + 5 GHz + 6E Tri-Band Omnidirectional Indoor Antenna:

https://store.rokland.com/products/alfa-ars-wifi6e-m2-2-4ghz-5-ghz-6e-tri-band-omnidirectional-indoor-antenna

FWIW: Rokland is one of the very few retailers of WiFi products that does a really good job with Linux support. 

They carry several WiFi adapters that are Plug and Play because of in-kernel drivers on Linux:

$27 - ALFA AWUS036NHA (N150, 2.4 GHz) Atheros 9271 chipset 

$40 - ALFA AWUS036ACHM (AC600, dual band, long range) Mediatek 7610u chipset

$22 - ALFA AWUS036ACU (AC1200, dual band) Realtek 8812bu chipset, as of kernel 6.2, the driver is in-kernel, best performance with kernel 6.3 or later, out-of-kernel driver also available

$43 - ALFA AWUS036ACM (AC1200, dual band) Mediatek 7612u chipset

$79 - ALFA AWUS036AXML (AXE3000, tri-band) Mediatek 7921au chipset) as of kernel 5.19, the driver is in-kernel,

Note: This is not a paid advertisement. I am simply a satisfied customer that is forwarding information that might be useful to those that are users of this site. If you see a good sale at a reputable dealer, let me know.

ALFA has dealers around the world. If Rokland does not ship to your location, contact ALFA for information about the closest ALFA:

https://www.alfa.com.tw/

kflmiami420, 2023-06-13:
$79 - ALFA AWUS036AXML (AXE3000, tri-band) Mediatek 7921au chipset) as of kernel 5.19, the driver is in-kernel,   Its does not seem to work on a fresh install of kali linux on a RPI4 8gig .   it shows up when the lsusb command is used but thats it .   Drivers have been installed again per the instructions --- MT7921 - mt7921au, mt7921, and mt7921k (AMD RZ608) chipsets

To install or update the firmware:

Go to the following site:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek

Click on WIFI_MT7961_patch_mcu_1_2_hdr.bin

Click on plain

Save file

Click on WIFI_RAM_CODE_MT7961_1.bin

Click on plain

Save file

If your adapter/card has Bluetooth support:

Click on BT_RAM_CODE_MT7961_1_2_hdr.bin

Click on plain

Save file

Create the needed directory (if necessary):

$ sudo mkdir /lib/firmware/mediatek
Copy the files to the following locations:

$ sudo cp WIFI_MT7961_patch_mcu_1_2_hdr.bin /lib/firmware/mediatek
$ sudo cp WIFI_RAM_CODE_MT7961_1.bin /lib/firmware/mediatek
$ sudo cp BT_RAM_CODE_MT7961_1_2_hdr.bin /lib/firmware/mediatek

When running Ifconfig or iwconfig the wifi adapter does not show up

morrownr, 2023-06-14:
> When running Ifconfig or iwconfig the wifi adapter does not show up

Recommend you discontinue use of ifconfig and iwconfig as they have been depreciated for several years and are no longer maintained. Please use ip and iw.

> fresh install of kali linux on a RPI4 8gig

What kernel is in use? $ uname -r

> Drivers have been installed again per the instructions --- MT7921 - mt7921au, mt7921, and mt7921k (AMD RZ608) chipsets

That instruction is for the firmware. The kernel has to have the driver for the adapter to work.

Firmware is handled by the distros maintainers or you. The driver is handled by Linus (the kernel maintainers).

With in-kernel drivers, there are 2 parts of the driver. The firmware and the part that is normally called the driver. You must have both for the driver to work. There are still supported distros out there that are using kernels that do not have kernels that have the in-kernel driver yet. This is likely the problem in this case.

Main Menu item 2, Plug and Play list..., has details about the mt7921au chipset not far from the top of the document. You need at least kernel 5.18. Examples of distros that are Plug and Play in this case:

Debian 12
Ubuntu 22.10 and later
RasPiOS 05-03-2023

bjlockie, 2023-06-14:
What is the output from lsusb?