morrownr/USB-WiFi #269: Kernel problems with mt7921au on OpenWrt 22.03.5

jclark, 2023-05-13:
Since #244 was closed, I'm making this into its own issue.
 
I tried a  Comfast CF-953AX as an AP on OpenWrt 22.03.5 (on an x86 machine), and it wasn't reliable for me.  I used the wpad-openssl package, and didn't change any defaults other than to enable WPA2-PSK. Running a speed test from an iPhone 14 hung almost immediately with some kernel messages:

```
Thu May  4 05:39:03 2023 kern.err kernel: [  450.578724] mt7921u 2-2:1.3: Message 00020002 (seq 10) timeout
Thu May  4 05:39:03 2023 kern.err kernel: [  450.808726] mt7921u 2-2:1.3: timed out waiting for pending tx
Thu May  4 05:39:03 2023 kern.warn kernel: [  450.837648] ------------[ cut here ]------------
Thu May  4 05:39:03 2023 kern.warn kernel: [  450.846659] WARNING: CPU: 3 PID: 38 at 0xffffffff810d8ad5
Thu May  4 05:39:03 2023 kern.warn kernel: [  450.856529] Modules linked in: pppoe ppp_async option nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet usb_wwan qmi_wwan pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_
reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt792
1u mt7921_common mt76x2u mt76x2e mt76x2_common mt76x02_usb mt76x02_lib mt7603e mt76_usb mt76_connac_lib mt76 mac80211 cfg80211 usbserial usbnet slhc r8169 nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag
_ipv6 nf_defrag_ipv4 libcrc32c igc forcedeth e1000e crc_ccitt compat cdc_wdm bnx2 i2c_dev ixgbe e1000 amd_xgbe mdio nls_utf8 ena sha256_ssse3 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac nls_iso8859_1 nls_cp437 vfat fat igb
 button_hotplug tg3 ptp realtek pps_core mii
Thu May  4 05:39:03 2023 kern.warn kernel: [  450.978211] CPU: 3 PID: 38 Comm: kworker/u8:3 Not tainted 5.10.176 #0
Thu May  4 05:39:03 2023 kern.warn kernel: [  450.986818] Hardware name: Default string Default string/Default string, BIOS 5.13 09/13/2022
Thu May  4 05:39:03 2023 kern.warn kernel: [  450.997567] Workqueue: mt76 0xffffffffa0324560 [mt7921_common@00000000d115bbdf+0xc000]
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.007554] RIP: 0010:0xffffffff810d8ad5
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.013220] Code: 32 31 c0 5b 41 5c 5d c3 cc cc cc cc 0f 0b 49 8b 9c 24 20 05 00 00 a8 04 74 ab 0f 0b b8 da ff ff ff 5b 41 5c 5d c3 cc cc cc cc <0f> 0b b8 f0 ff ff ff eb d0 0f 0b eb cc 66 66 2e 0f
 1f 84 00 00 00
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.036016] RSP: 0018:ffffc9000014fd88 EFLAGS: 00010202
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.043003] RAX: 0000000000000004 RBX: ffff8881014e4d00 RCX: 0000000000000002
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.052048] RDX: ffffffff824f0748 RSI: 0000000000000297 RDI: ffff88810156cf80
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.061089] RBP: ffffc9000014fd98 R08: 0000000000000400 R09: 0000000000000000
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.070046] R10: 0000000000000001 R11: 0000000000000000 R12: ffff88810156cf80
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.079157] R13: ffffc9000014fdb0 R14: 0000000000000100 R15: ffff888100a99f98
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.088213] FS:  0000000000000000(0000) GS:ffff888277d80000(0000) knlGS:0000000000000000
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.098283] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.105892] CR2: 00007fb4199c2000 CR3: 000000000340a000 CR4: 0000000000350ee0
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.115009] Call Trace:
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.119166]  0xffffffffa02f3436 [mt76_usb@00000000f6400edc+0x5000]
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.127262]  ? 0xffffffff810f60c0
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.132264]  0xffffffffa0330e99 [mt7921u@00000000d4dc2d3c+0x4000]
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.140198]  0xffffffffa03245eb [mt7921_common@00000000d115bbdf+0xc000]
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.148603]  0xffffffff810d2644
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.153448]  0xffffffff810d27ee
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.158203]  0xffffffff810d8403
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.162989]  ? 0xffffffff810d27a0
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.167881]  ? 0xffffffff810d82e0
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.172805]  0xffffffff81003c1f
Thu May  4 05:39:03 2023 kern.warn kernel: [  451.177505] ---[ end trace 2b75a9992dca1424 ]---
Thu May  4 05:39:03 2023 kern.info kernel: [  451.322002] mt7921u 2-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20220311230842a
Thu May  4 05:39:03 2023 kern.info kernel: [  451.322002]
Thu May  4 05:39:03 2023 kern.info kernel: [  451.351824] mt7921u 2-2:1.3: WM Firmware Version: ____010000, Build Time: 20220311230931
Thu May  4 05:39:10 2023 kern.err kernel: [  458.178722] mt7921u 2-2:1.3: Message 00020003 (seq 9) timeout
Thu May  4 05:39:13 2023 kern.err kernel: [  461.218721] mt7921u 2-2:1.3: Message 00020003 (seq 10) timeout
Thu May  4 05:39:13 2023 kern.err kernel: [  461.448722] mt7921u 2-2:1.3: timed out waiting for pending tx
Thu May  4 05:39:14 2023 kern.info kernel: [  461.613176] 7mt7921u 2-2:1.3: HW/SW Version: 0x8a108a10, Build Time: 20220311230842a
Thu May  4 05:39:14 2023 kern.info kernel: [  461.613176]
Thu May  4 05:39:14 2023 kern.info kernel: [  461.643854] mt7921u 2-2:1.3: WM Firmware Version: ____010000, Build Time: 20220311230931
```

I switched over to a MT7612u (a TEROW ROW02CD) and everything seemed to work fine in the brief test I did.

_Originally posted by @jclark in https://github.com/morrownr/USB-WiFi/issues/244#issuecomment-1534154689_

jclark, 2023-05-13:
I investigated this further. I installed OpenWrt 22.03.5 on an ARM box (a NanoPi R4S), used the same wifi card and got the exact same problem. It seems to happen only at high speed (I'm getting 400Mbps or more).

My wifi-config was:

```
config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/usb@fe800000/fe800000.usb/xhci-hcd.0.auto/usb6/6-1/6-1:1.3'
        option channel '149'
        option band '5g'
        option htmode 'HE80'
        option cell_density '0'
        option country 'TH'
```

I then tried the suggestion from @morrownr in https://github.com/morrownr/USB-WiFi/issues/244#issuecomment-1535206511 and updated there firmware.  The description there is out of date, because in 22.03.5 the firmware two files are installed in /lib/firmware/mediatek, but they were different from the ones in https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek .  Updating to the ones in git seems to have solved the problem. I haven't tested very much, but previously I was getting a kernel problem almost immediately when running at high speed.

morrownr, 2023-05-13:
> The description there is out of date, because in 22.03.5 the firmware two files are installed in /lib/firmware/mediatek, but they were different from the ones in https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek .

I went to correct the "out of date" problem but I can't find anything that is out of date. Can you be more specific?

jclark, 2023-05-13:
I believe this is no longer accurate as of 22.03.5:

> Note: As of OpenWRT 22.03.2, the mt7921au firmware is still included in the mt7921e driver file instead of a common package. Therefore, loading the mt7921u driver does not load the firmware. 

The two firmware files are now in kmod-mt7921-firmware, which gets installed when you install kmod-mt7921u.

jclark, 2023-05-13:
I guess I or somebody should file a bug report with OpenWrt to get kmod-mt7921-firmware updated with the right firmware to work with the driver they ship in kmod-mt7921u. Not sure where though.

morrownr, 2023-05-13:
> I believe this is no longer accurate as of 22.03.5:
>
>
> The two firmware files are now in kmod-mt7921-firmware, which gets installed when you install kmod-mt7921u.

Ah, yes. thank you. I have made changes. Do you mind taking a look to see if the changes make sense?

morrownr, 2023-05-13:
> I guess I or somebody should file a bug report with OpenWrt to get kmod-mt7921-firmware updated with the right firmware to work with the driver they ship in kmod-mt7921u. Not sure where though.

https://github.com/openwrt/mt76

Before you file a bug, let me explain.

The mt76 repo at OpenWRT is a downstream site for the mt76 in Linus' kernel. It is maintained by the Mediatek kernel devs just as they handle the mt76 driver in Linus' kernel. Felix regularly updates the OpenWRT mt76. He knows what firmware has come in and handles it as needed. What you will find is that the 2 mt76 repos do not stay in lockstep with each other at all times and sometimes a patch may need testing before being submitted to Linus' kernel. When it is time, patches are pulled both directions, upstream and down. 

Mediatek takes their Linux support very seriously.

The official way to report bugs and provide patches is via message to linux-wireless but let me tell you, you have to read up on how to do it and it is best to subscribe first and watch for a while so you learn how things are done. Don't wander in and say "my adapter doesn't work." That will be ignored. All kernel wireless devs use that list and they are very busy. In other words, no time for BS. I can provide links if you want to investigate further. I monitor linix-wireless so as to provide info of interest here at USB-WiFi.

Cheers

fakemanhk, 2023-05-15:
I just bought the CF-953AX (since the system detects no extra stuff so I believe it's single state), connected to my NanoPi R4S (OpenWrt 22.03.3), with kmod-mt7921u I can see it being detected (the USB ID is 0e8d:7961 so I don't need to change), `iw phy0 info` was able to read it as well, but the WiFi detect seems to have problem (it couldn't see my device).

On the other hand I was wondering if Chrome OS has a linux kernel which is capable to use this USB WiFi and the result is negative at the moment.

jclark, 2023-05-15:
@fakemanhk Are you using the FriendlyWrt version of OpenWrt or the official OpenWrt from openwrt.org (which 23.03.5)? I am using the official OpenWrt version.

I tried the FriendlyWrt version on a R5C, and installing kmod packages doesn't work, because they are using the official repo but the kernel version of kmod packages in the official repo doesn't match the kernel version of their OpenWrt build (which is super annoying).

morrownr, 2023-05-15:
MT7912u first entered kernel 5.18 and OpenWRT uses kernel 5.10. The OpenWRT Mediatek devs backported the driver to kernel 5.10. I suspect what has happened is that FeiendlyWRT does not know this.

> On the other hand I was wondering if Chrome OS has a linux kernel which is capable to use this USB WiFi and the result is negative at the moment.

Google is not the friendly toward usb wifi support. I converted my Chromebook to run regular Linux distros which fixed this problem.

fakemanhk, 2023-05-15:
No I don't use FriendlyWrt, I use vanilla OpenWrt on my R4S (I want
something really FLOSS).

jclark, 2023-05-16:
@morrownr The problem is this: https://github.com/friendlyarm/Actions-FriendlyWrt/issues/19

jclark, 2023-05-16:
@fakemanhk Have you tried 22.03.5? That's the only difference I can see between our two setups.

Are you connecting the CF-953AX directly or through a cable/hub? I had a problem when I tried to use an unpowered hub.