morrownr/USB-WiFi #271: Feature Request -- Make it easier to find adapters suitable for APs

owendelong, 2023-05-17:
Right now, it takes some digging in each adapter to find the Valid Combinations and then parse it for maximum AP and Maximum Total (the minimum of these two numbers is the maximum SSIDs that can be simultaneously supported on an AP AIUI).

It would be really nice if there could be a column in the various lists for Maximum AP SSIDs that contained this value pre-computed so one could scan the list for that information.

I'm willing to write a parser to go through the files in the iw list and produce the values if that's helpful, but I'd want to know how to make the output useful before going to the effort.

Thanks for this fantastic resource.

owendelong, 2023-05-18:
Yes, that's exactly what I'm talking about.

I didn't realize this was a driver thing. Presumably each chipset also has a hardware limit for this, right?

How do I go about finding the hardware (if any) limit for the maximum number of simultaneous SSIDs as an AP?
(sorry if it's inappropriate to ask that question in this context), I'm more of a user than a developer in this arena.

Techwolf, 2023-08-01:
I have one PCIx card that has:

```
valid interface combinations:
                 * #{ managed } <= 2048, #{ AP } <= 8, #{ P2P-client, P2P-GO } <= 1,
                   total <= 2048, #channels <= 1, STA/AP BI must match
```

Would love to find an USB one that can handle more then one managed client. I sometimes create upto 10 virtual MAC connections on one AP. Note that is is different then virtual IPs. The AP think 10 clients are connected to it with different MAC addresses, but it really is one device that is connected.