morrownr/USB-WiFi #273: rtl8832cu adapters

aquilarubra, 2023-05-24:
Hi, there are new adapters based on the new rlt8832cu chip. They are tri band AX3000 6e (2.4/5/6Ghz) and cost around $30.

Realtek just furnished the driver, which is same as rtw8852cu:
https://github.com/lwfinger/rtw8852cu

I created packages for Arch Linux:

https://aur.archlinux.org/packages/rtl8852cu-git
https://aur.archlinux.org/packages/rtl8852cu-dkms-git

You may wish to add this to your guide. I think they are a good alternative to the Comfast CF-953AX. So far, it was tested by two of us. I am getting a 6Ghz router, so I will test that band too.

morrownr, 2023-05-25:
Hi @aquilarubra 

Thank you for the information. This site would not be what it is without users posting information to add to what we have.

> You may wish to add this to your guide.

I assume you are talking about the Plug and Play list? USB-WiFi main menu item 2?

I'll assume that the Plug and Play list is the guide for the purposes of explaining things. This may be a longer explanation than you are looking for but I think I need to take the time and explain.

The Plug and Play list does not contain any adapters that are only supported with out-of-kernel drivers. There are adapters have out-of-kernel drivers available in the list. There are adapters with the  rtl88x2bu and rtl8821cu chipsets in the list now that there is also an in-kernel driver available for both.

So, the question is "why does he do that?" 

Answer: I've been using Linux since about 1994. I keep up with things and have a reasonable understanding of the development model of Linux and how it came about. The development model of Linux is very different than Windows or OSX or other closed source operating systems. Closed source tends to release a new version (meaning the API's are basically locked) and that version sticks around for a few years or in some cases, for a lot of years. Linux is not like that A new version of Linux is released every 2.5 month or so. The API's are locked for each version, mostly, but can change with when a new version is released. This model is a result of the resources that are available. Linux developers are not dedicated Linux employees... well, maybe Linus and a few others at the top of the food chain are full time employees of the Foundation but most of the developers are Red Hat, Intel, Mediatek, AMD or other large company employees or hobby programmers or programmers are smaller companies that take care of speicifc things. Linux is a worldwide effort. Given where the work comes from and the imability to set solid deadlines, the current Linux development model evolved and it seems to make the most out of the resources that are available.

Once you understand the Linux devlopment model, you will have a better understanding of why out-of-kernel drivers are problematic. Out-of-kernel drivers sitting on your harddriver do not get automatically updated. When things blow up is when I get calls for help with the 5 out-of-kernel drivers that I maintain here. On the other hand, Mediatek went to a fully in-kernel method as called for by Linux Wireless Standards. When API's are changed or added during the 2.5 month development cycle, the are several Mediatek employees that work on the Mediatek drivers to make sure everything works in the next release of Linux. Most users have not idea of the work that is required, all they know is that when the distro sends an updated kernel into their installation, the drivers still work.

Realtek tries to use a Windows dev model on their Linux USB WiFi drivers and it is problematic. Most of the work I do on the 5 out-of-kernel Linux drivers I maintain here is updating the drivers to make them continue to work on new kernels. Realtek doesn't do it. In fact, Realtek has no public location where we can download their drivers and they sure as hell have no way for users like you and I report bugs. On the other hand, Mediatek is open. Here is the link to their kernel.org website:

https://wireless.wiki.kernel.org/en/users/drivers/mediatek

Notice the email addresses of some of the Mediatek devs.

You can also subscribe to "linux-wireless" to watch the work as it happens.

Mediatek does not have a company site to download their drivers...here is the site:

https://github.com/torvalds/linux

That is the primary Linux repo. That is where the drivers are and that is where you can look at the code. You can submit PATCHes for problems you find or to add capabilities. There are users here that eariler this year submitted patch to submit the VID/PIDs for new adapters that uses company ID's.

I don't plan to add any adapters where the only driver available is an out-of-kernel driver because it is not in the best interest of Linux users. I hope you understand and I'm here for additional questions.

@morrownr

aquilarubra, 2023-05-26:
Hi, thank you for your explanation.

I was thinking more about point 5. Anyway, I did my job signaling compatibility of the rtl8832cu chip. I also created packages for Arch.

I am 25 years on Linux now and I also run my own distro. I understand there are currents in the Linux world that tend to be biased in one way or another. Truly, I am not interested in that.
I just see that with Realtek you can ask and they give you. Kernel compatibility maintenance is really trivial, changes are mostly minimal.
Realtek does not spend human resources to work in kernel. So what? Linux is based on volunteers. You cannot pretend every hardware company to have personnel dedicated to Linux.
For me, 2 criteria are important. Driver availability (in kernel or not) and price.
With the rtl8832cu I can buy 4 adapter at the price of a Mediatek adapter. Excluding that one that can be made to work on Linux on 6Ghz, but won't on Windows (yes, most Linux users use dual boot systems), and that other model you cannot even find on the market. I can buy an rtl8832cu adapter now, and what will come next in next years for 4 times at the same price, without an upfront payment. I will have always a new adapter instead of keeping the old Mediatek, which at a certain point could fail at any time. Not to mention antenna signal degradation.

There is room for everybody and every current of thought. Just take or disregard my post, as your liking.

closed 2023-05-26 by aquilarubra