morrownr/USB-WiFi #274: Updated information on Alfa Network AWUS036AXML

patrakov, 2023-05-30:
I have bought this adapter, and I am not happy.

Here is the text of the review I sent to Amazon, which is still pending approval.

---

This WiFi adapter comes with a long USB-C cable and accessories that help mount it in a spot with a good signal - e.g., high on the wall. The experience is plug-and-play on Linux, but a driver (https://info.alfa.com.tw/awus036axmldriver) needs to be installed on Windows. Windows 11 complains that not many people run the driver installer and that it could be malicious, which is, of course, false.

Compared to Alfa Network AWUS036ACM, this WiFi adapter has a better download speed but a much worse download speed, especially under Windows. The reason is that it, while being sensitive, sends very weak radio waves, and the access point cannot hear it well. The access point is reporting a significant diminishment of the signal levels as compared to AWUS036ACM, with a difference of 11 dBm on Linux and a whopping 18 dBm on Windows.

The issue has been reported as https://github.com/openwrt/mt76/issues/783 for the Linux driver.

I strongly recommend refraining from purchasing this adapter due to the serious signal issues mentioned above. For all practical purposes, it is a defective product.

bjlockie, 2023-05-30:
Typo?

On 2023-05-29 20:52, Alexander E. Patrakov wrote:
> Compared to Alfa Network AWUS036ACM, this WiFi adapter has a better 
> download speed but a much worse download speed,

Is the weak signal due to a low power problem?
What happens if you turn off power save in the Linux driver?

patrakov, 2023-05-30:
Yes, a typo - fixed, and fixed on Amazon, too. Thanks!

Regarding the second question, I think yes - we have both the low transmit power indicated by `iw` on the tablet, the low received signal indicated by the AP in its web interface (which is the same thing minus the propagation loss), and low upload speed.

Regarding the power-saving mode, I have tried turning it off. Unfortunately, due to the WiFi adapter antennas not being exactly in the same position as yesterday, it does not make sense to compare the results directly. However, to the first approximation, it makes no difference.

With the AP in AC mode:

The AP reports -77 dBm received from the WiFi adapter.

With power saving (i.e. the default settings): 363 Mbits/s down, 132 Mbits/s up (from the viewpoint of the tablet)

Without power saving (`iw dev wlp0s2f0u2i3 set power_save off`): 353 Mbits/s down (good!), 146 Mbits/s up (still worse than what AWUS036ACM can achieve with the default settings).

With the AP in AX mode:

The AP reports -78 dBm received from the WiFi adapter.

With power saving (i.e. the default settings): 373 Mbits/s down, 142 Mbits/s up (from the viewpoint of the tablet)

Without power saving: 351 Mbits/s down, 141 Mbits/s up (from the viewpoint of the tablet)

morrownr, 2023-05-30:
Hi @patrakov 

Can you provide the Amazon link that you used to purchase the adapter?

I'd like to see who the seller is. I've had somewhat better success by reporting problems to the seller first, thereby allowing them to correct the problem.

This is  something that needs to be investigated to see if it is a one-off manufacturing problem or if it is a general engineering problem. While I have the Alfa ACM, I do not have an AXML to test so I am wondering if you were to make a short step by step checklist of how you tested, others that do own the AXML could post their results so we can see if this is a general engineering problem.

Observation:

> The AP reports -77 dBm received from the WiFi adapter. Just a reminder, with AWUS036ACM in the same place, the AP reports -69 dBm.

Both of the results above are readings that are marginal. Well, the -69 is marginal and the -77 is a difficult situation. I always work to be get a reading of < -60 to have a stable connect and < -52 if I want a speedy, stable connection. Do you see the same weak tx situation if you move the adapter to where is have a better signal?

Overall, it would be nice if we could get some standardized testing from other users of the AXML. Problems happen. The question that needs answered is whether this is a one-off manufacturing problem or a general engineering problem. 

Regards

bjlockie, 2023-05-30:
If the -77db is bad but a different adapter has better signal then isn't that a problem? :-)

jopdyke, 2023-05-30:
We can work with ALFA and Mediatek to update the driver if that's needed. What brand WiFi router are you using, and do you broadcast separate SSIDs for each frequency or just a single SSID? Do you know what band your ACM and AXML are operating on in each of your tests?

What are the down/up speeds of your ACM in the same location/same router?

patrakov, 2023-05-31:
Regarding the "one antenna is faulty" theory, isn't this refuted by the excellent download speed?

patrakov, 2023-05-31:
The Amazon link is https://www.amazon.com/dp/B0BY8GMW32

Regarding the comparison of the signal levels in a less-difficult situation, that's a valid point, but I am 200% busy until Friday.

patrakov, 2023-05-31:
Regarding the -69 dBm reading on the AP, yes, I have deliberately tested from the worst place on the ground floor.

The WiFi router is a Linksys E8450 on the 2nd floor.

The step-by-step checklist is:

* Create a 5 GHz-only SSID on the router.
* Find a badly covered spot where you can place the adapter. Place a sticky note there.
* Put the AXML adapter there, then connect to a Lenovo Duet 7i tablet using the included cable. Edit the connection profile in NetworkManager to force the use of the AXML adapter. Disconnect and reconnect the connection without unplugging the adapter.
* Double-check that `iw reg get` gives the correct result, as well as `iw phy`.
* Run `iperf3 -s -p 9999` on the router.
* Run `iperf3 -c 192.168.10.1 -p 9999` where 192.168.10.1 is the router IP.
* Reboot into Windows without moving the adapter or the tablet, retest using the Windows binaries of iperf3.
* Remove the AXML adapter, use the included cable and a USB-C to USB-A adapter to place the ACM wireless card there.
* Retest both with Windows and with Linux.

patrakov, 2023-05-31:
> What are the down/up speeds of your ACM in the same location/same router?

see https://github.com/openwrt/mt76/issues/783#issuecomment-1567285862

morrownr, 2023-05-31:
I am going to address several messages in one here so as keep from adding numerous messages to this thread.

> Regarding the "one antenna is faulty" theory, isn't this refuted by the excellent download speed?

It very well could be that this is the case. I am not locked into any specific theory about what is causing this problem. I will likely need to do some heavy duty research as I am able to fully engage which will likely happen next week. Our ability to narrow down the list of causes could be very valuable to finding the solution.

> The Amazon link is https://www.amazon.com/dp/B0BY8GMW32

Okay, the seller is Top123Edge. I am not familiar with that seller but I will point out that you already have one reply above from an Alfa products reseller: @jopdyke 

FYI: It may not be obvious but this site is monitored by far more individuals than users. There are Mediatek devs that monitor the site. Alfa monitors the site as they have sent messages on a couple of occasions to ask me to correct information. Resellers monitor the site. The people needed to narrow down the cause are here but it may take time to get to the bottom of this issue.

> The step-by-step checklist is:

Thank you. This is needed so we are on the same sheet of music.

> Regarding the -69 dBm reading on the AP, yes, I have deliberately tested from the worst place on the ground floor.

I understand. It is sort of funny in that I may have to find a place well into my back yard to get a reading that bad but I will do so. It is good that you are finding this problem but I am curious as to the reason you were looking for a bad signal

Here is an example of how complicated the wifi and computer stuff can be. The below paste is from a patch submitted to linux-wireless this morning:

```
[PATCH 0/9] Support Wifi RFI interference mitigation feature

Due to electrical and mechanical constraints in certain platform designs there may
be likely interference of relatively high-powered harmonics of the (G-)DDR memory
clocks with local radio module frequency bands used by Wifi 6/6e/7. To mitigate
possible RFI interference producers can advertise the frequencies in use and
consumers can use this information to avoid using these frequencies for
sensitive features.

The whole patch set is based on 6.4-rc3. With some brief introductions as below:
Patch1:     Core ACPI interfaces needed to support WBRF feature.
Patch2 - 4: Enable WBRF support for some Mediatek and Qualcomm wifi drivers.
Patch5 - 9: Enable WBRF support for AMD graphics driver.
```

This is interesting in that it points out the various RFI issues involved and it reminds me of last year when mt7921 based adapters started shipping, We knew that the mt7921, mt7921k and mt7921au chipsets had Bluetooth (BT) capability but when the first mt7921au chipset based adapters shipped, BT was not working which created a lot of questions. At some point I remembered an old paper by Intel that is 10+ year old that described an unforeseen problem that USB3 connectors and cables emit RFI in the 2.4 GHz range that BT uses. This caused problems. After finding the document and doing some research, it was clear why none of the USB WiFi adapters makers over the last several years can exceed USB2 if they support BT. An example is the rtl8822bu and the rtl8812bu. The rtl8812bu chipset is single-function...wifi only and it supports USB3. The rtl8822bu uses the same wifi chipset but also supports BT so the wifi has to be limited to USB2 which is a speed limiter on AC1200 and later adapters. We know why the makers of USB WiFi adapters that use the mt7921au chipset are not turning on the BT capability... because it would seriously limit wifi performance. This is one issue of many that has to be handled by adapter makers. Supporting new chipsets and the drivers they require is a challenge. Are we going to find issues as the new WiFi 6 USB adapters come to market? Yes. As the quote from Dorthy goes, Toto, we are not in Kansas anymore. WiFi was MUCH more simply back in the WiFi 4 and earlier days. I have been looking at the WiFi 6 driver code for some time and the complexity is astounding. I have been heard to say "I don't know which is more complicated, Rocket Science or Wifi 6 drivers. Remeber that those working on the WiFi 6 driver are humans. It will take time to get everything sorted out. I actually think Mediatek has done a really good job with their WiFi 6 drivers but you can tell it has taken a toll on them as they are slowing adding features as time passes. The orginal release of the usb driver only supported managed and monitor modes. Latest AP and AP/VLAN support was added. P2P just went into kernel 6.4. When will mesh point happen? I don't know but Mediatek keeps kicking the can down the road and now AMD appears to be rolling into help. 

Something interesting I noticed yesterday: A series of patches was submitted to linux-wireless from a person that was not a usual suspect to submit a long series of patches for Mediatek. We regularly see single patches from individuals but patch series are usually from Mediatek devs such as Felix, Lorenzo or all of the devs with email addresss that end with mediatek.com. In this case, the email address ended with amd.com and a light bulb came on for me. It appears that AMD has assigned devs to assist with Mediatek WiFi drivers. This is likely coming as a result of a big contract Mediatek and AMD signed last year for Mediatek to support wifi chipsets for AMD so that AMD is able to provide a more complete lineup of components to system makers. AMD has been a very good supporter of Linux for a long time and this relationship will likely help with adding features and fixing bugs with the mt7921 and mt7922 chipset drivers. Modern WiFi 6 capable driver are incredibly complex as I said above.

@morrownr

patrakov, 2023-05-31:
> I am curious as to the reason you were looking for a bad signal

I was researching whether I could recommend this adapter to a customer who has a T-shaped office (and thus expects problems) and wants to use a software router + software AP. Yes, I understand that a more correct test case would be as an AP, not as a client. Anyway, I went for a "worst case in my apartment" test. We ended up agreeing on 2x AWUS036ACM as a temporary solution that can be upgraded later.

morrownr, 2023-06-01:
> I was researching whether I could recommend this adapter to a customer who has a T-shaped office (and thus expects problems) and wants to use a software router + software AP. Yes, I understand that [...]

Ah, I see. I do things like this a lot so I can add info about specific adapters. I've used the ACM a lot in AP mode. It is a solid adapter in AP mode. Main Menu, item 9 is a guide to setting up an AP and it has the specific info required for a lot of adapters including the ACM. The guide uses a RasPi4B but with minor mods can work on most computers running Linux. I do have a Comfast CF-951AX (mt7921au chipset) that I test and use but it has an internal antenna and it probably would not meet your range needs

Are you going to install both ACM's on the same system? The reason I ask is that I usually recommend folks use 2 adapters that use different chipsets in they are making a dual band adapter. The Realtek out-of-kernel drives won't support two adapters with the same chipset in the same machine. I have lighty tested 2 adapters with mt7612u adapters in the same machine and it worked but my favorite setup on the Pi4B is an ACM for 5 GHz and a ACHM for 2.4 GHz. The ACHM is a Max Power model. It's range is impressive.

I have no experience with the AXML but hope to soon. I'll bet we can figure out what is going on. What I would like to see is a Max Power adapter with the mt7922 chipset that has dedicated optimized antennas for each band. That would be an AP monster. I'm writing this message with the system connected with a PCIe card using the mt7922 chipset. Distro is Ubuntu 23.04 with the 6.2 kernel. Plug and Play.

Cheers

morrownr, 2023-06-01:
Opps...

> The reason I ask is that I usually recommend folks use 2 adapters that use different chipsets in they are making a dual band adapter.

I meant to say ...I usually don't recommend...

morrownr, 2023-06-02:
@patrakov 

Depending on your system, you might be interested in #261 

I am testing this PCIe card because it contains the mt7922 chipset. I think we may see the usb version of this chipset at some point this year and maybe adapters early next year. I have only tested managed mode so far but can expedite AP mode testing if it is of interest to you.

jopdyke, 2023-07-03:
We have tested this product with the Linksys E8450 router (which is not WiFi 6E) and are not able to produce the upload speed problem or tx power problem. Your Amazon review did appear to publish and is currently the only written review on the product. It is 1 star (but notes it gets better download speed than another product in your test but a slower upload speed). If this is not your review, please let us know as this could indicate another person has experienced something similar. 

For further testing may you please get in touch with me through Rokland.com support so we can exchange the product for you and see if this problem is unit specific? We can attempt to reproduce testing on your returned unit.

patrakov, 2023-07-03:
Hello.

The review is mine. OK, I will contact you on that website.

patrakov, 2023-07-11:
Update: this is actually an issue on the AP side. The AP is a Linksys E8450 reflashed to OpenWrt. There is a discussion that the mere presence of the 160 MHz AC capability exposed by the driver leads to problems - and this is indeed the case.

I have tested this fix: https://forum.openwrt.org/t/802-11ax-worse-than-802-11ac-with-mt76-driver/126466/562

It makes the upload speed much better under Linux, but hurts even more under Windows.

Feel free to close this issue.

DirtyOptics, 2024-09-26:
Is it possible to force the AWUS036AXML into using USB2 drivers as opposed to XHCI/USB3 drivers? The only reason I ask is because above you talk about RFI issues with USB3 and I believe to be having this issue. Whenever I plug my AWUS036AXML in, its basically killing my local GPS feed... Of note also, the Linux drivers on my RaspiOS 64bit show that tx: 3db also. Perhaps I need to look at a few things here.

morrownr, 2024-09-26:
Hi @DirtyOptics 

> Is it possible to force the AWUS036AXML into using USB2 drivers ?

Yes:

- Plug the axml cable into a USB2 port.
- Use a short USB2 extension cable or USB2 right angle adapter.

Anything in the connection between the port and adapter that only supports USB2 will cause the adapter to drop back to USB2.

DirtyOptics, 2024-10-01:
> Hi @DirtyOptics
>
>
> Yes:
>
> * Plug the axml cable into a USB2 port.
> * Use a short USB2 extension cable or USB2 right angle adapter.
>
> Anything in the connection between the port and adapter that only [...]

haha I knew you were going to say that.....sorry. Silly question. Via software? I don't have USB2 ports available.
also no space for adapters inside the enclosure. I will look to block the additional pins.

morrownr, 2024-10-05:
> Via software?

Not that I am aware of. I think it is a reasonable capability but those running linux-wireless are of the opinion that the use of module parameters should be limited.