morrownr/USB-WiFi #275: Monitor mode for ALFA AWUS036AXML in Kali?

iangregsondev, 2023-06-03:
Hi,

Can anyone confirm that this new model (wifi6e) ALFA AWUS036AXML is a good candidate for Kali Linux and working in Monitor mode.

I am trying to go through a course that requires a usb adapter that is supported in Monitor mode.

There is a list of recommended devices but this new model is not listed.

I was hoping to find some info on it but I could not, maybe I wasn't looking in the right place.

Any pros / cons for this adapter to work with Kali Linux in Monitor mode etc ?

Thanks in advance

morrownr, 2023-06-04:
Hi @iangregsondev 

Let me say right up front that I do not yet have an AXML but hope to soon. I remember a couple of Kali users last year reporting that the driver for the mt7921au chipset was working well in monitor mode in all 3 bands. I think that happened last September or so and I think the adapter was the Comfast cf-953ax. I have a cf-951ax and have done basic monitor mode testing like testing like injection and deauth but nothing extensive. The cf-951ax is probably not what you want as the lack of an external antenna may make the range less than you want.

I hear a rumor that a new good quality adapter that is smaller and less costly than the AXML should be available very soon. It runs on the same chipset as the AXML.

Would you mind posting the recommended list you are looking at? I'm sure we can get some posts with pro's and con's. I have been recommending the ALFA ACHM as the overall best monitor mode adapter for a few years. 

For what it is worth, I know there are a lot of "Best adapters for Kali" lists out there but most are really bad. I think the best list to start with is the "Plug and Play" list here. It is menu item 2. Which is best depends on a lot of factors such as will it be used with a desktop, laptop or RasPi type of computer. Do you need or want a very long range adapter? Do you need a dual band or tri-band adapter? Will you use it with a directional antenna? Will you travel with it?

How soon do you needs to buy?

Regards

iangregsondev, 2023-06-04:
Thank you for the reply! This is very helpful.

I honestly wasn't looking at many others, I wanted a Tri-band supporting Wifi6e so the Alfa AXML was the only one I was initially looking at but maybe there are others that I better suited.

Here is what I was looking at.

1. New laptop ordered (Novacustom NV41 Series 14) - but not expecting delivery until around July it seems :-( So I do have time. Although its a laptop, its will be connected to external monitor/keyboard/mouse - so very much used as a desktop.
2. Looking at some general  wifi hacking courses, it will be used  internally on the laptop but I probably won't be travelling.
3. Looking for a triband.
4. Won't be travelling, will just need standard antennas that come with it, be nice if an external atenna could be added which I believe is the case with eh Alfa AXML i mentioned above. Although thats now my initial thought but it would be great if it was supported.
5. It will be used on Kali - probably exclusively to be truthful.

I would like to purchases it before the laptop arrives, so maybe in around 1 month or a little more.

Thanks in advance for your help

Ian

morrownr, 2023-06-06:
Ian,

My AXML arrived today. My to-do list  is full so if you have specific things you want tested, i will find time.

I confirmed that the antennas are removable. The extension cable is really nice and supports USB-C and USB-A. Overall quality appears to be very good. The color is flat black and does not show fingerprints. Overall, it is a very modern looking adapter. I just to start testing as able but I will test what you want when you let me know what to test.

iangregsondev, 2023-06-06:
Very nice, I suppose its confirmed as supporting Monitor Mode and Packet Injection ?

Sounds like a great piece of kit!

Out of interest, are you using it in Kali Linux or some other OS ?

morrownr, 2023-06-06:
> I suppose its confirmed as supporting Monitor Mode and Packet Injection ?

The mt7921au chipset and the mt7921u driver are confirmed to support monitor mode and packet injection but I hesitate to say yes until I see it myself or get numerous reports saying it does. The AXML is still a fairly new product. The only two recent issues I have seen are AP mode related. I can't say that I remember single complaint about monitor mode yet but my memory is not as good as it used to be. I will test.

> Out of interest, are you using it in Kali Linux or some other OS ?

I usually keep a partisan with Kali on it in my small lab but it made me mad last fall. I have better things to do than fix distros so I deleted it. I am over being pissed now so I will get Kali back up soon. Kali is somewhat less stable than many distros and I have more work to do working on adapter drivers, docs and answering questions. To ensure good testing, I frequently blow out old installations and install new. I have little tolerance for stuff that is broken.

I can test monitor mode but I have 2 high priority jobs I need to work on right now. If you haven't see a report here by next Monday, ping me.

iangregsondev, 2023-06-06:
Sounds great, thanks

ZerBea, 2023-06-07:
@morrownr it would be great if you can confirm/not confirm that active monitor mode is working, too.
I asked it, because some MediaTek chipsets announce (NL80211_FEATURE_ACTIVE_MONITOR) that they support active monitor mode, but unfortunately they don't, e.g. mt7601u:
https://bugzilla.kernel.org/show_bug.cgi?id=217465

In advance,  thanks.
Regards
Mike

ZerBea, 2023-06-07:
Hello @morrownr 
Thanks for your answer. Doing this via a script is not so trivial.

The first part (set monitor mode) is simple. It can be done via iw or via hcxdumptool:

the iw way:
$ sudo ip link set INTERFACE down
$ sudo iw INTERFACE set type monitor
$ sudo iw INTERFACE set monitor active
$ sudo ip link set INTERFACE up

the hcxdumptool way:
$ sudo hcxdumptool -m INTERFACE

The second part is tricky, because mt7601u report "NL80211_FEATURE_ACTIVE_MONITOR" feature.
The only way to detect that it isn't working is to analyze the incoming traffic by e.g. tshark or Wireshark.
If we see only packets to addressed to broadcast (ff:ff:ff:ff:ff:ff) or to the device MAC, active monitor mode is not working.
I guess, this is not feasible via a simple script.

morrownr, 2023-06-08:
@iangregsondev 

Short test of packet injection on 2.4 GHz band:

19:48:30  Trying broadcast probe requests...
19:48:30  Injection is working!
19:48:32  Found 2 APs

19:48:32  Trying directed probe requests...
19:48:32  8C:59:73:FE:8B:F4 - channel: 1 - 'XXXX'
19:48:32  Ping (min/avg/max): 1.421ms/5.613ms/37.082ms Power: -26.10
19:48:32  30/30: 100%

19:48:32  12:59:32:04:A7:B8 - channel: 1 - ''
19:48:33  Ping (min/avg/max): 0.753ms/4.356ms/9.120ms Power: -42.55
19:48:33  22/30:  73% 

Just a quick test. I'll need to do more testing and get my ACHM out to do a comparison before I can offer any opinions.

morrownr, 2023-06-08:
Quick managed mode test:

5 GHz band
channel 100 DFS
80 MHz channel width
3 walls, about 12 meters

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.198 port 38282 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  72.1 MBytes   605 Mbits/sec    0   1.82 MBytes       
[  5]   1.00-2.00   sec  72.5 MBytes   608 Mbits/sec    0   1.82 MBytes       
[  5]   2.00-3.00   sec  75.0 MBytes   629 Mbits/sec    0   1.82 MBytes       
[  5]   3.00-4.00   sec  73.8 MBytes   619 Mbits/sec    0   1.82 MBytes       
[  5]   4.00-5.00   sec  72.5 MBytes   608 Mbits/sec    0   1.82 MBytes       
[  5]   5.00-6.00   sec  73.8 MBytes   619 Mbits/sec    0   1.82 MBytes       
[  5]   6.00-7.00   sec  73.8 MBytes   619 Mbits/sec    0   1.82 MBytes       
[  5]   7.00-8.00   sec  73.8 MBytes   619 Mbits/sec    0   1.82 MBytes       
[  5]   8.00-9.00   sec  72.5 MBytes   608 Mbits/sec    0   1.82 MBytes       
[  5]   9.00-10.00  sec  75.0 MBytes   629 Mbits/sec    0   1.82 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   735 MBytes   616 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   732 MBytes   613 Mbits/sec                  receiver

iperf Done.
```

Not bad. Need to test other channels and distances as well as do some comparison testing.

ZerBea, 2023-06-08:
@morrownr , thanks for the injection test.
Why do you use aireplay-ng? It only count (short) ack frames

morrownr, 2023-06-08:
> @morrownr , thanks for the injection test. Why do you use aireplay-ng? It only count (short) ack frames to determine that injection is working.

I have no preference for aireplay-ng. It just happened to be the closest thing at hand. Tell me what you want tested and the instructions and I will test it.

Did you see my post today about the new Alfa AXM adapter?

morrownr, 2023-06-08:
> The first part (set monitor mode) is simple. It can be done via iw or via hcxdumptool:
> 
```
> the iw way: $ sudo ip link set INTERFACE down $ sudo iw INTERFACE set type monitor $ sudo iw INTERFACE set monitor active $ sudo ip link set INTERFACE up
```

I am aware of this. I wrote start-mon.sh.
 
> The only way to detect that it isn't working is to analyze the incoming traffic by e.g. tshark or Wireshark. If we see only packets addressed to broadcast (ff:ff:ff:ff:ff:ff) or to the device MAC, active monitor mode is not working. I guess, this is not feasible via a simple script.

Don't bet on it. I think you just wrote the logic to make it work. Tell me about tshark.

> BTW: Active monitor mode is an amazing feature (if targeting a PMKID). 

You can educate me more. We could use parts of start-mon.sh to start a new script. It appears that what we need is a text mode app to handle the capturing. BASH is actually capable of of doing complicated work.

ZerBea, 2023-06-09:
> I have no preference for aireplay-ng. It just happened to be the closest thing at hand. Tell me what you want tested and the instructions and I will test it.
>
> Did you see my post today about the new [...]

Thanks for your answer. Aireplay-ng is fine do to an initial test but it doesn't cover some  things.
E.g After I wrote this comment:
https://github.com/morrownr/USB-WiFi/issues/275#issuecomment-1581041731

I tested an ALFA AWUS036ACM and noticed that frequency change via NETLINK is very slow.
My default scan time was one second - and that is not enough. To make sure, the device is on the new frequency and ready to transmit, I had to increase it to at least two seconds:
https://github.com/ZerBea/hcxdumptool/commit/0a940f251237c032d83833549b36e6543f089d7c#diff-8125ff62a279e92c823e77b6ba7da82b2544b5db078008535e0bf8034cbc96f7R2793

Yes, I noticed you post and I'm surprised about the new L adapter, too.

Some words about hcxdumptool/hcxtools. This penetration testing tools are kind of a WiFi pre-processor to hashcat or to John the Ripper (JtR). Goal is to discover weak points in (own) NETWORKs in a very short time.
`hcxdumptool -> hcxtools -> hashcat/JtR`
Using this workflow, we discovered the PMKID attack (for the first time):
https://hashcat.net/forum/thread-7717.html

The PMKID attack is not the only one. There are several other weak points e.g. transmit password in the clear:
https://github.com/evilsocket/pwnagotchi/issues/835#issuecomment-598597214

or get IMEI of a cell phone:
https://hashcat.net/forum/thread-6661-post-44789.html#pid44789

or downgrade WPA3 CLIENT to WPA2
or get EAPOL M2 from an unconnected WPA CLIENT
or get EAP-ID

The simplest way to discover if your ACCESS POINT or your CLIENTs are weak is to run hcxdumptool against them:
stop all NETWORK services that take access to the interface (e.g. NetworkManager and wpa_supplicant)
$ sudo hcxdumptool -i interface -c channel_o_ap (e.g. 1a) --rds=1

If you got a PMKID, an EAPOL M3 (AP) or and EAPOL M2 (CLIENT) you can stop the attack, convert the pcapng dump file to a format that hashcat or JtR understand (by pcapngtool). Now you're ready to break the password.
hcxpcapngtool is part of hcxtools, which are running in back ground of several online services e.g.:
https://wpa-sec.stanev.org/
https://hashcat.net/cap2hashcat/

Additional you can use hcxdumptool as a simple WiFi scanner (active or passive):
$ sudo hcxdumptool -i wlp22s0f0u4 -F --rcascan=active

You directly see if you got a PROBERESPONSE to a PROBEREQUEST:
```
  CHA  FREQ   BEACON  RESPONSE A   MAC-AP   ESSID  SCAN-FREQUENCY:   2457
--------------------------------------------------------------------------
 [ 10  2457] 08:03:50 08:03:50 + 0a96d798f10e TEST-G
 [ 10  2457] 08:03:50 08:03:50 + 0896d798f10e AP_7272
```
An ACK is a short frame that doesn't contain useful information. If a channel is busy, there is mostly a time to transmit this short frame.
A PROBERESPONSE frame is big and contain a lot of information. If a channel is busy, there is no time slot to transmit this big frame. In that case you can stop your attacks, because you will not get useful (big) frames from the target.
That makes the difference: counting small ACKs or retrieving full frames.

In attack mode you can see what kind of EAPOL MESSAGEs the AP respond (upper display) or the CLIENT respond (lower display). And for sure, most of them respond to hcxdumptool's layer 2 attacks.

tshark is a great tool. Mostly I run it in parallel to hcxdumptool (while coding new features) to make sure that hcxdumtool is exactly doing what I want to do it. 
It is the little brother of Wireshark. It came with Wireshark, but doesn't provide a GUI.
Running bash tools like awk on the columns of tshark terminal output it might be possible to detect if active monitor is working as expected. But I haven't tested this.

morrownr, 2023-06-11:
@ZerBea 

Thanks for the info. I want to get back to this topic but it may be a few days as I am busy.

Foolian2, 2024-01-09:
any updates @morrownr ?

ZerBea, 2024-01-09:
Now it is completely broken on kernel:
$ uname -r
6.6.10-arch1-1

https://github.com/ZerBea/hcxdumptool/discussions/361#discussioncomment-7567045

ZerBea, 2024-02-28:
That highly depend on the kernel.
As of today running Arch Linux (kernel 6.7) and Debian (kernel 6.6).
https://github.com/openwrt/mt76/issues/839#issuecomment-1968509383

morrownr, 2025-06-16:
Hi @HalluxPollex 

> I’m having trouble monitoring 5 GHz networks. When I run airodump-ng, it only shows 2.4 GHz networks and WPA2 — no WPA3 or 5 GHz networks appear. I use kali.

Have you successfully used the adapter on 5 GHz in managed mode?

What is the result when you run:

$ iw reg get

morrownr, 2025-07-25:
Hi @leosdreams 

> Does ALFA AWUS036AXM and ALFA AWUS036AXML supports packet injection & monitor mode in Kali Linux?

Yes, they even support active monitor mode. I currently do not list either in The Plug and Play List or the Recommended for Kali List as both ran into a significant problem last year that was related to their bluetooth support. This problem was messing up wifi support and was hard to find and fix. My general recommendation is to buy USB WiFi adapters that do not include bluetooth support and this case shows one reason why. As of kernel 6.14, the problem appears to be fixed. I am continuing to test both adapters and am watching reports. When I am convinced users will not have problems, I may add them back to the lists.

If you go to the site Main Menu, you will see menu item 2, That is where The Plug and Play List and the Recommended for Kali lists are:

https://github.com/morrownr/USB-WiFi

What specific capabilities do you need? long range, dual band, active monitor. etc.

There are new adapters that will be available soon that should work well. New tri-band WiFi 6 adapters with the mt7921au chip as well as new tri-band WiFi 7 adapters with the mt7925 chip should be available soon. In fact I now have a new Netgear A9000 USB WiFi adapter. It is a WiFi 7 adapter. I am testing it but am not ready to recommend it yet. 

The Alfa ACH is an old adapter but a good one if you only need dual band. The new in-kernel driver in kernel 6.14 and later gives it fresh life and it has wonderful range. I am kind of fond of the Edup AX1672 as it is a reasonably prices adapter that is such that it almost seems like it was made for Linux.

morrownr, 2025-07-25:
Give me a report when you have time.

enisoku, 2025-11-11:
@morrownr hey man i am new to pentesting, ethical hacking etc. and i need a usb wifi adapter for my macbook pro m3 with a kali linux vm running on parallels. i am looking for active monitor mode, packet injection, decent range, dual/triband. i've done some research, looked at your lists, and checked the aircrack site and decided to get the ALFA AWUS036AXML. can you tell me if that was a good decision for my demanded capabilities as of the newest state of things. my other two options were ALFA AWUS036ACM and EDUP EP-AX1672. should i have went for one of those two instead?

best regards

Shallisteria, 2025-11-11:
> [@morrownr](https://github.com/morrownr) hey man i am new to pentesting, ethical hacking etc. and i need a usb wifi adapter...

I'm not morrownr but AWUS036AXML has the same chipset as AX1672 but if there is a difference then maybe in their supporting components like the amplifier, the capacitor or something else. I have AX1672 and it's my most satisfying USB Wi-Fi i have ever had, it runs cool so you can use it 24/7 and cheaper than Alfa AXML adapter. But if you're concerned by range then you need to check Alfa AWUS036ACHM

ZerBea, 2025-11-12:
@enisoku , as of today, you won't be happy with an AXML:

General information about the Linux kernel and tested device:
```
$ uname -r
6.17.7-arch1-1

$ lsusb
Bus 001 Device 008: ID 0e8d:7961 MediaTek Inc. Wireless_Device

$ hcxdumptool -l
  0	  3	00c0cab571e9	b025aa19f08f	*	wlp22s0f0u4i3   	mt7921u	NETLINK
```

Real world monitor mode and packet injection test:
```
$ sudo hcxdumptool --tot=2 --rcascan=active

This is a highly experimental penetration testing tool!
It is made to detect vulnerabilities in your NETWORK mercilessly!
Misuse within a network, without specific authorization, may cause
irreparable damage and result in significant consequences!
Not understanding what you were doing is not going to work as an excuse!

starting...

900 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
0 BEACONs received (monitor mode is possibly not working)
0 PROBERESPONSEs received (packet injection is possibly notworking)
exit on TOT
```

The driver only handles DATA packets, which is confirmed by tshark - other frame types are ignored!
```
$ tshark -i wlp22s0f0u4i3 -o 'gui.column.format:"No.","%m","Time","%t","Protocol","%p","Length","%L","Info","%i"'
Capturing on 'wlp22s0f0u4i3'
    1 0.000000000 802.11 142 Data, SN=2002, FN=0, Flags=.p....F.
    2 0.051878381 802.11 175 Data, SN=787, FN=0, Flags=.p....F.
    3 0.096973698 802.11 175 Data, SN=3630, FN=0, Flags=.p....F.
    4 0.147462483 802.11 142 Data, SN=788, FN=0, Flags=.p....F.
    5 0.199042049 802.11 134 Data, SN=3632, FN=0, Flags=.p....F.
    6 0.506265084 802.11 196 Data, SN=3636, FN=0, Flags=.pm...F.
    7 0.508294521 802.11 274 Data, SN=3637, FN=0, Flags=.pm...F.
    8 0.509312145 802.11 148 Data, SN=3638, FN=0, Flags=.pm...F.
    9 0.511798292 802.11 241 Data, SN=3639, FN=0, Flags=.p....F.
   10 0.564046712 802.11 196 Data, SN=794, FN=0, Flags=.pm...F.
   11 0.566041905 802.11 274 Data, SN=795, FN=0, Flags=.pm...F.
   12 0.566962368 802.11 142 Data, SN=796, FN=0, Flags=.pm...F.
   13 0.567951720 802.11 148 Data, SN=797, FN=0, Flags=.pm...F.
   14 0.569701326 802.11 241 Data, SN=798, FN=0, Flags=.p....F.
   15 0.614838621 802.11 196 Data, SN=2009, FN=0, Flags=.pm...F.
   16 1.485693465 802.11 196 Data, SN=810, FN=0, Flags=.pm...F.
   17 1.487706242 802.11 274 Data, SN=811, FN=0, Flags=.pm...F.
   18 1.488809615 802.11 148 Data, SN=812, FN=0, Flags=.pm...F.
```

iw confirms that it is really in monitor mode:
```
$ iw dev
phy#0
	Interface wlp22s0f0u4i3
		ifindex 3
		wdev 0x1
		addr b0:25:aa:19:f0:8f
		type monitor
		channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```

ZerBea, 2025-11-12:
Not easy to answer, because:
- I don't use KALI
- I don't use the aircrack-ng suite
- I don't inject frames on 5, 6 or 7GHz because the range is far too short and because I don't need to do [that](https://github.com/ZerBea/hcxdumptool/blob/master/README.md).

Running in this environment:
- OpenWRT (Raspberry Pi Zero) & hcxdumptool/hcxlabtool
- Arch Linux Notebook and Desktop & hcxdumptool/hcxlabtool

I prefer:
devices with external antenna:
- TP-Link TL-WN722N v2/3
- ALFA AWUS036NH
- ALFA AWUS036NHA

devices with fixed mounted antenna:
- TP-Link TL-WN822N
- TP-Link TL-WN823N
- EDIMAX EW-7711UAn V2

 
BTW:
Here are some real adapter tests:
https://github.com/ZerBea/hcxdumptool/discussions/361

ZerBea, 2025-11-12:
> > i mentioned the ALFA AWUS036ACM and EDUP EP-AX1672 are any of those better and if so which one?
> 
> Why not check AWUS036ACHM? It's range is longer than the ACM counterpart. While for AX1672 after using it heavily from July this year I don't find any issues both in monitor mode and packet injection, but the adapter is wide enough to block the adjacent USB port if you're plugging it directly to the USB port on the PC

The AWUS036ACM  is affected by the broken driver, too:
```
0 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
0 BEACONs received (monitor mode is possibly not working)
0 PROBERESPONSEs received (packet injection is possibly notworking)
exit on sigterm
```

Just for reference - a 12,49€ device:
```
$ lsusb:
Bus 005 Device 005: ID 2357:0108 TP-Link TL-WN822N Version 4 RTL8192EU

$  sudo hcxdumptool --tot=2 --rcascan=active
1033 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on TOT
```

ZerBea, 2025-11-12:
@morrownr 

> I have some testing to do for monitor mode for another guy anyway but my health problems have slowed me down.

I'm very excited about the (detailed) results of your tests. Especially the environment (target AP, target CLIENT, used tools).

ZerBea, 2025-11-12:
@morrownr thanks, I'm interested in 2 tests on mt7925:

Is monitor mode and packet injection working:
```
$ sudo hcxdumptool -i interface_name --tot=2 --rcascan=active
...
? Packet(s) captured by kernel
? Packet(s) dropped by kernel
```

How long does it take to successful attack a target:
```
$ time sudo hcxdumptool  --exitoneapol=14
...
? Packet(s) captured by kernel
? Packet(s) dropped by kernel
exit on EAPOL M1M2

real ?m?s
user ?m?s
sys  ?m?s
```

and if the target provide a PMKID:
```
$ time sudo hcxdumptool  --exitoneapol=1
...
? Packet(s) captured by kernel
? Packet(s) dropped by kernel
exit on PMKID

real ?m?s
user ?m?s
sys  ?m?s
```

My reference for  both attacks TP-Link TL-WN822N:

```
$ time sudo hcxdumptool  --exitoneapol=14
...
445 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on EAPOL M1M2ROGUE

real 0m5,468s
user 0m0,008s
sys  0m0,004s

$ time sudo hcxdumptool  --exitoneapol=1
...
488 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on PMKID

real 0m7,217s
user 0m0,005s
sys  0m0,007s
```

enisoku, 2025-11-12:
@morrownr 

> For you, I will start by using Aircrack-ng, except for where I have my own tools, and will test going into monitor mode and doing packet injection plus maybe a couple of others things.

that's very nice of you man 😁 so will you be testing the mt7921au or mt7612u chipset or both? hcxdumptool would also be interesting. i've seen @ZerBea 's tests but maybe it's different with different drivers?

Shallisteria, 2025-11-12:
> And [@Shallisteria] do you use your AX1672 only for 2.4ghz for packet injection or do u use dual band?

I plan to use every feature that exist in every adapter that I have, but right now i can only use 2.4 and 5 GHz band.

> Can I get you guys to tell me what tools you plan to use and what you will be doing?

For now I use Airgeddon that if I remember correctly use Aircrack-ng? I'm still newbie so I plan to learn on what kind of environment penetration test works and what makes it don't work, so far I have tested in Wi-Fi 4, Wi-Fi 5 and Wi-Fi 6 routers and as ZerBea has said modern APs have some kind of protection (In my case it's my Wi-Fi 6 router) so deauth isn't working there.

I use Kali that is still on 6.8, and Linux Mint that I believe use 6.14 and last time I played with it (I think it's around september) the monitor mode and packet injection is working

And thanks ZerBea I finally found out what's the name for that protection, I thought my adapter or my Linux installation is faulty when that protection kicked in

Shallisteria, 2025-11-13:
> don't worry about MFP

Oh? I'll remember it for my next trial run. I don't know what tools to start so I went with alphabetical order in the menu and it just so happen aircrack and airgeddon start with "a". But as much as i wanted to I can't start right now so I hope my brain don't forget this when I can start trying stuff again

Shallisteria, 2025-11-13:
> * do  not know how to set up a Berkeley Packet Filter
> * running (a mostly unconfigured) KALI Linux. Just take a look at hashcat forum and you'll notice that the robot replace "KALI" with "The-Distribution-Which-Does-Not-Handle-OpenCL-Well (Kali) linux".
> 
> If you like scripts like airgeddon, consider to take a look at [WiFite2](https://github.com/kimocoder) which runs hcxtools as [default](https://github.com/kimocoder/wifite2/commit/2143b22c41d925432f3f9b65fb5a4eff0adad2bb). If you like a beautiful GUI, please take a look at [angryoxide](https://github.com/Ragnt/AngryOxide).

Ok I admit those two bullet point are new info to me. I mostly run Linux Mint tho since Kali is.. idk doesn't click for me so I'm still distro hopping. And thanks for the script recommendation, and I really mean it I checked it and it's kind of awesome, especially WiFite2 I like it more than the included WiFite in Kali.

But is it okay talking like this here? As much as i want to know more it seems like this is out of topic

enisoku, 2025-11-13:
@Shallisteria is there a way i can contact you privately? i want to know more about how you got into cyber security, wifi hacking etc. like an email address

morrownr, 2025-11-14:
@ZerBea 

Okay, good. I'll go ahead and make a full run of your items to check with this mt7921 based adapter. Let me know when you see things that implicate the mt7921u driver in bad behavior and I'll apply the patch and compile kernels as you see fit.

enisoku, 2025-11-14:
> Great, thanks. It is the right direction. The 9480 by kernel dropped packets are related to your distribution and its configuration, which isn't optimized to run such penetration tests. Your Raspberry is too slow to handle all incoming RAW packets using this configuration. This is possibly caused by too many running services.

@ZerBea 

may I ask what's the best way to configure my distribution? Take Kali as an example are there guides or anything to configure it the most efficient way?

ZerBea, 2025-11-14:
@enisoku , just disable all services that you do not need.
https://www.tecmint.com/list-all-running-services-under-systemd-in-linux/
And make sure that no other tool/service takes access to the interface.

ZerBea, 2025-11-15:
I neither use VNC nor ssh. Szraight sfter initisl boot I start hcxdumptool vis init.d.
If VNC or ssh slows down, you can expect packet drops.

ZerBea, 2025-11-16:
@morrownr , your script to set monitor mode is fine, and it is doing a good job.
Your first test confirmed that the configuration of a distribution hss a huhe impact on the performance.
How to apply the patch depends on the kind of the patch (e.g. git diff) and on the distribution (are the kernel headers installed).

morrownr, 2025-11-17:
@ZerBea 

Can your hcxdumptool test for active monitor mode?

Testing continues.

hcxdumptool --tot=2 --rcascan=active

42318 Packet(s) captured by kernel
4183 Packet(s) dropped by kernel
exit on TOT

The above test was with a Alfa ACM adapter, mt7612u driver, RasPi4B hardware, RasPiOS 64 bit distro, 6.12 kernel.

It appears that the Alfa ACM is also passing and supports monitor mode and injection. The reason for the dropped packets could very well be that the ACM, with its nice antennas, has longer range and is having to process many more packets than the Netgear A9000.

-----

List of Drivers and Adapters that are Passing Monitor Mode and Injection Testing
- I will keep building this list as I continue with additional tests.
- Range is based on a scale of short, medium, long and extra long and is based on what I am seeing.

Command line: sudo hcxdumptool --tot=2 --rcascan=active

Driver: mt7925u
Adapter: Netgear A9000 (WiFi 7)
Result: Pass
27038 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on TOT
Estimated range: medium

Driver: mt7612u
Adapter: Alfa ACM (WiFi 5)
Result: Pass
42318 Packet(s) captured by kernel
4183 Packet(s) dropped by kernel
exit on TOT
Estimated range: Long

-----

ZerBea, 2025-11-17:
> Estimated range: medium.

We will see how the adapter performs in attack mode.
For a succesful attack we need both directions:
- packets received by target
- packets received by attack device

I mentoned that in several comments:
- RSSI is meaningless
- own TX power is meaningless

albaluz35, 2025-11-20:
@morrownr what is the Patch that you ere talking about related to mt7921u dongles like the edup AX1672? Where can I find it?

ZerBea, 2025-11-20:
As of today, in no case the AXML is working flawlees in monitor mode for me.

enisoku, 2025-11-22:
@ZerBea 

oh wow 😳

should i rather go with 6.12 or 6.8?

morrownr, 2025-11-26:
@ZerBea 

> looks like the device is very sensitive: "108355 Packet(s) captured by kernel"

I noticed that this number is very different than what we saw with kernel 6.12. I'm planning on 3 data sets for the drivers/adapters that I am testing. Basic kernel 6.12, kernel 6.18 as is and kernel 6.18 with one or more patches. Hopefully this will give us enough data to know where we are and I'll work specific issues if needed. I noticed that the patch we discussed above is going into kernel 6.19 but it was not scheduled to backport to 6.18 and 6.17. I see that as a mistake that needs attention but will wait for the results of the testing before deciding what to do.

> unfortunately you system isn't able to handle all the incoming packets: "4578 Packet(s) dropped by kernel" - but this is not a fault of the device or the driver.

If I am understanding you correctly, for the purposes of this testing, which is simply to determine if specific monitor mode capabilities are working on specific drivers, my test setup should work okay. My expertise is more in the AP and managed modes so I depend on you to guide me for monitor mode testing. I appreciate what you do.

Just to clarify the drivers that are part of this test:

mt7925u
mt7921u
mt7612u
mt7610u
rt2800usb (RT5370 chip)
rtl8814au
rtl8812au
rtl8821/11au

The reason for those drivers: The purpose of this site is primary to help users find USB WiFi adapters that they can buy new and expect very reliable results for what they are doing. That eliminates a lot of chips/drivers/adapters due to many factors.

ZerBea, 2025-11-30:
@morrownr , based on your tests, I've pushed an update to hcxdumptool git.

Now hcxdumptool print a warning if the user request active monitor mode, but the driver can't provide it:
```
$ sudo hcxdumptool --rcascan=active -A

Warning: active monitor mode requested by user, but not supported by driver!

This is a highly experimental penetration testing tool!
It is made to detect vulnerabilities in your NETWORK mercilessly!
Misuse within a network, without specific authorization, may cause
irreparable damage and result in significant consequences!
Not understanding what you were doing is not going to work as an excuse!

starting...
```
Please update to latest git head and do the test again.

ZerBea, 2025-12-02:
@morrownr 

Driver reports that active monitor mode is supported.
```
$lsusb
Bus 005 Device 008: ID 7392:7710 Edimax Technology Co., Ltd Edimax Wi-Fi

$ hcxdumptool -I wlp48s0f4u2u4
interface information:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0   3 74da38f2038f 980ee49f85c2 * wlp48s0f4u2u4    mt7601u (NETLINK)

available frequencies: frequency [channel] tx-power of Regulatory Domain: DE

  2412 [  1] 20.0 dBm	  2417 [  2] 20.0 dBm	  2422 [  3] 20.0 dBm	  2427 [  4] 20.0 dBm
  2432 [  5] 20.0 dBm	  2437 [  6] 20.0 dBm	  2442 [  7] 20.0 dBm	  2447 [  8] 20.0 dBm
  2452 [  9] 20.0 dBm	  2457 [ 10] 20.0 dBm	  2462 [ 11] 20.0 dBm	  2467 [ 12] 20.0 dBm
  2472 [ 13] 20.0 dBm
[zerobeat@tux1 tmp]$ hcxdumptool -L wlp48s0f4u2u4
available physical wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0   3 74da38f2038f 980ee49f85c2 * wlp48s0f4u2u4    mt7601u (NETLINK)

modes reported by the driver:
* active monitor mode available (do not trust it)
S active monitor mode available on shared interface (do not trust it)
+ monitor mode available
s monitor mode available on shared interface
- no monitor available
```

If user set active monitor mode, it does not transmit any longer.
```
$ sudo hcxdumptool -A --rcascan=active
^C
418 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
0 PROBERESPONSEs received (packet injection is possibly notworking)
exit on sigterm
```

If user does not set active monitor mode, everythig is fine:
```
$ sudo hcxdumptool --rcascan=active
^C
177 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on sigterm
```

ZerBea, 2025-12-02:
Next one (mt76x0u)::
```
$ lsusb
Bus 005 Device 010: ID 0e8d:7610 MediaTek Inc. WiFi

$ hcxdumptool -l
  6	  9	00c0cab10a87	a06260145fe0	*	wlp48s0f4u2u3   	mt76x0u	NETLINK
[zerobeat@tux1 tmp]$ hcxdumptool -L wlp48s0f4u2u3
available physical wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  6   9 00c0cab10a87 a06260145fe0 * wlp48s0f4u2u3    mt76x0u (NETLINK)

modes reported by the driver:
* active monitor mode available (do not trust it)
S active monitor mode available on shared interface (do not trust it)
+ monitor mode available
s monitor mode available on shared interface
- no monitor available

$ sudo hcxdumptool --rcascan=active
^C
291 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on sigterm

$ sudo hcxdumptool -A  --rcascan=active
^C
386 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
0 PROBERESPONSEs received (packet injection is possibly notworking)
exit on sigterm
```

ZerBea, 2025-12-02:
> Do you know the last kernel and firmware where it was working?

Active monitor mode has never worked for me on mt76 devices. It has worked on mt7601u, but that's a long time ago.

ZerBea, 2025-12-02:
I only have a TP-Link Archer T2U PLUS (RTL8821AU) - no mt7925u, no rtl8814au, no rtl8812au:
```
$ lsusb
Bus 005 Device 013: ID 2357:0120 TP-Link Archer T2U PLUS [RTL8821AU]

$ hcxdumptool -L wlp48s0f4u2u4
available physical wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  8  11 e4fac44d0096 9c93e4bb92a9 + wlp48s0f4u2u4    rtw88_8821au (NETLINK)

modes reported by the driver:
* active monitor mode available (do not trust it)
S active monitor mode available on shared interface (do not trust it)
+ monitor mode available
s monitor mode available on shared interface
- no monitor available

$ sudo hcxdumptool  --rcascan=active
^C
168 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on sigterm

$ time sudo hcxdumptool --exitoneapol=14 -c 6a --rds=1
275 Packet(s) captured by kernel
0 Packet(s) dropped by kernel
exit on EAPOL M1M2ROGUE

real	0m8,:00s
user	0m0,004s
sys	0m0,011s
```

ZerBea, 2025-12-03:
@morrownr 

>  Active monitor mode is just a specific capability within monitor mode and, as best I can tell, most people do not need it.

...and there is much more that people don't need when running hcxdumptool -> hcxtools -> hashcat/JtR
Just take a look at the [adapter tests](https://github.com/ZerBea/hcxdumptool/discussions/361) and you'll notice that especially these cheap devices (with external antenna connectors) are working well!
But anyway, I love the glossy fairy tails about TX power, range, need for 5,6,7 GHz, written especially for newbies!

If you run hcxdumptool:
- you don't need an oversized all in one distribution - OpenWRT or Arch Linux is more than enough
- you don't need high TX power devices, because we are packet oriented. Attack TX power should always be the same as TX power of the target.  If it is higher you jam the channel!
- you do not need active monitor mode, because hcxdumtool requests missing frames
- you only need 2.4 GHz because hcxdumptool is able to request hash information from 5,6,7 GHz on 2.4 GHz
- in case of MFP your attacks are successful, because hcxdumtool doesn't transmit stupid and useless DEAUTHENTICATION frames
- in case of WP3 transition mode, hcxdumptool downgrades the CLIENT to WPA2

If you run hcxpcapngtool, the captured hashes will be converted to a format that the big GPU crackes hashcat and JtR (john) understand - that's the royal class of hash cracking!

If you run other hcxtools, the converted hash file can be additional filtered or pre-processed. That decrease the cracking time of the GPU crackers.

ZerBea, 2025-12-03:
@morrownr , great idea. I'll add my test results there, too.