morrownr/USB-WiFi #276: Comfast adapters (PCB, board scheme, circuit design)

user723045, 2023-06-07:
Hello. 
I'm looking for a cheap but affordable quality replacement for Alfa adapters.
I am planning to buy a couple of adapters from Comfast and therefore I have a request for PCB photos and information about adapter circuitry from this manufacturer.

I was able to find photos of some adapters and I was disappointed with the stuffing of the adapters.

COMFAST CF-WU785AC (chipset MT7612UN)
[image] [image]
The adapter contains fake antennas, the central core of the antenna is not soldered anywhere.
Also, the board does not have any signal amplifiers (PA, LNA) for comparison, take a look at the board from ALFA
[image]

COMFAST CF-926AC V2 (chipset MT7612UN)
[image]

---
If there are owners of the following adapters:
- Comfast CF-951AX (mt7921au)
- Comfast CF-953AX (mt7921au)
- Comfast CF-WU782AC (mt7612un)

Could you provide photos of the internals of your adapters to evaluate the quality of their circuit designs?

---
It would also be helpful if someone could share pictures of the PCBs of these noname adapters which can be easily found on aliexpress.
Photos of these adapters:
[image] [image]

morrownr, 2023-06-08:
[image]

Comfast CF-951AX

> I'm looking for a cheap but affordable quality...

Using both cheap and quality in the same sentence is suspect but I get what you are saying. If you compare athe verage quality of Alfa products vs Comfast, there is no comparison as Alfa produces high quality products on the whole and Comfast produces low cost products but if we look at individuals products, we can probably find something that hits the sweet spot for you.

To help, we really need to know some details about how you intend to use the adapter:

- What distro are you using?
- What kernel are you using?
- What modes do you use? Managed, AP, monitor?
- How do you use the modes? Basic managed mode or packet capture with Wireshark or something else?
- Do you need long range? Or same room only or something else?
- Do you intend to take the adapter on the road?

Regards

ea4eoz, 2023-06-23:
Comfast CF-953AX

Same PCB as CF-951AX but soldered antennas.

[image]

[image]