morrownr/USB-WiFi #278: Possible issue with MT7921U reporting TX power

ea4eoz, 2023-06-12:
Hello!

Today I noticed the MT7921U kernel module is probably reporting a bad TX power:

for example: 

iwconfig wlan0
wlan0  IEEE 802.11  ESSID:"xxxxx"  
          Mode:Managed  Frequency:5.18 GHz  Access Point: A6:64:XX:XX:XX:XX   
          Bit Rate=866.7 Mb/s   Tx-Power=3 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=62/70  Signal level=-48 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

I really doubt it would be outputting 3 dBm (2 mW) because AP is distant and it is working like a charm. Same output with other utilities, like wavemon. The adapter is the Comfast CF-953AX with USB id 3574:6211

Can anyone confirm this?

Using debian unstable with kernel 6.3.5-1

morrownr, 2023-06-13:
> Today I noticed the MT7921U kernel module is probably reporting a bad TX power:

A bad report is the prevailing theory. This issue is listed in the sticky issue pinned at the top. I probably need to add some detail.

For what it is worth: Of the supported MT76 chipsets, the only one that does not show this problem is the mt7612u. I have devices with the following chipsets and they all show txpower -3 dBm:

mt7610u
mt7921au
mt7922 (PCIe card)

> AP is distant and it is working like a charm.

That is what I see as well. There is no way on a distant AP that you would be seeing the signal you are if the txpower really was -3 dBm. Maybe we can get one of the Mediatek devs to take a look.

ea4eoz, 2023-06-13:
> Maybe we can get one of the Mediatek devs to take a look.

That would be nice.

morrownr, 2023-06-15:
Let me ponder the best way to present this issue. I know the devs are very busy adding features to the WiFi 6 drivers and creating the WiFi 7 drivers. This may be on their to-do list but since it is a cosmetic issue, it may not be high priority.