morrownr/USB-WiFi #279: OpenWRT BrosTrend AC1200 Linux Compatible USB WiFi Adapter - AC3L

oOEnyaliosOo, 2023-06-12:
Hey Guys, i try to connect my new BrosTrend adapter, but i always fail. I add kmod-mt7921u to the system like this howto:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

"Note about OpenWRT: There is an exception to the above for OpenWRT. MT7921u has been backported to the kernel (5.10) used in OpenWRT 22.03.x. Starting with OpenWRT 22.03.3, simply install the following package:

kmod-mt7921u "

But i can´t finish it. It doesn´t work.

Some Infos:

Mon Jun 12 19:39:41 2023 kern.info kernel: [ 2118.630395] usb 1-1.1: USB disconnect, device number 3
Mon Jun 12 19:39:45 2023 kern.info kernel: [ 2122.765476] usb 1-1.1: new high-speed USB device number 7 using xhci_hcd
Mon Jun 12 19:39:45 2023 kern.info kernel: [ 2122.896488] usb 1-1.1: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
Mon Jun 12 19:39:45 2023 kern.info kernel: [ 2122.905021] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mon Jun 12 19:39:45 2023 kern.info kernel: [ 2122.912398] usb 1-1.1: Product: 802.11ac NIC
Mon Jun 12 19:39:45 2023 kern.info kernel: [ 2122.916696] usb 1-1.1: Manufacturer: Realtek
Mon Jun 12 19:39:45 2023 kern.info kernel: [ 2122.920964] usb 1-1.1: SerialNumber: 123456

lsusb:

root@OpenWrt:~# lsusb
Bus 001 Device 005: ID 1997:2433   Mini Keyboard
Bus 002 Device 001: ID 1d6b:0003 Linux 5.10.176 xhci-hcd xHCI Host Controller
Bus 001 Device 003: ID 0bda:b812 Realtek 802.11ac NIC
Bus 001 Device 006: ID 12d1:14dc HUAWEI_MOBILE HUAWEI_MOBILE
Bus 001 Device 002: ID 2109:3431  USB2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux 5.10.176 xhci-hcd xHCI Host Controller

OpenWRT= openwrt-22.03.5-99c242c2ad02-bcm27xx-bcm2711-rpi-4-ext4-factory.img

Let me know if u need more infos. My Huawei e5332 works great as tethering.

Thx for reading!
Maik

bjlockie, 2023-06-12:
It's Realtek.
I'd send it back if you can.

On 2023-06-12 15:45, oOEnyaliosOo wrote:
> Bus 001 Device 003: ID 0bda:b812 Realtek 802.11ac NIC

oOEnyaliosOo, 2023-06-12:
Ok i will send it back. Thx.

morrownr, 2023-06-12:
@oOEnyaliosOo 

@bjlockie speaks words of wisdom. Brostrend uses Realtek chipsets. Realtek chipsets on OpenWRT are a challenge. Mediatek chipsets will work MUC?H better.

>I will try this one:
https://www.amazon.de/gp/product/B08BJS8FXD/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

> is it ok ?

Yes. I have used my Alfa ACM a lot as an AP. Sometimes with a RasPi4B using my setup guide and sometimes with OpenWRT on my router that has a USB3 port. The driver you will need to use is: kmod-mt76x2u

Do you have any questions about the Plug and Play list (menu item 2)? Hopefully we did not cause the confusion.

oOEnyaliosOo, 2023-06-12:
As i understand this chipset isn´t p&p, right? It isn´t nessessory for my OpenWRT project on the Raspberry Pi 4B.

oOEnyaliosOo, 2023-06-13:
Ok it will arrive tomorrow. If i have probs i will ask in a new Thread with the right topic name. But may u can answer an other question. which 100/1000 ethernet stick would you recommend to me? I dont want to open an new Thread for this little questen.

Thx for your help.

bjlockie, 2023-06-13:
I have a TP-Link USB to Ethernet Adapter (UE300) but I think most will work with Linux (I am guessing but the 1st one I bought worked :-)).

oOEnyaliosOo, 2023-06-13:
I want an extra usb ethernet port on my raspberry pi 4 with OpenWRT.

oOEnyaliosOo, 2023-06-15:
Both Alfa Sticks worked out of the box. Without any problems. Thanks for that hint.

closed 2023-06-15 by oOEnyaliosOo

morrownr, 2023-06-15:
Glad it worked out for you.

> I want an extra usb ethernet port on my raspberry pi 4 with OpenWRT.

I don't have a specific recommendation other than go snoop around the RasPi Forums... maybe ask what users use. I know many RasPi users use these adapters and some get better results than others so be careful.