morrownr/USB-WiFi #281: Summer sale: Comfast CF953AX - WiFi 6e - Plug and Play - 17.63 USD plus free shipping

morrownr, 2023-06-16:
AliExpress

https://www.aliexpress.us/item/3256804283254522.html?gatewayAdapt=glo2usa

The above link is the same link that is used in the Plug and Play list. Many of the reviews are from Linux users. This low cost adapter can meet the needs of many users so research it and see if it can work for you.

It appears the sale will last for 2 more days.

@morrownr

bjlockie, 2023-06-16:
It's the same as I got from aliexpress.
I use it on my raspberry pi 4 as a client.

morrownr, 2023-06-17:
How is it performing?

When I saw the sale, I reread the reviews and looked around here to see if I could find any reported problems and it sure looks to me like folks seem reasonably happy with it.

For what it is worth, I removed the Comfast CF-951AX from the Plug and Play list today. I have one and I never could get it to work with extension cables or powered hubs which I think is important. And I was not getting any reports from others. With more adapters becoming available, I think it was time for it go.

bjlockie, 2023-06-17:
On 2023-06-16 21:18, morrownr wrote:
>
> How is it performing?
>
I have it plugged into an extension cable and a powered hub.
I think I got lucky with both though.
I am not using it as an AP but it is probably close to directly on the 
floor above the router (the router is on the top shelf of a shelf and 
the pi is on the bottom of a shelf, so may 10ft s=away with a floor in 
between).
$ iw dev wlan0 link
         freq: 5180
         signal: -52 dBm
         rx bitrate: 1200.9 MBit/s 80MHz HE-MCS 11 HE-NSS 2 HE-GI 0 HE-DCM 0
         tx bitrate: 960.7 MBit/s 80MHz HE-MCS 9 HE-NSS 2 HE-GI 0 HE-DCM 0

         bss flags:      short-slot-time
         dtim period:    1
         beacon int:     100
I don't have a wired iperf server.
> When I saw the sale, I reread the reviews and looked around here to
> see if I could find any reported problems and it sure looks to me like
> folks seem reasonably happy with it.
>
> For what it is [...]

I doubt it is very different on the inside than a 951 except for the 
antennas.

morrownr, 2023-06-17:
> I have it plugged into an extension cable and a powered hub.

Which is more than I could get from my 951.

> signal: -52 dBm

That is not bad at all but you may be in the cone of silence if you are directly above the router.

In a recent discussion, another user and I checked the amount of power an adapter with a mt7921au chipset was asking for:

$ lsusb -vv

Scoll down to your adapter and look for "Maxpower". What does it say?

bjlockie, 2023-06-17:
`MaxPower              100mA`

morrownr, 2023-06-19:
That is such a small amount of power. I'm using my Alfa ACH right now on my dev system as I am working on the driver:

MaxPower 800mA

Don't let your Pi see that or it will run away and hide.