morrownr/USB-WiFi #282: No MT7922u-based adapters after all this time??

Aqua1ung, 2023-06-18:
See subject line. What's the mystery?

Is everybody holding their breath for WiFi7 chipsets?

bjlockie, 2023-06-18:
They're probably seeing how 7921 sells and waiting for wifi 7.
I think the only improvement with 7922 is the channel width which is 
probably not useful in crowded areas.
I don't know why they can't just replace a 7921 with a 7922 in a current 
design but maybe 7922 chips cost too much still.

morrownr, 2023-06-18:
> What's the mystery?

Not really a mystery. Those involved in wifi consider usb wifi adapters to be a niche market. The mt7921e driver showed up in kernel 5.12. The mt7921u driver showed up in kernel 5.18, over a year later. We are just now starting to see a pickup in availability of the mt7922 chipset based cards. USB chipsets are almost always available 1-2 years after the card chipsets.

The information that I am aware of seems to indicate that we will get a usb version of the mt7922 chipset but we won't know for sure until we see it.

Aqua1ung, 2023-06-19:
> The information that I am aware of seems to indicate that we will get a usb version of the mt7922 chipset but we won't know for sure until we see it.

But ... but ... the mt7922**u** has been out for some time, right?

bjlockie, 2023-06-19:
mt7922 pci cards are not even common.

morrownr, 2023-06-19:
> But ... but ... the mt7922u has been out for some time, right?

I'm running a PCIe card with a mt7922 chipset and it uses the mt7921e driver... but PCIe is not USB. The USB version of the mt7922 may be in the hands of some adapter makers or it may not. I have no information one way or the other.

If I had to guess, it would be that we might see adapters early next year.

morrownr, 2023-06-19:
For what it is worth:

WiFi 6 USB Adapters: In chronological order, I think)

Realtek released the its first USB WiFi 6 (not WiFi 6e) adapter chipset and it was called the rtl8852/32au. The driver was provided as an out-of-kernel driver. All known adapters are multi-state for this chipset by the way. I started work on the driver with the intent of making it public with the other Realtek out-of-kernel drivers here at this site. It soon became clear that the driver was really really bad. I paused work pending a new release that would hopefully be better. That new release never came so there is no driver at this site from that chipset. It appears production of the chipset has been terminated. This very well could be another case where things don't work out well for Realtek and they stop supporting a product. This dropping support is not new from Realtek. My recommendation is for Linux users (or any users) to avoid adapters with this chipset as driver support is likely not going to ever be good and you simply do not need multi-state adapters.

Mediatek released its first USB WiFi 6e (6 GHz capable) adapter chipset and it is called the mt7921au. Mediatek never produced a WiFi 6 only chipset. They went directly to WiFi 6e. 

Realtek released the its second USB WiFi 6 (not WiFi 6e) adapter chipset and it was called the rtl8852/32bu. The driver was provided as an out-of-kernel driver. All known adapters are multi-state for this chipset by the way. I started work on the driver with the intent of making it public with the other Realtek out-of-kernel drivers here at this site. While the currently released driver is better than the driver for the previous generation, it has some problems. I have paused my efforts to make a driver available here at this site pending a new improved release from Realtek. My recommendation is for Linux users (or any users) to avoid adapters with this chipset as out-of-kernel drivers do not meet Linux Wireless Standards and will be problematic even if a better out-of-kernel driver is released in the short term. Additionally, it appears that all available adapters with this chipset are multi-state and Linux users simply do not need the problems that can come with that.

Realtek recently released the its first USB WiFi 6e (6 GHz capable) adapter chipset and it is called the rtl8852/32cu. The driver is provided as an out-of-kernel driver. Not much else is known at this point.

Mediatek may release its first 160 MHz channel width USB WiFi 6e (6 GHz capable adapter chipset at some point this year. It is called the mt7922u.

Summary: while Realtek beat Mediatek to the WiFi 6 usb market, Mediatek zoomed by and took a commanding lead with 6e and standards compliant drivers. We'll have to see where things go. Realtek's continued use of out-of-kernel drivers is troubling as out-of-kernel drivers simply do not lead to long term, stable performance. Code is already in the current Linux kernel to prevent WiFi 7 drivers that use the old Wireless Extensions from working. Most of the current in-kernel drivers that had Wireless Extensions code have had the code removed. So Realtek has been warned. There is hope as the Realtek rtw88 in-kernel driver now support several of the modern WiFi 5 dual band usb chipset such as the rtl8812bu.

Things are getting better in the USB WiFi adapter for Linux world. Let's hope it continues.