morrownr/USB-WiFi #283: MT7921u: Driver doesn't support chan-switch with channel contexts

ea4eoz, 2023-06-19:
Hello!

After using a Comfast CF-953AX for about 10 days I noticed disconnections from time to time.

Looking into syslog I found for every disconnection this:

kernel: driver doesn't support chan-switch with channel contexts
wpa_supplicant: CTRL-EVENT-DISCONNECTED bssid=a6:64:xx:xx:xx:xx reason=4 locally_generated=1

After a quick search in the kernel tree, this happens when the AP issues a frequency / channel change because of DFS (using 802.11h:  IEEE80211_HW_CHANCTX_STA_CSA) and the kernel driver does not handle it. The action in response is to disconnect the link.

So... in the current status of MT7921u driver is it does not handle DFS issues, so expect frequent link disconnections while working in the 5 GHz band. Keep it in mind if you need a rock solid connection to your AP.

Probably this must be added to the known issues list, because it can be a no-go for many users.

Tested using Debian Sid with kernel 6.3.7

ea4eoz, 2023-06-19:
Never mind. I just saw it is listed in the pinned issues.

Sorry for the inconveniences.

ea4eoz, 2023-06-19:
I noticed the DFS issue is listed in AP mode. I'm experiencing it in client mode, so the kernel driver does not support DFS at all, in any mode. This must be noted as it can be a serious problem for typical users.

morrownr, 2023-06-19:
> I noticed the DFS issue is listed in AP mode. I'm experiencing it in client mode, so the kernel driver does not support DFS at all, in any mode. This must be noted as it can be a serious problem for typical users.

Thanks for the report. The DFS problem in the pinned issue is a result of the support intentionally not being there. Same with the mt7612u and mt7610u chipsets. DFS for AP mode requires separate testing and FCC approvals and I suspect Mediatek management simply did not think it was a capability that needed to be in usb adapters. I am simply trying to get Mediatek's attention to the fact that 5 GHz AP mode DFS support is and will continue to be important and is feature we want.

Now, the issue you are reporting really needs its own separate bug report in the pinning issue and if we can get others to confirm it and tighten it down, that will help. I have the ability to test and will start doing so soon. I love a good bug hunt.

If we can tighten it down to something very repeatable on multiple distros, it will be time to report it to those who work these issues... or if we can develop a patch, we can submit the patch. I will say this, my main 5 GHz radio on my main wifi router is set to a DFS channel and I have not noticed this but I have not been specifically looking for it and I am likely using a different setup than you so work to be done.

@morrownr

morrownr, 2023-06-20:
I started the bug hunt on my end today.

My wifi router is a dual band Zyxel running OpenWRT 23.05 rc1. 5 GHz is running with channel 128 and 80 MHz channel width.

My client is on my main dev box running Ubuntu 23.04 x80_64. The adapter is a CF-951AX. First run of operf3:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.185 port 56756 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  66.6 MBytes   559 Mbits/sec    0   1.84 MBytes       
[  5]   1.00-2.00   sec  73.8 MBytes   619 Mbits/sec    0   1.84 MBytes       
[  5]   2.00-3.00   sec  72.5 MBytes   608 Mbits/sec    0   1.84 MBytes       
[  5]   3.00-4.00   sec  72.5 MBytes   608 Mbits/sec    0   1.84 MBytes       
[  5]   4.00-5.00   sec  73.8 MBytes   619 Mbits/sec    0   1.84 MBytes       
[  5]   5.00-6.00   sec  72.5 MBytes   608 Mbits/sec    0   1.84 MBytes       
[  5]   6.00-7.00   sec  73.8 MBytes   619 Mbits/sec    0   1.84 MBytes       
[  5]   7.00-8.00   sec  72.5 MBytes   608 Mbits/sec    0   1.84 MBytes       
[  5]   8.00-9.00   sec  70.0 MBytes   587 Mbits/sec    0   1.84 MBytes       
[  5]   9.00-10.00  sec  72.5 MBytes   608 Mbits/sec    0   1.84 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   720 MBytes   604 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   719 MBytes   602 Mbits/sec                  receiver

```

I looked around in dmesg and clean so far.

My understanding is that my router that is on a DFS channel needs to initiate a channel change and that should trigger a drop, right?

I'll just work as usual and if drops don't show up, I will start changing the chennal until hopefully I trigger a drop.

One possibly major difference in our client systems is that I am not running wpa_supplicant as Ubuntu changed over to IWD.

ea4eoz, 2023-06-21:
Yes, everything works fine until the AP decides to change the channel. At that point the connection drops, the syslog shows the "driver doesn't support chan-switch with channel contexts" message and it reconnects again in the new channel. It is just a matter of 2 - 3 seconds.

morrownr, 2023-06-21:
> Yes, everything works fine until the AP decides to change the channel. At that point the connection drops, the syslog shows the "driver doesn't support chan-switch with channel contexts" message and it reconnects again in the new channel. It is just a matter of 2 - 3 seconds.

Okay, good information.

So, when the AP changes channel, there is a drop but the drop only last 2-3 seconds before the connection is reestablished.

Question: You are only seeing this with DFS channels?

I have a little lab so I can, given enough time, test many adapters on many distros.

One thing I have to say is that there are a LOT of patches flowing into linux-wireless currently and have been for a while. The wireless stack is getting a lot of new features so it is possible that the stack has a new feature and the driver simply has not added the feature yet. I am actually amazed at how solid the mt7921u/e/s drivers are at this point given the very large size required for WiFi 6/6e support. Oh, I can also test a mt7922e client as I have a PCIe card based on the mt7922 chipset. It uses the same mt7921 driver as the mt7921 chipsets. It uses different firmware but the same driver.

ea4eoz, 2023-06-21:
> Question: You are only seeing this with DFS channels?

It is hard to answer this. Here where I live (Spain) only the first four channels are DFS Free (36, 40, 44 and 48) and they are so crowded we usually configure our APs in auto channel mode. The problem is channel jump can be a rare event (one in two or three days) or very very common (dozens in just an hour)

So channel changes can occur jumping into a DFS one, or jumping out of a DFS one.

I will keep an eye to track the channels my AP is using.

morrownr, 2023-06-21:
Yo intiendo todo. I understand.

We can report back here as we have additional information.

ea4eoz, 2023-06-22:
And while I was writting the previous message, I had another channel change, from 52 (DFS) to 36 (no DFS).

Same syslog sequence as you can see:

```
2023-06-22T18:29:29.712949+02:00 waterhole kernel: [18992.145011] wlxe0e1a9xxxxxx: driver doesn't support chan-switch with channel contexts
2023-06-22T18:29:29.893858+02:00 waterhole wpa_supplicant[676]: wlxe0e1a9xxxxxx: CTRL-EVENT-DISCONNECTED bssid=a6:64:xx:xx:xx:xx reason=4 locally_generated=1
2023-06-22T18:29:29.932674+02:00 waterhole wpa_supplicant[676]: wlxe0e1a9xxxxxx: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
2023-06-22T18:29:29.939286+02:00 waterhole NetworkManager[674]: <info>  [1687451369.9384] device (wlxe0e1a9xxxxxx): supplicant interface state: completed -> disconnected
2023-06-22T18:29:29.999804+02:00 waterhole NetworkManager[674]: <info>  [1687451369.9994] device (wlxe0e1a9xxxxxx): supplicant interface state: disconnected -> scanning
2023-06-22T18:29:31.072845+02:00 waterhole wpa_supplicant[676]: wlxe0e1a9xxxxxx: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
2023-06-22T18:29:31.366126+02:00 waterhole wpa_supplicant[676]: wlxe0e1a9xxxxxx: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
2023-06-22T18:29:33.671828+02:00 waterhole wpa_supplicant[676]: wlxe0e1a9xxxxxx: SME: Trying to authenticate with a6:64:xx:xx:xx:xx (SSID='Wireless' freq=5180 MHz)
2023-06-22T18:29:33.672902+02:00 waterhole kernel: [18996.105029] wlxe0e1a9xxxxxx: authenticate with a6:64:xx:xx:xx:xx
2023-06-22T18:29:34.184962+02:00 waterhole kernel: [18996.616730] wlxe0e1a9xxxxxx: send auth to a6:64:xx:xx:xx:xx (try 1/3)
2023-06-22T18:29:34.187128+02:00 waterhole NetworkManager[674]: <info>  [1687451374.1862] device (wlxe0e1a9xxxxxx): supplicant interface state: scanning -> authenticating
2023-06-22T18:29:34.188902+02:00 waterhole kernel: [18996.621383] wlxe0e1a9xxxxxx: authenticated
2023-06-22T18:29:34.190418+02:00 waterhole wpa_supplicant[676]: wlxe0e1a9xxxxxx: Trying to associate with a6:64:xx:xx:xx:xx (SSID='Wireless' freq=5180 MHz)
2023-06-22T18:29:34.196079+02:00 waterhole NetworkManager[674]: <info>  [1687451374.1959] device (wlxe0e1a9xxxxxx): supplicant interface state: authenticating -> associating
2023-06-22T18:29:34.196921+02:00 waterhole kernel: [18996.626607] wlxe0e1a9xxxxxx: associate with a6:64:xx:xx:xx:xx (try 1/3)
2023-06-22T18:29:34.204931+02:00 waterhole kernel: [18996.636098] wlxe0e1a9xxxxxx: RX AssocResp from a6:64:xx:xx:xx:xx (capab=0x511 status=0 aid=81)
2023-06-22T18:29:34.232945+02:00 waterhole kernel: [18996.666164] wlxe0e1a9xxxxxx: associated
2023-06-22T18:29:34.232970+02:00 waterhole kernel: [18996.666500] wlxe0e1a9xxxxxx: Limiting TX power to 23 (23 - 0) dBm as advertised by a6:64:xx:xx:xx:xx
2023-06-22T18:29:34.390878+02:00 waterhole wpa_supplicant[676]: wlxe0e1a9xxxxxx: Associated with a6:64:xx:xx:xx:xx
2023-06-22T18:29:34.391123+02:00 waterhole wpa_supplicant[676]: wlxe0e1a9xxxxxx: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
2023-06-22T18:29:34.391373+02:00 waterhole wpa_supplicant[676]: wlxe0e1a9xxxxxx: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=ES
2023-06-22T18:29:34.397233+02:00 waterhole NetworkManager[674]: <info>  [1687451374.3970] device (wlxe0e1a9xxxxxx): supplicant interface state: associating -> associated
2023-06-22T18:29:35.297810+02:00 waterhole NetworkManager[674]: <info>  [1687451375.2969] device (wlxe0e1a9xxxxxx): supplicant interface state: associated -> 4way_handshake
2023-06-22T18:29:35.381020+02:00 waterhole wpa_supplicant[676]: wlxe0e1a9xxxxxx: WPA: Key negotiation completed with a6:64:xx:xx:xx:xx [PTK=CCMP GTK=CCMP]
2023-06-22T18:29:35.381255+02:00 waterhole wpa_supplicant[676]: wlxe0e1a9xxxxxx: CTRL-EVENT-CONNECTED - Connection to a6:64:xx:xx:xx:xx completed [id=0 id_str=]
```

And this time reconnection was done in 6 seconds.

ea4eoz, 2023-06-22:
And now from channel 36 (no DFS) to channel 100 (DFS).

Same syslog sequence, so I will not repeat it here.

No matter what channels are involved (DFS or not DFS) , there is a connection drop every time the AP changes the channel.

morrownr, 2023-06-23:
I was not seeing anything here for the first 2-3 days of testing most likely because my AP was not seeing any need to switch channels. Therefore I decided to manually switch channels. I run OpenWRT. I did a few times using DFS and non-DFS channels with a consistent result.:

```
[  5]   8.00-9.00   sec  71.2 MBytes   598 Mbits/sec    0   1.50 MBytes       
[  5]   9.00-10.00  sec  73.8 MBytes   619 Mbits/sec    0   1.59 MBytes       
[  5]  10.00-11.00  sec  71.2 MBytes   598 Mbits/sec    0   1.67 MBytes       
[  5]  11.00-12.00  sec  73.8 MBytes   619 Mbits/sec    0   1.72 MBytes       
[  5]  12.00-13.00  sec  60.0 MBytes   503 Mbits/sec    0   1.75 MBytes       
[  5]  13.00-14.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  5]  14.00-15.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  5]  15.00-16.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  5]  16.00-17.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  5]  17.00-18.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes 
```

The connection would drop but would not recover. The only way to get the connection going again was to cycle wifi off and back on. Now I need to look at the logs. This does seem to indicate the problem is with my client. FYI: My client system currently has a PCIe adapters with a mt7922e chipset and a USB WiFi adapter with the mt7921au chipset. Both showed exactly the same result... which is probably what we should expect since the driver for both share a lot of code.

The key line you are seeing in your log is? : driver doesn't support chan-switch with channel contexts

I think some research is in order. Rsearch into that message and research into what the expected outcome should be.

ea4eoz, 2023-06-23:
> The key line you are seeing in your log is? : driver doesn't support chan-switch with channel contexts

Yes, a quick search in syslog files must be enough to see is the problem is present or not.

The string is present in net/mac80211/mlme.c, for example:

https://github.com/torvalds/linux/blob/master/net/mac80211/mlme.c#L1963

```
if (local->use_chanctx &&
    !ieee80211_hw_check(&local->hw, CHANCTX_STA_CSA)) {
	sdata_info(sdata,
		   "driver doesn't support chan-switch with channel contexts\n");
	goto drop_connection;
}
```

and as you can see, the action after syslogging the message is to drop the connection.

Using kernel 6.3.7 and loading these firmware files:

```
kernel: [ 1304.469711] usb 2-4: new SuperSpeed USB device number 6 using xhci_hcd
kernel: [ 1304.497670] usb 2-4: New USB device found, idVendor=3574, idProduct=6211, bcdDevice= 1.00
kernel: [ 1304.497677] usb 2-4: New USB device strings: Mfr=2, Product=3, SerialNumber=4
kernel: [ 1304.497679] usb 2-4: Product: Wireless_Device
kernel: [ 1304.497680] usb 2-4: Manufacturer: MediaTek Inc.
kernel: [ 1304.497681] usb 2-4: SerialNumber: 000000000
kernel: [ 1304.503517] mt7921u 2-4:1.0: firmware: direct-loading firmware mediatek/WIFI_RAM_CODE_MT7961_1.bin
kernel: [ 1304.630234] usb 2-4: reset SuperSpeed USB device number 6 using xhci_hcd
kernel: [ 1304.788356] mt7921u 2-4:1.0: firmware: direct-loading firmware mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
kernel: [ 1304.788375] mt7921u 2-4:1.0: HW/SW Version: 0x8a108a10, Build Time: 20230302150916a
kernel: [ 1304.798222] mt7921u 2-4:1.0: firmware: direct-loading firmware mediatek/WIFI_RAM_CODE_MT7961_1.bin
kernel: [ 1304.798241] mt7921u 2-4:1.0: WM Firmware Version: ____010000, Build Time: 20230302150956
```

mike-silva, 2023-07-19:
I've been seeing the same thing on my Raspberry Pi running Ubuntu Lunar for months.

In my case, the RPI is running headless, so when this occurs, I must go connect via ethernet to restart WiFi or power cycle. A real pain.

Jul 18 06:58:23 Vancouver systemd-networkd[1033]: wlxe0e1a9368b21: Lost carrier
Jul 18 06:58:23 Vancouver kernel: wlxe0e1a9368b21: driver doesn't support chan-switch with channel contexts

I'm located in the Bay Area in California, so I imagine there are quite a few sources of DFS strikes nearby, ergo this becomes a tedious daily, or sometimes multiple times a day occurrence.

Has anyone reached out to the mediatek mailing list to ask them to fix this?

morrownr, 2023-07-19:
Hi @mike-silva 

> Has anyone reached out to the mediatek mailing list to ask them to fix this?

Not that I am aware of. I monitor linux-wireless and not the Mediatek list. Obviously this needs to get some attention. Maybe you guys can collaborate on a good [bug report] and post it accordingly.

There is the downstead mt76 repo:

https://github.com/openwrt/mt76

No matter where this is reported, include a link to this thread for background.

I'm not sure that the 2 links provided about the 7915 driver will be much help.

mike-silva, 2023-07-19:
Thanks @morrownr for your reply and your work here tracking these devices on Github.

I'm about to post to the mediatek list, and for sure will reference the excellent reports here, so they can see it's a common problem that needs addressing.

If anyone wants a handy link to peruse an archive of the mediatek specific mailing list for linux-wireless, it's here:
https://lore.kernel.org/linux-mediatek/

If you search there for the terms "mt76 dfs" or "mt76 radar", you'll see that there are several threads about fixing such issues on other chipsets...just not the ones we care about here.

And, of course, it probably wouldn't hurt if folks here keep an eye on my report on the list, and pipe up if there is additional relevant info or corroboration needed. Thanks!

morrownr, 2023-07-19:
Mike,

I saw your bug report on the Mediatek list this morning. Hopefully this is fixed soon. I also run a couple of Pi's headless and they connect to a DFS channel. I think the difference is that I live in an area where DFS switching is just not triggered on the channel I use. Keep us posted on anything you find out.

@morrownr

mike-silva, 2023-07-19:
Yah, I'm a bit worried, because I seemed to have some bad luck in my bug report timing. If you noticed, last night there was like 80 or so patches that hit the list. Hopefully, my report doesn't get lost in the shuffle.

No response as of now. I'll give them until the weekend, since things seem busy, then bump it if I haven't heard back.

Or, if folks here want to pipe up with their own logs and corroboration, I'm guessing that can't hurt. Just sayin'.

mike-silva, 2023-07-22:
Unfortunately, I've gotten no response to my mailing list bug report. But, I just CC'd the mediatek development team and linux wifi lead maintainer directly, as their emails are listed on the linux-wireless documentation site. Hopefully, this will bring some visibility.

To be fair, the list has been incredibly busy this week, for obvious reasons with the new in development linux kernel being open to changes.

morrownr, 2023-07-22:
> To be fair, the list has been incredibly busy this week...

They also have numerous new drivers for WiFi 7 in progress as if they need more to do. Remember to be very kind and not push too hard so that you don't end up in the penalty box.

mike-silva, 2023-07-22:
Yup, noticed that too. Plus, a bunch of SOC work for Single Board Computers.

I'm a software/network engineer. I'm pretty good about pinging, without being a dick. :-) But, I'm persistent. And, I'll hit up Comfast at some point too. Vendor requests for support from their chipset makers pull a lot more weight.

morrownr, 2023-07-22:
> It's an egregious enough bug. 

Yes it is and speaking of egregious bugs... somebody made the decision to backport something from kernel 6.5 and patch it into everything back to and including kernel 6.1.39 so I am trying to patch six Realtek drivers in a way that results in the least damage. There is no good fix.

mike-silva, 2023-07-23:
Just to be clear, you know that Realtek sold a few of their chipsets to mediatek a year or two ago, right?

Thus, some of the upheaval in the code base. It seems like mediatek is trying to clean up some technical debt, but they definitely carried over old Realtek bugs. And, when they then derived new chipsets from the IP they bought, carried it forward to those derivatives. (And to make the situation more comical, one is now suing the other for monopoly business practices, squeezing them out of vendor contracts.)

My main point though, is if there are silly Realtek issues, you might want to hit up that mailing list/dev team instead. If they are still true Realtek chipsets.

bjlockie, 2023-07-24:
Maybe Mediatek can counter sue Realtek for making sh*tty drivers. :-)

mike-silva, 2023-07-24:
Sorry to gum up the git with blather, but I have to say that your comment just made me spit-take my coffee.

morrownr, 2023-07-24:
> Sorry to gum up the git with blather,

I'm all for it. That is what makes life worth living. Leave it to a Canadian to stir things up. I spent a lot of time in Canada earlier in my life and I grew fond of the Canadian sense of humor.

Now back to business,,,

https://www.reuters.com/legal/realtek-sues-semiconductor-rival-mediatek-over-patent-bounty-agreement-2023-06-06/

That link is just to add to the discussion. A quick google and you can get numerous articles about the same topic.

> you know that Realtek sold a few of their chipsets to Mediatek a year or two ago, right?

I have not seen anything to verify that, I have suspected some things having to do with anti competitive business practices were happening between the two companies but that was not one of them. Mediatek is somewhat dominant in wifi for TV's and that seems to be what started this problem but then Realtek is dominant in usb wifi and I suspect Realtek is trying to lock down buyers of its usb chipsets in exclusive contracts, which would also be illegal in the US.

Mediatek got the IP for its wifi from Ralink when it bought them in 2013 and that includes up to the mt7612u chipset. Mediatek beat Realtek to the usb wifi market for 6 GHz by around 2 years. Search for Mediatek and Samsung. 

These cases take forever to go to court, if they go to court so the cat fight will likely continue for a few years. But I do like the idea of suing Realtek for their sh*tty drivers. We would have standing in this case but I don't think Mediatek would.

mike-silva, 2023-07-24:
Ah crap. I guess I misremembered and confused Ralink and Realtek. I'll double check tonight. Sorry. On second thought, what am I saying, if anyone is going to know who acquired what chipsets, it's gonna be you @morrownr.

Ars also did a good piece on the anti-competitive practice suit:
https://arstechnica.com/tech-policy/2023/06/smart-tv-industry-rocked-by-alleged-patent-conspiracy-from-chip-maker/

And, hosts a PDF of the legal filing:
https://cdn.arstechnica.net/wp-content/uploads/2023/06/Realtek-v-MediaTek-5-23-cv-02774-SK-6-6-2023.pdf

mike-silva, 2023-07-24:
I haven't had a lot of time to look yet, but these 2 patches from Mediatek today look promising.

https://lore.kernel.org/linux-mediatek/8fd42ac8e1c97246f6e65225a14fc8a029ac3aaa.1690232804.git.objelf@gmail.com/T/#t

Cursory take on the run: kinda hard to support changing channel switch with channel contexts, if your driver can't see that your chipset supports it.

morrownr, 2023-07-25:
> Ah crap. I guess I misremembered and confused Ralink and Realtek.

Wait until you are my age. The misremembering happens more frequently.

> Cursory take on the run: kinda hard to support changing channel switch with channel contexts, if your driver can't see that your chipset supports it.

We can look forward to seeing where this goes and what it fixes.

mike-silva, 2023-07-27:
Those two patches cleared testing by a member of the Chrome OS team. I'm expecting them to be released in the next few days, at which point I'll start testing them. I'll add a comment here, letting everyone know when the binaries are available.

mike-silva, 2023-08-04:
Well, nothing has been released yet. To be frank, I'm not sure exactly what the linux firmware team's release management methodology/schedule is. In the past, I've seen patches hit the mediatek mailing list, then hit the binary firmware releases in git.kernel.org in 2-3 days.

I'll ping back here when it happens.

morrownr, 2026-01-18:
@a-gavin 

Thanks for this report. Unfortunately these patches and problems with the support in mac80211 broke monitor mode support badly with mt7921/22.

morrownr, 2026-01-18:
@a-gavin 

> Edit: I see discussion in this thread

I had reports coming into several issues so I establish this new thread to try to consolidate reports.

>  but that should be fixed for kernels 6.18.2 and newer (changelog here).

My last testing was with kernel 6.18.1 so let me fire up my compiler and see what happens.

> I realize this is off thread, but is there another issue that I'm not aware of?

I was posting results in the new Monitor Mode issue you posted above. mt7925 was doing great. mt7921, not so much. We were testing injection and capabilities beyond capture. It was bad. I'll see what happens with 6.18.6.

a-gavin, 2026-01-19:
Gotcha, I don’t regularly test frame injection, so I can’t speak to that.

ea4eoz, 2026-08-27:
Three years later I tried again my MT-7921u dongles and they still do not support channel switch in the 5 GHz band.

Running Debian 13, kernel 6.12.105+deb13-amd64, This was dmesg output:

```
[   10.893781] mt7921u 8-4:1.0: HW/SW Version: 0x8a108a10, Build Time: 20241106151007a
[   11.151439] mt7921u 8-4:1.0: WM Firmware Version: ____010000, Build Time: 20241106151045

[ 5771.438906] wlan0: driver doesn't support chan-switch with channel contexts
[ 5771.943831] wlan0: authenticate with a0:64:8f:dc:b1:74 (local address=e0:e1:a9:38:98:03)
[ 5772.063489] wlan0: send auth to a0:64:8f:dc:b1:74 (try 1/3)
[ 5772.067645] wlan0: authenticated
[ 5772.108566] wlan0: associate with a0:64:8f:dc:b1:74 (try 1/3)
[ 5772.115030] wlan0: RX AssocResp from a0:64:8f:dc:b1:74 (capab=0x1431 status=0 aid=2)
[ 5772.131268] wlan0: associated
[ 5799.003937] wlan0: AP is in CSA process, reject auth
[ 5804.104994] wlan0: deauthenticating from a0:64:8f:dc:b1:74 by local choice (Reason: 2=PREV_AUTH_NOT_VALID)
```

After this, Network Manager asked me for the wifi password.

I'm very surprised that such a basic feature hasn't been implemented yet, considering that WiFi 6e adapters are becoming obsolete hardware with the arrival of WiFi 7... or maybe is a specific Debian issue?

I see there are newer versions of firmware-mediatek in backports and in unstable. Does it worth to take the risk?

Any help will be appreciated.

morrownr, 2026-08-27:
Hi @ea4eoz 

Earlier this year we added a repo to help solve issues like this:

https://github.com/morrownr/mt76

Can I get you to first install the driver at this repo. It will modernize your drivrer from the one in kernel 6.12 to the one in kernel 7.3. If you then still see the problem, please open an issue at that repo and we will work it.

The installation and uninstallation is very easy.

a-gavin, 2026-08-27:
> Three years later I tried again my MT-7921u dongles and they still do not support channel switch in the 5 GHz band.

@ea4eoz mt7921/mt7921u Linux drivers have supported channel switching in client/STA mode since Linux v6.14 (patch [here](https://lore.kernel.org/linux-wireless/20241107061440.6545-1-mingyen.hsieh@mediatek.com/), v6.14 [released March 2025](https://lore.kernel.org/lkml/CAHk-=wg7TO09Si5tTPyhdrLLvyYtVmCf+GGN4kVJ0=Xk=5TE3g@mail.gmail.com/)). The next most recent LTS Linux release after v6.12 is v6.18 which was released [November 2025](https://lore.kernel.org/lkml/CAHk-=whnC+hRftevTLeVs3tyyqwn+7un=jUES2-WX+pZhDdKNw@mail.gmail.com/).

As a user you choose to run Debian, which is a great choice for many reasons. However, Debian is known for ensuring stability and quality rather than "chasing" the latest and greatest (whereas Fedora is a generally tracks latest much more closely). While I cannot speak to why Debian chose v6.12 for their newest release (Debian 13 in July this year), v6.18 has been available for some time.

To enable channel switching with your mt7921u radio, I encourage you to follow @morrownr's suggestion to install a more recent driver, assuming you'd like to stick with Debian 13.

morrownr, 2026-08-28:
@ea4eoz 

@a-gavin speaks words of wisdom. Debian is a great distro. In fact, my main dev box runs the testing version of Debian 14. That allows me to stay current with updated wireless things. I also run other distros on test machines. The distro that seems to keep firmware and other parts of Linux very close to the bleeding edge while maintaining a good level of stability is Ubuntu. Not to say others don't as well I can't stay up on everything.

The mt76 repo that I pointed you to is not something one person put together to work on one problem. It is the entire mt76 series of drivers. The installer compiles and installs all drivers except for the ones that are designed for routers. We keep the code such that it will install and work on kernels 6.12 through 7.2. The actual code level in the drivers is already up to kernel 7.3. Some really talented devs work at that repo and we upstream all fixes. We have upstreamed dozens of fixes and enhancements so far this year. 

@morrownr

ea4eoz, 2026-08-28:
Yes, I know, I know... Debian is obsolete even on the launching day. That's why I asked if it is a Debian issue... and it is.

I will try kernel 7.1.8 from backports and the most recent firmware-mediatek, and if it does not work, I'll try Morrow's drivers.

Thank you!