morrownr/USB-WiFi #284: Kernel 6.2 Realtek RTL8814AU

birdlinux, 2023-06-22:
Could you add support for kernel 6.2 for the RTL8814AU chipset? i cant use my laptop anymore because of that very please!!! its a need i cant do any work at all anymore nor browse the net nor anything because it simply doesn't work it detects the wifi networks but it wont connect! and i sadly cant use a LAN cable since im way to far away and i always travel

morrownr, 2023-06-22:
Hi @birdlinux 

> Could you add support for kernel 6.2 for the RTL8814AU chipset? 

That was added a long time ago. I've been running the driver with kernel 6.4.

Since I can't update the driver on your harddrive, maybe you need to updare? I've tried to make it easy:

$ git pull
$ sudo sh install-driver.sh

You don't have to remove the current driver as the install script will detect and remove it.

Hopefully this helps.

birdlinux, 2023-06-22:
This sadly didnt help i dont know why i did update with git pull but sadly nothing happened as it said it was up to date! I provided a small video on the issue to try to identify it.

https://www.youtube.com/watch?v=sRMtaB8z6fA

morrownr, 2023-06-22:
Post the results of the following:

$ iw dev

What distro are you using? 

Have you tried rebooting?

birdlinux, 2023-06-22:
[image]
yeah i have tried rebooting but nothing happened

morrownr, 2023-06-22:
Try running the installation script again and this time capture the on screen log and post it.

I need more info.

I'm not seeing anything so far that indicated the driver is causing a problem. The driver is installed and an interface is being provided. It appears scanning for APs is working.

Are you using Ubuntu?

birdlinux, 2023-06-22:
I use pop OS

[image]

https://www.youtube.com/watch?v=FgzO5OYxsFw

morrownr, 2023-06-22:
I can't see a problem at this point. I'm on the road right now so my ability to do much more will be limited for a few days.

My best guess: It could be something in Pop OS. I just checked and the 8814au repo is averaging around 150 clones day.

bjlockie, 2023-06-22:
Does the 8814au have an in kernel driver now?
Maybe something needs blacklisting?

birdlinux, 2023-06-23:
@morrownr 

I checked again its very weird it worked on previous Pop OS installation, I reinstalled and suddenly it doesn't work anymore. It seems that SSID scanning works it just wont connect? It's very very weird i don't know what to do at this point

morrownr, 2023-06-23:
I've been on the road but am back now. I will try to set aside some time to do some testing today or maybe this weekend. I have seen nothing that tells me it is a problem with the driver so far. It really appears to be something in Pop.

Have you looked around the Pop forums to see if anyone else is having this problem?

birdlinux, 2023-06-24:
yeah i have looked but i sadly havent found anything at all

birdlinux, 2023-06-25:
Hey, @morrownr 

Any updates yet?

birdlinux, 2023-06-26:
[image]

this is after i removed everything i reinstalled with the installation steps but nothing happened 

<hr>

after installation
[image]

morrownr, 2023-06-26:
What is the result of:

$ sudo mokutil --sb-state

birdlinux, 2023-06-27:
[sudo] password for xor:
SecureBoot disabled
Platform is in Setup Mode

morrownr, 2023-06-28:
Hmmm... still no hints.

Try running the following and posting the output:

$ sudo dmesg | grep wl
$ sudo lsmod | grep 8814

birdlinux, 2023-06-28:
```
[    7.498567] iwlwifi 0000:00:14.3: api flags index 2 larger than supported by driver
[    7.498585] iwlwifi 0000:00:14.3: TLV_FW_FSEQ_VERSION: FSEQ Version: 89.3.35.37
[    7.499074] iwlwifi 0000:00:14.3: loaded firmware version 72.daa05125.0 QuZ-a0-hr-b0-72.ucode op_mode iwlmvm
[    7.636091] iwlwifi 0000:00:14.3: Detected Intel(R) Wi-Fi 6 AX201 160MHz, REV=0x351
[    7.760643] iwlwifi 0000:00:14.3: Detected RF HR B5, rfid=0x10a100
[    7.826285] iwlwifi 0000:00:14.3: base HW address: a0:e7:0b:68:a8:c6
[    7.855056] iwlwifi 0000:00:14.3 wlp0s20f3: renamed from wlan0
[   11.867942] wlp0s20f3: authenticate with 38:10:d5:5a:89:fd
[   11.867973] wlp0s20f3: 80 MHz not supported, disabling VHT
[   11.870329] wlp0s20f3: send auth to 38:10:d5:5a:89:fd (try 1/3)
[   11.929793] wlp0s20f3: authenticated
[   11.932996] wlp0s20f3: associate with 38:10:d5:5a:89:fd (try 1/3)
[   11.938885] wlp0s20f3: RX AssocResp from 38:10:d5:5a:89:fd (capab=0x1431 status=0 aid=8)
[   11.946835] wlp0s20f3: associated
[   12.025721] wlp0s20f3: Limiting TX power to 20 (20 - 0) dBm as advertised by 38:10:d5:5a:89:fd
[   12.059732] IPv6: ADDRCONF(NETDEV_CHANGE): wlp0s20f3: link becomes ready
[   92.633112] RTW: rtw_ndev_init(wlan0) if1 mac_addr=00:c0:ca:ae:5d:86
[   92.731328] rtl8814au 1-4:1.0 wlx00c0caae5d86: renamed from wlan0
[   93.211098] RTW: wlx00c0caae5d86- hw port(0) mac_addr =00:c0:ca:ae:5d:86
[   93.224985] IPv6: ADDRCONF(NETDEV_CHANGE): wlx00c0caae5d86: link becomes ready
[  246.652184] wlp0s20f3: deauthenticating from 38:10:d5:5a:89:fd by local choice (Reason: 3=DEAUTH_LEAVING)
[  246.882135] wlp0s20f3: authenticate with 38:10:d5:5a:89:fe
[  246.886590] wlp0s20f3: send auth to 38:10:d5:5a:89:fe (try 1/3)
[  246.915429] wlp0s20f3: authenticated
[  246.917007] wlp0s20f3: associate with 38:10:d5:5a:89:fe (try 1/3)
[  246.919676] wlp0s20f3: RX AssocResp from 38:10:d5:5a:89:fe (capab=0x1511 status=0 aid=1)
[  246.927040] wlp0s20f3: associated
[  246.935418] wlp0s20f3: Limiting TX power to 20 (23 - 3) dBm as advertised by 38:10:d5:5a:89:fe
[  246.945894] IPv6: ADDRCONF(NETDEV_CHANGE): wlp0s20f3: link becomes ready
[  285.512770] wlp0s20f3: deauthenticating from 38:10:d5:5a:89:fe by local choice (Reason: 3=DEAUTH_LEAVING)
[  288.392320] wlp0s20f3: authenticate with 38:10:d5:5a:89:fe
[  288.400973] wlp0s20f3: send auth to 38:10:d5:5a:89:fe (try 1/3)
[  288.430233] wlp0s20f3: authenticated
[  288.434554] wlp0s20f3: associate with 38:10:d5:5a:89:fe (try 1/3)
[  288.438022] wlp0s20f3: RX AssocResp from 38:10:d5:5a:89:fe (capab=0x1511 status=0 aid=1)
[  288.447587] wlp0s20f3: associated
[  288.519272] IPv6: ADDRCONF(NETDEV_CHANGE): wlp0s20f3: link becomes ready
[  288.519295] wlp0s20f3: Limiting TX power to 20 (23 - 3) dBm as advertised by 38:10:d5:5a:89:fe
[  294.636511] wlp0s20f3: deauthenticating from 38:10:d5:5a:89:fe by local choice (Reason: 3=DEAUTH_LEAVING)
[  299.412732] wlp0s20f3: authenticate with 38:10:d5:5a:89:fd
[  299.412750] wlp0s20f3: 80 MHz not supported, disabling VHT
[  299.415125] wlp0s20f3: send auth to 38:10:d5:5a:89:fd (try 1/3)
[  299.476552] wlp0s20f3: authenticated
[  299.482182] wlp0s20f3: associate with 38:10:d5:5a:89:fd (try 1/3)
[  299.496645] wlp0s20f3: RX AssocResp from 38:10:d5:5a:89:fd (capab=0x1431 status=0 aid=8)
[  299.509167] wlp0s20f3: associated
[  299.567234] wlp0s20f3: Limiting TX power to 20 (20 - 0) dBm as advertised by 38:10:d5:5a:89:fd
[  299.595278] IPv6: ADDRCONF(NETDEV_CHANGE): wlp0s20f3: link becomes ready
[  304.549514] wlp0s20f3: deauthenticating from 38:10:d5:5a:89:fd by local choice (Reason: 3=DEAUTH_LEAVING)
[  307.839417] wlp0s20f3: authenticate with 38:10:d5:5a:89:fe
[  307.845067] wlp0s20f3: send auth to 38:10:d5:5a:89:fe (try 1/3)
[  307.874251] wlp0s20f3: authenticated
[  307.878223] wlp0s20f3: associate with 38:10:d5:5a:89:fe (try 1/3)
[  307.881465] wlp0s20f3: RX AssocResp from 38:10:d5:5a:89:fe (capab=0x1511 status=0 aid=1)
[  307.886789] wlp0s20f3: associated
[  307.958481] IPv6: ADDRCONF(NETDEV_CHANGE): wlp0s20f3: link becomes ready
[  308.070032] wlp0s20f3: Limiting TX power to 20 (23 - 3) dBm as advertised by 38:10:d5:5a:89:fe
[  313.911549] wlp0s20f3: Connection to AP 38:10:d5:5a:89:fe lost
[  320.932219] wlp0s20f3: authenticate with 38:10:d5:5a:89:fe
[  320.942457] wlp0s20f3: send auth to 38:10:d5:5a:89:fe (try 1/3)
[  320.972417] wlp0s20f3: authenticated
[  320.974239] wlp0s20f3: associate with 38:10:d5:5a:89:fe (try 1/3)
[  320.977572] wlp0s20f3: RX AssocResp from 38:10:d5:5a:89:fe (capab=0x1511 status=0 aid=1)
[  320.985424] wlp0s20f3: associated
[  321.792419] wlp0s20f3: Limiting TX power to 20 (23 - 3) dBm as advertised by 38:10:d5:5a:89:fe
[  337.064404] wlp0s20f3: Connection to AP 38:10:d5:5a:89:fe lost
[  342.780280] wlp0s20f3: authenticate with 38:10:d5:5a:89:fe
[  342.787625] wlp0s20f3: send auth to 38:10:d5:5a:89:fe (try 1/3)
[  342.816470] wlp0s20f3: authenticated
[... 530 lines cut ...]
[184418.736343] wlp0s20f3: authenticate with 38:10:d5:5a:89:fd
[184418.736376] wlp0s20f3: 80 MHz not supported, disabling VHT
[184418.741204] wlp0s20f3: send auth to 38:10:d5:5a:89:fd (try 1/3)
[184418.769648] wlp0s20f3: authenticated
[184418.772293] wlp0s20f3: associate with 38:10:d5:5a:89:fd (try 1/3)
[184418.779975] wlp0s20f3: RX ReassocResp from 38:10:d5:5a:89:fd (capab=0x1431 status=0 aid=1)
[184418.786963] wlp0s20f3: associated
[184418.892707] wlp0s20f3: disconnect from AP 38:10:d5:5a:89:fd for new auth to 38:10:d5:5a:89:fe
[... cut ...]
[184419.100277] wlp0s20f3: send auth to 38:10:d5:5a:89:fe (try 2/3)
[184419.102006] wlp0s20f3: authenticated
[184419.104283] wlp0s20f3: associate with 38:10:d5:5a:89:fe (try 1/3)
[184419.106691] wlp0s20f3: RX ReassocResp from 38:10:d5:5a:89:fe (capab=0x1511 status=0 aid=2)
[184419.111251] wlp0s20f3: associated
[184419.139422] wlp0s20f3: Limiting TX power to 20 (23 - 3) dBm as advertised by 38:10:d5:5a:89:fe
```

second

```
8814au               3076096  0
cfg80211             1241088  5 8814au,iwlmvm,iwlwifi,mac80211,88XXau

```

morrownr, 2023-06-28:
Okay, things are becoming clearer.

> cfg80211             1241088  5 8814au,iwlmvm,iwlwifi,mac80211,88XXau

That line tells me you have 2 usb wifi adapters in your system (or 1 usb adapter and one internal).

The log tells me there are a lot of problems with the adapter that uses the 88XXau driver.

The log and your earlier post of the results of $ iw dev tell me that the 8814au driver is correctly loading and supports the wlx00c0caae5d86 wifi interface. I see no evidence that this interface has been connected to an AP.

The log shows a lot problems with the wlp0s20f3 interface that is supported by the **XXau driver. This interface is connected  as shown with $ iw dev but the log shows it trying to connect but mostly failing.

So, the problem is not with this driver. This driver is doing its job. It just is not connected to an AP. However, the other wifi driver is raising hell about something and that probably has to be fixed before you are going to make much progress.

birdlinux, 2023-06-29:
oof so what should i do now i got no idea to be honest just stuck

bjlockie, 2023-06-29:
>The log tells me there are a lot of problems with the adapter that uses 
> the 88XXau driver.

.Maybe blacklist the 88XXau driver?

morrownr, 2023-06-29:
> oof so what should i do now i got no idea to be honest just stuck

The best approach depends on you. If it were me, I'd open up the laptop and remove the wifi card. But that is me and I have many years of experience building and working on computers. I might even replace it with a card that uses a Mediatek or Intel chipset if I need the internal wifi capability.

However, I like @bjlockie 's idea about blacklisting the driver that is causing a mess in your log. See if this helps:

https://linuxconfig.org/how-to-blacklist-a-module-on-ubuntu-debian-linux

Blacklisting shuts down the driver for unneeded hardware.

If you have problems with any of the details, let us know.

morrownr, 2023-07-01:
@birdlinux 

Any progress?

birdlinux, 2023-07-16:
Nah man not at all sadly :/

morrownr, 2023-07-16:
Were you able to blacklist the other wifi driver?

Have you used "Forget connection" on any previous connections?

morrownr, 2023-07-17:
> Can other wifi adapter work ?

Probably so. Realtek out-of-kernel drivers are problematic and have been for a long time. 

Here but here is the url to the README that contains the site Main Menu:

https://github.com/morrownr/USB-WiFi

Scroll down to see the Main Menu. Recommend you read Menu Item 1 for background information and then Menu Item 2, The Plug and Play List gives information and links to 50-60 adapters. The adapters in this list meet the following criteria:

1. They use in-kernel drivers. That means the adapters are Plug and Play. You do not have to find and compile a driver. Plus the in-kernel drivers are well maintained and are Linux Wireless Standards compliant.
2. They are all single-function. That is, they are wifi adapters and do not support bluetooth. You do not want bluetooth in your wifi adapter as it can limit wifi speeds and be a problem.
3. They are all single-state in that they do not have a Windows driver onboard. This is something else that can cause problems.

In other words, The Plug and Play List is filtered down to adapters that are much more problem free on Linux. You still have to make a decision regarding what best meets your needs within your budget so once you take a look and decide on 3-4 to ask about, please start a new issue and post what things like long range, short range, cheap, single band, dual band or tri band you need and we can give you some input.

@morrownr

birdlinux, 2023-10-17:
Hey, i finally had time to work it. I had uni etc. I got it work! It seems like i had some other rtl drivers preinstalled and that broke the support.

Kind regards
Bird

closed 2023-10-17 by birdlinux

manuelpaulo, 2024-01-30:
Here working and widely used. Why not adding it to kernel?

https://github.com/aircrack-ng/rtl8812au

morrownr, 2024-01-30:
Hi @manuelpaulo 

Hola, como esta usted hoy?

> Here working and widely used.

I am aware of the driver that you posted the link for. It is maintained primarily to support Aircrack-ng and it is based on rather old code. For a more general purpose version of this driver that is based on much more modern source code:

https://github.com/morrownr/8812au-20210629

Yes, that driver is right here at this site.

> Why not adding it to kernel?

Only drivers that are fully Linux Wireless Standards compliant will be accepted into the kernel. These Realtek out-of-kernel drivers are not standards compliant. In fact, there is code in the Linux kernel to prevent WiFi 7 drivers based on the same methods from operating in Linux. This effectively kills this type of driver going forward. I would argue that it should have been done for WiFi 6. Realtek's Wifi 6 out-of-kernel drivers are really bad.

For more information, here is the link to the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

Menu items 1 and 2 should help you understand more about usb wifi on Linux.

If you have additional questions, please ask.

Saludos,

@morrownr