morrownr/USB-WiFi #286: Extremely slow performance with the mt7610u driver

SimplyTadpole, 2023-06-26:
Hi, I'm gonna preface this by clarifying that I'm not a very tech-savvy person at all, so I'm not sure if I'm posting the issue on the right page (you do not have a project for the 76x0 driver, only the 76x2), so I figured the safer bet would be to put it here.

I've been stuck with a buggy Realtek wifi USB dongle (the RTL8821AU) for a good while, and was often running into trouble due to the third-party driver not playing nice with system upgrades. I followed your guide to buy the [Anddear MT761003](https://www.aliexpress.com/item/4000079918154.html) dongle off of AliExpress after too many failed attempts to buy a Mediatek device here - I had to settle for it because international shipments are very expensive - but the performance I'm getting from it is slow. Like, **_abnormally_** slow - contrary to the reviews I've seen on its AliExpress page. I went from getting over 40 mbps of internet speed to less than 2! Additionally, although it's advertised as supporting 5G, in fact only the 2G mode works - my and others' 5G wifi connections rarely show up, and if they do, connecting to them is impossible.

Common sense would say that the most likely possibility is that the model I ordered is simply just that weak - and that's been my initial belief - but, for starters, the reviews I've seen don't suggest that is the case. Plus, it states it can reach up to 600 mbps of internet speeds and you mentioned reaching over 100 mbps, so it doesn't make sense that I'm only getting 1.6 mbps. Not to mention its inability to access 5G altogether despite being advertised as capable. Could you shine some light on this?

I've been attempting to look up information on this to see if anyone else had similar issues, and I saw [two](https://askubuntu.com/questions/1453381/mediatek-inc-mt7612u-802-11a-b-g-n-ac-wireless-adapter-slow-on-kubuntu-22-10) [such](https://github.com/morrownr/7612u/issues/17) pages, but I couldn't really understand what was going on with them (as mentioned earlier, I'm *not* tech-savvy - in fact, I ordered this dongle precisely to not have to fiddle with drivers anymore, yet here we are).

If it helps, I'm using Fedora Kinoite 38. I tried putting the USB on both USB 2.0 and 3.0 slots to see if there'd be any differences, but while 3.0 helped out a little, it's still unusually slow (went from 1.6 to 5 mbps).

morrownr, 2023-06-26:
Hi @SimplyTadpole 

The first thing I would look at is the signal level you are getting. You can install an app called wavemon and run it from a terminal:

$ wavemon

Tell what the signal level and link quality numbers are.

SimplyTadpole, 2023-06-26:
The link quality is around 84% (59/70) and the signal level is -51 dBm (7.94 nW);

morrownr, 2023-06-26:
> The link quality is around 84% (59/70) and the signal level is -51 dBm (7.94 nW). Is that good or bad?

That is pretty good.

I just did a quick test with my CF-951AX adapter. It is showing 87% and -48 dBM so my numbers are close to yours. Here is the result:
```
[  5]   0.00-1.00   sec  65.2 MBytes   547 Mbits/sec    0   1.87 MBytes       
[  5]   1.00-2.00   sec  75.0 MBytes   629 Mbits/sec    0   1.87 MBytes       
[  5]   2.00-3.00   sec  73.8 MBytes   619 Mbits/sec    0   1.87 MBytes       
[  5]   3.00-4.00   sec  72.5 MBytes   608 Mbits/sec    0   1.87 MBytes       
[  5]   4.00-5.00   sec  73.8 MBytes   619 Mbits/sec    0   1.87 MBytes       
[  5]   5.00-6.00   sec  72.5 MBytes   608 Mbits/sec    0   1.87 MBytes       
[  5]   6.00-7.00   sec  75.0 MBytes   629 Mbits/sec    0   1.87 MBytes       
[  5]   7.00-8.00   sec  72.5 MBytes   608 Mbits/sec    0   1.87 MBytes       
[  5]   8.00-9.00   sec  73.8 MBytes   619 Mbits/sec    0   1.87 MBytes       
[  5]   9.00-10.00  sec  72.5 MBytes   608 Mbits/sec    0   1.87 MBytes 
```

That is an average of 609 Mbps. The test is with a 5 GHz DFS channel where there are no other APs on the channel.

The wavemon info leads me to think that your problem may not be with the adapter or with the driver. It may be external. What I would do is change the channel in your wifi router to 1 and post the results as you did above and then do the same with channel 6. We do this to so which of your available wifi channels is the least congested. WiFi technology is based on sharing so an APs on the same channel as your WiFi router are going to cause sharing with you. The idea is to find and use the least congested channel.

morrownr, 2023-06-26:
> I also had some other people suggest that a BIOS update might fix or at least help this issue. Since I'm apparently getting good signal quality, do you think that might be the case? I never really did a BIOS update on my own, so I'm scared of doing that ^^'

It is unlikely a bios update will help in this situation. It is good that you are scared of doing it because you can kill you computer if you have not studied how to do it or have someone helping you the first time. Let's put that low of the list of things to do.

> Sorry, but could you explain how channel changing works and how to do it, and if it will interfere with other wifi connections? I'm not really familiar with any of this, so I'm quite confused.

Let's fix the confusion. It may take a few days but this is a public forum with a lot of traffic so I'm sure others can benefit as well.

First tell me the brand and model number of your wifi router/modem. 

What type of internet service do you have? Cable, DSL, Satellite or fiber?

What I am looking to do it find the information that will allow you to log into your wifi router so that we can check and possible change some of the settings?

Do you have another way, other than the adapter, to connect to your wifi router?

> (Also, how did you get such results printed out? What command did you use?)

I use iperf3. It would be cool if you could set iperf3 up but it is an advanced capability. iperf3 requires that a copy is running on another system in your LAN. In my case, iperf3 runs in server mode on my main router. That is possible because I use OpenWRT. We need to work other issues and maybe we can try this at some point. Do you have a RaspBerry Pi?

SimplyTadpole, 2023-06-28:
Jeez, thanks for warning me about the BIOS. 😓

The router is a Nokia G-140W-H, and the internet service is provided via Fiber. Sadly, the router's password was changed two years ago and none of us remember it - we tried guessing it, but it seems the router has a security measure against brute-forcing as it locks itself if you fail to guess the password three times (we tried twice).

Sorry for the late response - was hoping we could call the ISP about it but haven't had the chance to. However, I do have an update - it seems that this issue is indeed not related to the dongle or the driver, as other devices (ranging from other computers/laptops - which run Windows - and even phones) are also having unusually slow internet. Sometimes it fluctuates and gets better, while other times it doesn't.

I am guessing that means this issue is likely out of scope for this github issue page, and as such I understand if you would rather have it closed - but I do wonder if your suggestion of changing the wifi channel to something less congested will help. We found out that this issue only affects wifi - my parents tried to connect one of the other laptops to the router via ethernet while not moving it away from its position, and indeed the internet connection did get *much* better. (Sadly, using ethernet long-term isn't viable for us at the moment as our computers and laptops are in awkward positions related to the router.)

morrownr, 2023-06-28:
> I am guessing that means this issue is likely out of scope for this github issue page, and as such I understand if you would rather have it closed...

Not really. I prefer that users leave issues open and let me close the ones that seem to be no longer relevant. I periodically do a little house cleaning.

 > I do wonder if your suggestion of changing the wifi channel to something less congested will help.

There is only one way to find out. I did a search on "how to reset a Nokia G-140W-H wifi router." The instructions are out there. A reset will clean things like the router id and password (that you do not remember) and it may clear the wifi ssid's and passwords as well so if you want to keep the ones you have, you need to write them down. Very often problems with wifi have to do with either the wifi router location or its internal settings. The internal settings of almost wifi routers can be changed by the user to one degree or another.

If you are able to perform a reset and gain control of the router's web interface so as to check and change the settings, I will be glad to help. You really should try to download and manual for your router before you start as it will contain instructions for resetting the router and logging into the web interface.

morrownr, 2023-07-01:
@SimplyTadpole 

Any progress?

abawany, 2024-02-13:
FYI, I also have a Linksys AE6000 with this chipset and recently experienced a performance issue as well (1.8Mbps on the local network, only able to connect to 802.11b even though it can see the `g` networks). All other devices on my network perform well and putting in an old EDIMAX usb adapter in the affected machine gave me back a lot of the performance so I think something is failing on the original adapter. Trying it in different machines also led to similarly low performance. I haven't checked logs for associated errors but I will check soon.

morrownr, 2024-02-13:
> only able to connect to 802.11b

Ouch! 

I have a Linksys AE6000. I can do some testing but I need to know more about your hardware and distro.

abawany, 2024-02-13:
I'm running on a Dell with i5-7500 (~2017) with kernel `5.15.0-94-generic` and Lubuntu 20.x.

morrownr, 2024-02-14:
Understand. It is a handy, tiny adapter but failures do happen. I keep mine as a backup and haven't used it for a while. I do have an Alfa ACHM with the same chipset and it runs 24/7. It haven't noticed any slowdowns with it.

Maybe you can get one of the new adapters based on the new mt7925 chipset when they are available this year. The driver went into kernel 6.7 so you will need to upgrade to Lubuntu 24.04. The mt7925 is a WiFi 7 tri-band chipset. Something to look forward to.

@morrownr

morrownr, 2024-02-18:
I dug my little Linksys AE6000 adapter out. Chipset is mt7610u. Remember that this is an AC600 class adapter so don't expect any new speed records.

Client: Debian 12, kernel 6.5, 5 GHz AC 80 MHz

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.107 port 40610 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  17.6 MBytes   147 Mbits/sec    0    363 KBytes       
[  5]   1.00-2.00   sec  19.0 MBytes   159 Mbits/sec    0    399 KBytes       
[  5]   2.00-3.00   sec  18.8 MBytes   157 Mbits/sec    0    434 KBytes       
[  5]   3.00-4.00   sec  18.5 MBytes   155 Mbits/sec    0    455 KBytes       
[  5]   4.00-5.00   sec  18.6 MBytes   156 Mbits/sec    0    455 KBytes       
[  5]   5.00-6.00   sec  18.8 MBytes   158 Mbits/sec    0    455 KBytes       
[  5]   6.00-7.00   sec  18.1 MBytes   152 Mbits/sec    0    479 KBytes       
[  5]   7.00-8.00   sec  19.6 MBytes   164 Mbits/sec    0    479 KBytes       
[  5]   8.00-9.00   sec  18.6 MBytes   156 Mbits/sec    0    479 KBytes       
[  5]   9.00-10.00  sec  18.5 MBytes   155 Mbits/sec    0    479 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   186 MBytes   156 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   183 MBytes   154 Mbits/sec                  receiver

iperf Done.

```

That is not bad at all given that I have two walls and about 20 feet to the AP. This little Linksys adapter is basically a nano adapter so it is really doing a good job given what it is up against.

I also did a quick test with my mt7922 based PCIe card.

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.127 port 51062 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  87.9 MBytes   737 Mbits/sec    0    967 KBytes       
[  5]   1.00-2.00   sec   109 MBytes   912 Mbits/sec    0    967 KBytes       
[  5]   2.00-3.00   sec   109 MBytes   912 Mbits/sec    0    967 KBytes       
[  5]   3.00-4.00   sec   109 MBytes   912 Mbits/sec    0    967 KBytes       
[  5]   4.00-5.00   sec   111 MBytes   933 Mbits/sec    0    967 KBytes       
[  5]   5.00-6.00   sec   114 MBytes   954 Mbits/sec    0    967 KBytes       
[  5]   6.00-7.00   sec   111 MBytes   933 Mbits/sec    0    967 KBytes       
[  5]   7.00-8.00   sec   110 MBytes   923 Mbits/sec    0    967 KBytes       
[  5]   8.00-9.00   sec   111 MBytes   933 Mbits/sec    0    967 KBytes       
[  5]   9.00-10.00  sec   109 MBytes   912 Mbits/sec    0    967 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.06 GBytes   906 Mbits/sec    0             sender
[  5]   0.00-10.00  sec  1.05 GBytes   904 Mbits/sec                  receiver

iperf Done.

```

Have you been using wavemon to see what is going on?