morrownr/USB-WiFi #288: Need TESTERS: rtl8852bu/rtl8832bu driver

morrownr, 2023-07-01:
Hello Everyone,

I am working on a very recent release of the Linux source code for usb wifi adapters that use the rtl8852bu and rtl8832bu chipsets. If you have an adapter with either chipset and would be willing to test and help, please let me know by posting a reply.

Please do not buy a new adapter for this testing. Why? Part of the mission of this site is to help Linux users with information regarding USB WiFi adapters. I recommend that Linux users buy adapters that are single-state (no onboard Windows driver), single-function (no bluetooth) and use in-kernel drivers. Since this is an out-of-kernel driver, a recommendation to buy adapters based on these chipsets would go against one of the basic recommendations on this site. If fact, all adapters that I have seen with these chipsets are also multi-function which is not what we want.

With that said, in working on this driver over the last few days, it is of good enough quality that I think it would be a good thing to make it public for those who already have adapters based on these chipsets or have little alternative but to buy an adapter based on one of these chipsets. If anyone is aware of a good quality single-state adapter that uses these chipsets, let us know. Also, there is a possibility that these chipsets could be supported with in-kernel drivers, rtw89, at some point.

Question: Should I buy an adapter with a rtl8852au or rtl8832au chipset?

Answer: No. Those are the original WiFi 6 chipsets that Realtek released. The driver that was released was really bad and no updated version is available that I am aware of. These chipsets may have gone out of production. It is not clear to me that we will ever see reasonable Linux support for these chipsets so my recommendation is to avoid adapters based on these chipsets.

Question: Should I buy an adapter with a rtl8852cu or rtl8832cu chipset?

Answer: Adapters based on these chipsets are new to the market and are Realtek's first WiFi 6e (6 GHz) chipsets for usb adapters. It appears driver support will be out-of-kernel so I cannot recommend them to Linux users. There are several single-state, single-function, WiFi 6e adapters in the Plug and Play list.

@morrownr

bjlockie, 2023-07-01:
Is there any chance of making the rtl8832cu driver work for the 8832au?

morrownr, 2023-07-01:
> Is there any chance of making the rtl8832cu driver work for the 8832au?

I would say that the chance is slightly above zero.

Jibun-no-Kage, 2023-07-15:
Ah... how many decimal places is that above zero?  Is it 0.0001 or 0.0000001 or 0.000000001?  Or ?

briantbutton, 2023-07-15:
I don't know where to leave this, so . . . 
. . . grand thanks to you and everyone else who has worked on these.  You are a light in the night.

morrownr, 2023-07-15:
> . . . grand thanks to you and everyone else who has worked on these. You are a light in the night.

You are welcome and thank you for the kind words. I get a lot of help. Many folks just pick a part of what you see and maintain it as they have time.

Stop by anytime. You are always welcome to start a new issue.

@morrownr

morrownr, 2023-07-15:
> how many decimal places is that above zero? Is it 0.0001 or 0.0000001 or 0.000000001? Or ?

I'll go with 0.0000001. A few folks like me had hope for the rtl8832au chipset but it looks like it has turned into abandonware.

briantbutton, 2023-07-18:
I am in the midst of buying adapters for our own tests, so I thought to buy a couple with these chipsets to test.

I have not had much luck searching for adapters by chipset.  Can you suggest a make/model?