morrownr/USB-WiFi #291: SALE: $10 off ALFA AWUS036AXML or AWUS036AXM w/ code JULYXML

morrownr, 2023-07-04:
Rokland has a 4th of July sale in progress:

https://store.rokland.com/pages/weekenddeals-sale

The sale includes more than the above listed Linux Plug and Play USB WiFi adapters. It also includes directional antennas and more.

Note: If anyone has one of the new AWUS036AXM adapters, we could sure use a report.

@morrownr

fhteagle, 2023-07-18:
I thought I had posted in here but I do not see my comment, so I will repeat. The AXML was $20 off during Prime Day sale. It has since expired, but that does bode well for future price decreases.

morrownr, 2023-07-18:
Hi @fhteagle 

> that does bode well for future price decreases.

It does. There is a market surplus of chips right now and that may continue for a while. New fabs are coming online now and over the next few years. Another thing that will help bring prices down on the WiFi 6e adapters is that the new Realtek rtl8852cu WiFi 6e chipset is available.

I'd like to see the mt7922u chipset come on the market.

bjlockie, 2023-07-18:
Does the Realtek rtl8852cu have an in kernel driver?

morrownr, 2023-07-18:
> Does the Realtek rtl8852cu have an in kernel driver?

No. I am not saying that I recommend an adapter with this chipset, I'm just saying that are coming on the market.

The rtw89 driver is in-kernel and it has support for the rtl8852be and ce so I can see a future where it could have an in-kernel driver at some point similar to what has happened with the rtw88 driver and several WiFi 5 chipsets.

The out of kernel driver for the rtl8852bu is getting to the point that it is not bad. I'm on schedule to go public with the rtl8852bu repo here by August 01. I don't have a rtl8852cu adapter so can't say much about it but having some competition on the market should help prices come down.

rmandrad, 2023-08-31:
I have a rtl8852cu that works fine with https://github.com/lwfinger/rtw8852cu with AX as an AP ... hostapd recognizes the device on the 6ghz band