morrownr/USB-WiFi #297: News: rtl8852bu driver now available at this site

morrownr, 2023-07-31:
Good day,

I want everyone to understand that I am not encouraging Linux users to buy usb wifi adapters based on the rtl8852/32bu chipsets. However the new rtl8852bu repo is public for those that might benefit from it:

https://github.com/morrownr/rtl8852bu

I am posting the following warning in the README of this new driver:

Warning: Posting this driver is not a recommendation for Linux users to buy USB WiFi adapters based on the Realtek rtl8852/32bu chipsets. Realtek out-of-kernel wifi drivers, such as this one, are designed to be used by skilled programmers producing products such as embedded systems. Users of desktop distros such as Ubuntu, Debian, Manjaro, Fedora, Raspberry Pi OS and other mainline desktop distros will likely find adapters that use in-kernel drivers to be more stable and much more trouble-free.

Status of this driver:

The driver has only been tested by 2 individuals, including me. The lack of being able to find additional testers is likely because very few Linux users own adapters based on these chipsets. The point I am trying to make is that the driver has not had adequate wide testing.

FAQ:

Why did I work on and release this driver? There may be individuals that find it difficult to find adapters that use the preferred usb wifi chipsets and I want to make sure that they can use Linux and usb wifi adapters.

Where should I look for information about the preferred adapters and chipsets for Linux users? On the Main Menu (README), look at menu item 2, The Plug and Play List. It will show the preferred chipsets and many adapters that are known to work well with Linux.

Do you plan to release a driver for the rtl8852/32au chipsets? No. I tried a couple of years ago but the driver was simply too bad to release. It is a challenge to work on Reqltek out-of-kernel drivers due to the lack of documentation and it appeared to me that there may have been silicon bugs in those chipset. You can't fix all hardware bugs with software. Recommend all Linux user avoid adapters that use the rtl8852/32au chipsets.

@morrownr

bjlockie, 2023-07-31:
Does this driver happen to work with the cu?

morrownr, 2023-07-31:
No. The cu will require a new driver.