morrownr/USB-WiFi #298: Question: alfa network card range and battery drain

Okitask, 2023-08-03:
1)I am looking for the longest range network adapter and from the "performance comparison" awus036achm and awus036ach seem to be the winners. I also noticed awus036ach in heavy load power usage requires 800ma against the 380m of the achm.
So should it drain more battery (on laptop)?
But I know USB 3.0 has more power efficiency in idle status so who is better in term of battery drain?

2)awus036ach has faster packet injection but is the difference worth it? I mean, i think to do wep cracking or deauth attack achm works well. Use case where I can notice this pro?

3)Does the awus036axml have internal power range amplifier as achm and ach or a better range?

Thank you a lot!

morrownr, 2023-08-03:
Hi @Okitask 

Discussions like this are cool.

> I am looking for the longest range network adapter and from the "performance comparison" awus036achm and awus036ach seem to be the winners. 

They are the winners. I have about 35 USB WiFi adapters. Don't ask. I need to do another performance comparison but have been busy. I'll try to do one later this year. I do have a speed comparison on the Main Menu but that is not what you are looking for.

Let's look at what Alfa says:

[image]

Alfa uses 3 terms to differentiate range: max, high and standard. The only 2 max adapters they currently have are the achm and ach. The ach does use a lot of power. The heat is dissipated well as the adapter does not get hot. It is based on the rtl8812au chipset. The achm uses far less power and can run indefinitely at full speed without issue. It is based on the mt7610u chipset.

If you need an adapter right now for security analysis / pen testing, I give a hands down recommendation to the achm. Why?

1. You will have a well supported in-kernel driver indefinitely. The ach only has a Realtek out-of-kernel driver and support has already gone end-of-life. In fact, the rtl8812au chipset is end-of-life and is no longer being produced.
2. I'm not a professional pen tester but I have worked with many who are over the years and my impression is that speed is not that big of a deal.
3. The achm uses an in-kernel standards compliant driver. In-kernel driver do more of the monitor mode things you will want to do.

> Does the awus036axml have internal power range amplifier as achm and ach or a better range?

We don't have a `max` adapter yet from Alfa as far as WiFi 6e goes. I have been lobbying hard and I can see it happening maybe early next year. The mt7921u driver that supports the axm and axhm adapters is in kernel and is VERY solid. One user that bought an adapter with the mt7921au chipset last year after adapters were released said he tested monitor mode in every way he uses an adapter and said it worked well. He said monitor mode was even working with 6 GHz.

I think Rokland has a $10 off August sale on the axm right now:

https://store.rokland.com/collections/wifi-6-6e/products/alfa-awus036axm-wifi-6e-3000-mbps-tri-band-2-4-5-6-ghz-wifi-usb-adapter

I think you have to click on the sale ad at the top of the screen to get the code. I do not have an axm yet. Rokland actually has a staff that not only knows and supports Linux but they know and support pen testing so you could call them and ask questions about the axm.

@morrownr

Okitask, 2023-08-03:
Thank u for the response, i really appreciate your work!
So is it correct to say that achm consume less power than ach even if USB 3.0 has more power efficiency in idle status?
Also i think this should not be so relevant, because often network interfaces are in some mode and idle status is rarely achieved, isn't it?

Do you also know how many client achm can have in ap mode?

morrownr, 2023-08-04:
> i really appreciate your work!

You are welcome. I enjoy it but I just don't seem to have time to do everything I want to do.

> So is it correct to say that achm consume less power than ach even if USB 3.0 has more power efficiency in idle status?

While I have not testing this specifically, I could do some testing when I have time. I think this test would require information about how much of the adapters on time is spend idle. My opinion is that the achm would use far less electicity.

Here is something interesting. You saw the maxpower as measured in the document you read but something came up earlier this year somebody wanted to know how much power the mt7921au based adpter were using. I moved recently and still have not found my usb current device but will find it at some point. What did was use lsusb to see what the maxpower requested by the adapter will be:

$ sudo lsusb -vv
mt7921au        MaxPower              160mA

The lsusb numbers are usually pretty close. That is an incredibly low number but does explain why none of us were seeing any heating. Has Mediatek moved to a chip making process that is using much less power? That is the only explanation we could figure out for now.

> in some mode and idle status is rarely achieved, isn't it?

This can be true. It defends on your use case. Use as an AP with many IoT clients might make idle rare.

> Do you also know how many client achm can have in ap mode? (hardware limit i mean , I know this can be set via hostapd)

I really think this is determined by the driver and by the bandwidth that is required. Last year a guy was setting up an AP with a USB WiFi adapter and ask me basically the same question. I can get 12-15 clients going on a good day if I do some work so I am limited. He was wanting to connect 50+ clients. After listening to what he needed, I told him if it was me, I would start with the following adapter:

https://www.amazon.com/Panda-150Mbps-Wireless-Adapter-Antennas/dp/B004AC6X0K

It worked. I'm not recommending this adapter for your situation. The rt3070 chipset was first produced in 2009 (I think) and only ceased production in 2021.

What I think right now is either the Alfa achm or axm depending on whether range or power usage is more important.

morrownr, 2023-08-08:
Oh, I forgot to mention this fact that I discovered about the ACHM:

Even though the ACHM is a lowly AC600 class adapter, there are situations where it can outperform AC1200 class adapters or even AXE3000 class adapters. Why is that? When the AC1200 class adapter can't get a connection due to distance, you may find the ACHM smoking right along like the AP is in the same room.

Okitask, 2023-08-08:
Really happy to know this! Finally few days ago I ordered achm and I'll let you know. And yeah to have a more complete comparison sheet it would be great if paramaters like: number of client supported (in master mode), idle Vs heavy load power usage. Also I hope to see more network interface adapter in the test. Waiting from alfa a max range 6GHz network adapter with mediatek chipset.

Okitask, 2023-08-11:
@morrownr 
Ok finally i tested this beauty. Unfortunately i can't compare it with other adapter but it has sure long range because it scanned x2.5 AP more than my internal adapter (intel one).

I noticed the following "drawback":

1)In Windows, led on the adapter works well but in linux doesn't. I think something wrong with the driver but really it doesn't matter.
2)Seems txpower can't be changed, i tried whatever but nothing and this is so frustrating for evil twin attack.
3)iwconfig/iw dev shows txpower=3.0dBm but i think is only a bad indicator.
3)Windows says Mac randomization isn't supported with this adapter. Can't yet have the time to check but i'll give a look on this

morrownr, 2023-08-11:
> 1)In Windows, led on the adapter works well but in linux doesn't. I think something wrong with the driver but really it doesn't matter.

It is not clear to me why this is so but it seems the support is not in the modern in-kernel driver infrastructure at this point. What I have noticed is Alfa, with their recent AXML and AXM adapters, has accounted for this by making their LEDs power status lights instead of blinking madness. Folks like me that like to run the LEDs off have been accounted for because the LEDs are in a location and have low brightness so they don't don't bother you, even at night, but you can see the light if you know where to look. Interesting how an adapter maker is adjusting to support Linux.

> 2)Seems txpower can't be changed, i tried whatever but nothing and this is so frustrating for evil twin attack.

Yes, the txpower is locked but I have not seen any new adapters that can change txpower in a long time so I did not mention this. Can you explain what impact this has on a evil twin attack?

> 3)iwconfig/iw dev shows txpower=3.0dBm but i think is only a reported issue.

Yes, this is an issue that comes up but many Mediatek chipsets show the same 3 dBm reading. This is an issue that we currently  don't have information about but the reading has no bearing on performance.

You could send a nice email to Mediatek for an answer. The below is the Mediatek Linux Wireless webpage. I'd recommend using the email addresses of a couple of the devs whose email ended with @mediatek.com

https://wireless.wiki.kernel.org/en/users/drivers/mediatek

I'll be glad to post answers here at a good place.

> Windows says Mac randomization isn't supported with this adapter.

I have never been a regular user of Windows. I got nothing.

Okitask, 2023-08-11:
> Can you explain what impact this has on a evil twin attack?

When you need to create a rogue AP (as in evil twin) parameters like txpower is important because "you are screaming your fake AP more than the others" so it appears having good signal and so in the first position or near. I know this doesn't rely only on it but also on other parameters, txpower is one of them. 
Know i am just wondering if txpower changing works but because of the indicator it seems to not work. I have to try to low it to see what happens or if you have also tried let me know!

> I have never been a regular user of Windows. I got nothing.

Sure i'll let you know

amisix, 2023-08-12:
I'm also not able to get the tx power of my ACHM to adjust in linux. Spent a bit of time trying too. But I'm able to get it to adjust tx power in Windows with a registry hack and it runs great with an amp because it require a lower tx power (<20dB) feeding it. The tx power in linux is just too high - the adapter is rated at 475mW, 27dB. So it is possible to adjust the tx power, just not in the OS we'd like most.. 

Well, wait, I did find a way to lower the tx power actually. You can always change the country code to another country with lower maximum tx power, like JP (20dB). I tried it with my amp and it worked, hopefully you experience similar results.

Okitask, 2023-08-12:
> But I'm able to get it to adjust tx power in Windows with a registry hack

  Could you kindly demonstrate the process you used for this?

I have also observed the capability to conduct VIF (unlike realtek8812au and others, as indicated in this [list](https://github.com/v1s1t0r1sh3r3/airgeddon/wiki/Cards%20and%20Chipsets)). This feature is remarkably potent because it obviates the necessity for two or more adapters to execute a comprehensive attack on the same target (such as in the scenario of an evil twin attack).

Has anyone tested alternative omni antennas with a higher dbi? I came across this [9dbi Alfa omni antenna](https://store.rokland.com/en-de/products/alfa-9-dbi-aoa-2458-79af-2-4-5-ghz-dual-band-outdoor-n-female-wifi-omni-antenna).

amisix, 2023-08-12:
@Okitask 

Install the mt7610u Windows driver from Mediatek
https://www.mediatek.com/products/broadband-wifi/mt7610

Now you need to locate the registry key for your mt7610u adapter. It will be a long string with a number at the end, like this example:

HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{4D36E972-E325-11CE-BFC1-08002BE10318}\0012

Navigate to the below registry key via Regedit

HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\Class\{4D36E972-E325-11CE-BFC1-08002BE10318}\ 

There will be a long list of numbered folders. Search through those folders for the mt7610u adapter by looking for:
"Componentid" =  \USB\VID_0E8&PID_7610
"DriverDesc" =  "802.11AC Wireless LAN card"

Once you've located the correct registry entry, locate the setting "Countrytxpower" and adjust it accordingly. The number in parenthesis is dB.

>Has anyone tested alternative omni antennas with a higher dbi?

I'm running off-brand tall 9dB omni antennas with my dual ACHM Kismet rig and have had great success with them.

Okitask, 2023-08-12:
@amisix 

> I'm running off-brand tall dual-band 9dB omni antennas with my dual ACHM Kismet rig and have had great success with them.

Has your range improved significantly? It would be greatly appreciated if you could share some data illustrating the contrast with the stock version.

amisix, 2023-08-12:
>Has your range improved significantly? It would be greatly appreciated if you could share some data illustrating the contrast with the stock version.

I assume so but don't have/won't have any recorded metrics, sorry. It's feels over reals sometimes with antennas as you know. 
https://www.amazon.com/gp/product/B07YRSPRK6/