morrownr/USB-WiFi #302: Monitor Mod support on AWUS036AXML

svnas1994, 2023-08-23:
Hi All, 

Greetings. 

Just want to know whether Alfa AWUS036AXML dongle support monitor mode or not. 
If it support monitor mode, can someone please confirm what is Minimum System requirements it will support. 

Currently i have Ubuntu running on 22.04.3 LTS with 5.19 Kernel. 

Can someone please help me on this part. 

If it wont support monitor mode, please suggest which is the best AX dongle to be used to capture the ax traffic. 

Thanks, 
Sairam

morrownr, 2023-08-23:
Hi @svnas1994 

> Currently i have Ubuntu running on 22.04.3 LTS with 5.19 Kernel.

That will work. Main Menu item 2 has a lot of information.

> Just want to know whether Alfa AWUS036AXML dongle support monitor mode or not.

The driver for the AXML supports monitor mode. It would help us help you if you explain exactly what you will be doing with monitor mode so we can test to make sure you are good to go.

Regards

svnas1994, 2023-08-23:
Thanks for the reply @morrownr . 

I just want to check debug throughput related issues with WiFi 6 Protocol testing. 

Thanks, 
Sairam

morrownr, 2023-08-23:
> I just want to debug throughput related issues with WiFi 6 Protocol testing.

I have an AXML. If you provide the capabilities needed or a give me a guide on how to do it, I'll give it a show as I have time. There are several others here with AXML's so maybe someone else can speed things up.

svnas1994, 2023-08-24:
Thanks @morrownr for the offering. 

The main aim of our testing is to debug the OFDMA and MU MIMO related throughput and functional related scenarios. 

Thanks, 
Sairam

ZerBea, 2023-08-27:
Unfortunately several problems running latest kernel:

AWUS036AXML
https://github.com/ZerBea/hcxdumptool/issues/334

AWUS036AXM
https://github.com/ZerBea/hcxdumptool/issues/337

morrownr, 2023-08-27:
Hey @ZerBea 

Good info. USB3 is not mankind's greatest creation.

Since the ultimate issue that the OP wants solved is "what adapter should I buy?", let me throw out some general advice and since you are highly qualified to answer monitor mode questions, please jump in and correct me when you see things differently:

My advice to those seeking monitor mode support is to first and foremost, get an adapters that uses in-kernel drivers. While maintaining the Realtek out-of-kernel drivers here at this site over the last few years, I have learned a lot and I just can't recommend the Realtek out-of-kernel drivers or the adapters that use them to anyone except maybe those working on embedded projects but we are talking about skilled programmers in this case.

Of the adapters that use in-kernel drivers and are going to be used with monitor mode, it depends on some factors such as desired range and throughput capability. The Alfa ACHM has the best range currently but is an AC600 class adapter. It is popular for monitor mode as many uses of monitor mode do not require adapters that are high throughput. The AXML and AXM are both AXE3000 and are capable of high throughput but have less range than the ACHM. The ACM is an AC1200 class adapter that is dependable for many uses as it has good range and good throughput.

The real issue boils down to capabilities of the drivers:

mt7921u - AXML and AXM
mt7612u - ACM
mt7619u - ACHM

Since there are so many things that can be done in monitor mode, I recommend testing for that capability  as the only way to know for sure so I offer users the opportunity to send me a guide that I can follow and I will provide the results. I have AXML, ACM and ACHM which covers the main modern chipsets.

You have much more expertise with monitor mode and I would be interested in your thoughts.

@morrownr

ZerBea, 2023-08-27:
@morrownr , I fully agree.

Luckily in kernel support for Realtek chipsets is going to better since kernel 6.3:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/net/wireless/realtek/rtl8xxxu?h=v6.4.12

```
$ lsusb
Bus 001 Device 020: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 021: ID 2357:010c TP-Link TL-WN722N v2/v3 [Realtek RTL8188EUS]

$ hcxdumptool -L

Requesting physical interface capabilities. This may take some time.
Please be patient...

available wlan devices:

phy idx hw-mac       virtual-mac  m ifname           driver (protocol)
---------------------------------------------------------------------------------------------
  0   3 00c0cab067e9 00c0cab067e9 + wlp22s0f0u4      rtl8xxxu (NETLINK)
  1   4 503eaad5e035 503eaad5e035 + wlp22s0f0u3      rtl8xxxu (NETLINK)
```

And it was a hard way for me to learn NETLINK / RTNETLINK stuff to manage this driver (rtl8xxxu) and the MediaTek drivers.
Unfortunately not all combinations (USB controller - USB device) are working as expected.

Raspberry Pi (USB 2) - mt7612u == ACM working as expected
Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne USB 3.1 ==  ACM not working as expected (as of kernel >= 6.1.28)