morrownr/USB-WiFi #306: Three questions about the AVM FRITZ! AC 860

chronosirius, 2023-09-10:
1. What chipset does it use? Do they make their own or do they just package intel/broadcom/etc.?
2. Can it be gotten inside the United States?
3. Is it standalone or does it require a FRITZ! Box? The AVM website was unclear, for me at least, about this.

Thank you!

morrownr, 2023-09-10:
Hi @chronosirius 

>  What chipset does it use? 

Mediatek mt7612u

> Do they make their own or do they just package intel/broadcom/etc.?

The maker buys chips from Mediatek.

> Can it be gotten inside the United States?

It may be but I have not found a source. If you are in the US, you have easy access to many USB WiFi adapters. Go to:

https://github.com/morrownr/USB-WiFi

Scroll down. That is the Main Menu for this site. Recommend you read menu items 1 and 2. Menu item 2 has an entry for the Fritz adapter you are looking for. Look in the mt7612u section.

> Is it standalone or does it require a FRITZ! Box? 

USB WiFi adapters are standalone but if you are unfamiliar with them, tell us more information about what computer you have and what distro you use and what your objective is.

chronosirius, 2023-09-11:
Thanks for the quick response! Your recommendation for links 1-2 was really helpful, I looked through it. As for the last question, I had thought as much but thanks for clarifying. From your warning about the FRITZ! adapter about the unreliable chipset (as in the company isn't consistent in which chipset goes in), I will probably be purchasing a Netgear A6210. Thank you so much for assembling this Github repo, it is amazing and I cannot thank you enough!

morrownr, 2023-09-11:
Which RasPi are you using? And what is OWL?

Just trying to see if there is anything we need to pass on to you.

chronosirius, 2023-09-11:
4b, [open wireless link](https://github.com/seemoo-lab/owl) to run opendrop on.

morrownr, 2023-09-12:
In case you missed it, here is the repo that contains mt7612u specific information:

https://github.com/morrownr/7612u

Maybe something will be handy.

Good luck.