morrownr/USB-WiFi #308: News: Add support for Mediatek Wi-Fi7 driver mt7925

morrownr, 2023-09-13:
I follow the Linux-Wlireless list and today I see v2 of the `Add support for Mediatek Wi-Fi7 driver mt7925` patchset flowing in.

Info from v1 can be read at the following:

https://lwn.net/Articles/939966/

This gives us a more clear picture of where Mediatek is headed with USB Wireless. Things we now know:

- The mt7925 chipset  will support a usb version of the chipset.
- This is a WiFi 7 capable chipset.
- It is capable of 160 Mhz channel width.
- This patchset will be going mainline in the not so distant future.

@morrownr

LINGTIANJIAO, 2023-09-18:
When the USB network card of the Mediatek Wi-Fi7 driver mt7925 is released?

LINGTIANJIAO, 2023-09-18:
When the USB network card of the Mediatek Wi-Fi7 driver mt7925 will be released?

bjlockie, 2023-09-18:
>Linux support is on equal footing with Windows as far as Mediatek's 
> strategy is concerned.

I think Windows support by Mediatek is not very good.
Wifi 6 routers are not even common.

morrownr, 2023-09-18:
> I think Windows support by Mediatek is not very good.

I don't use Windows but have seen some complaints. I see this as normal given that it takes time to train people and get up to speed. The company seems to be committed. The complaints seem to be lessening but there are old drivers out there that will cause problems for a while as the company just can't reach out and take everything that is out in the wild back. That is actually an advantage for Linux as there are no mt7921 drivers out in the wild so everyone basically gets the latest for the kernel they are using and it will continue to get better as new kernels are released.

> Wifi 6 routers are not even common.

https://openwrt.org/toh/views/toh_available_16128_ax-wifi

LINGTIANJIAO, 2023-09-18:
But I think the mt7921 has not good performance to Linux. Even can't apply 160Mhz!
So now I'm trying to use the mt7922 with PCIe.

morrownr, 2023-09-18:
> But I think the mt7921 has not good performance to Linux. Even can't apply 160Mhz!

The first Realtek usb chipset that will do 160 Mhz is the recently available rtl8832cu. I have a mt7922 based PCIe card. It will do 160 Mhz width in managed mode. Currently it shows a max of 80 Mhz for AP mode but I suspect that will be corrected in firmware at some point or it could be in code as it is supported with the same driver, mt7921u, that the mt7921 chipsets use. A lot of changes and additions continue to flow into the drivers and firmware. The complexity and size of these modern WiFi 6+ drivers and firmware is astounding.

> So now I'm trying to use the mt7922 with PCIe.

As I said above. Managed mode is working fine with 160 Mhz with the mt7922.

LINGTIANJIAO, 2023-10-02:
I am trying to use the MT7922 in raspberry pi cm4.
Do you know how to set it?
: )

morrownr, 2023-10-02:
@LINGTIANJIAO 

G'day.

> I am trying to use the MT7922 in raspberry pi cm4. Do you know how to set it?

The mt7922 chipset uses the same in-kernel driver as the mt7921: mt7921u.ko

Raspberry Pi OS 05-03-23 is plug and play here with my mt7921au chipsets. I have a mt7922 based PCIe card and it works play and play with my desktop but I have not tested your specific combination but I'm sure we can get it going.

Tell me what version of the RasPiOS you are running?

The mt7922 chipset uses a different set of firmware files than the mt7921. Do you know how to check the version of the firmware files you have? You can read:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

LINGTIANJIAO, 2023-10-02:
> Tell me what version of the RasPiOS you are running?

I have 2 RasPis.
One of them is  
`~$ uname -a`
`Linux raspberrypiling 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`

I have been using this RasPi to set AP mode with USB MT7921.
And when I  
```
~ $ lspci
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20)
01:00.0 Network controller: MEDIATEK Corp. Device 0616
```

The other one is Ubuntu 23.04
And when I 
```
~$ lspci
00:00.0 PCI bridge: Broadcom Inc. and subsidiaries BCM2711 PCIe Bridge (rev 20)
01:00.0 Network controller: MEDIATEK Corp. MT7922 802.11ax PCI Express Wireless Network Adapter
```
 
But when I ran 
`~$ ip a`
both of them have no new interface.
: )

> The mt7922 chipset uses a different set of firmware files than the mt7921. Do you know how to check the version of the firmware files you have? You can read:

Yeah, I tryed. But it doesn't work.
: )

morrownr, 2023-10-02:
> 01:00.0 Network controller: MEDIATEK Corp. Device 0616

The 0616 is just AMD speak for mt7922. AMD and Mediatek have a close contractual relationship. Intel does its own wifi but AMD has a contract with Mediatek and uses their own branding. 0608 = mt7921.

I'm on my main dev box right now. I has a PCIe card using the mt7922 chipset and kernel 6.1 and it works well.

If the mt7921u.ko driver is in place, which it should be with kernel 6.1 and the firmware files are the latest, it should work unless your card is using a vid/pid that is not in kernel 6.1. Can you get the vid/pid for your card and we will do some kernel diving to see if your vid/pid is there.

morrownr, 2023-10-02:
$ grep PCI_ID /sys/bus/pci/devices/*/uevent

/sys/bus/pci/devices/0000:00:00.0/uevent:PCI_ID=8086:0150
/sys/bus/pci/devices/0000:00:01.0/uevent:PCI_ID=8086:0151
/sys/bus/pci/devices/0000:00:14.0/uevent:PCI_ID=8086:1E31
/sys/bus/pci/devices/0000:00:16.0/uevent:PCI_ID=8086:1E3A
/sys/bus/pci/devices/0000:00:1a.0/uevent:PCI_ID=8086:1E2D
/sys/bus/pci/devices/0000:00:1b.0/uevent:PCI_ID=8086:1E20
/sys/bus/pci/devices/0000:00:1c.0/uevent:PCI_ID=8086:1E10
/sys/bus/pci/devices/0000:00:1c.1/uevent:PCI_ID=8086:1E12
/sys/bus/pci/devices/0000:00:1c.3/uevent:PCI_ID=8086:1E16
/sys/bus/pci/devices/0000:00:1d.0/uevent:PCI_ID=8086:1E26
/sys/bus/pci/devices/0000:00:1f.0/uevent:PCI_ID=8086:1E4A
/sys/bus/pci/devices/0000:00:1f.2/uevent:PCI_ID=8086:1E02
/sys/bus/pci/devices/0000:00:1f.3/uevent:PCI_ID=8086:1E22
/sys/bus/pci/devices/0000:01:00.0/uevent:PCI_ID=1002:675D
/sys/bus/pci/devices/0000:01:00.1/uevent:PCI_ID=1002:AA90
/sys/bus/pci/devices/0000:03:00.0/uevent:PCI_ID=14C3:7922
/sys/bus/pci/devices/0000:04:00.0/uevent:PCI_ID=10EC:8168

The line for my card is:

/sys/bus/pci/devices/0000:03:00.0/uevent:PCI_ID=14C3:7922

LINGTIANJIAO, 2023-10-02:
> Can you get the vid/pid for your card and we will do some kernel diving to see if your vid/pid is there.

Yeah, my card has two lines:
`/sys/bus/pci/devices/0000:00:00.0/uevent:PCI_ID=14E4:2711`
`/sys/bus/pci/devices/0000:01:00.0/uevent:PCI_ID=14C3:0616`

ziswiler, 2024-02-11:
14C3 is the vendor ID for MediaTek and 0616 should be supported for quite a while:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=f395d41f2a0343fa9d989d8f53e5640329554dd8

Is there anything mt7921e related in dmesg? E.g. like:

```
[    9.430862] mt7921e 0000:01:00.0: enabling device (0000 -> 0002)
[    9.472196] mt7921e 0000:01:00.0: ASIC revision: 79220010
[    9.572245] mt7921e 0000:01:00.0: HW/SW Version: 0x8a108a10, Build Time: 20231120183400a
[    9.952435] mt7921e 0000:01:00.0: WM Firmware Version: ____000000, Build Time: 20231120183441
[   11.101994] mt7921e 0000:01:00.0 wlp1s0: renamed from wlan0
```

BTW: 14E4 is the vendor ID for Broadcom and 2711 their BCM2711 PCIe Bridge.

ilikenwf, 2024-05-29:
https://github.com/morrownr/USB-WiFi/issues/431

````
1b:00.0 Network controller: MEDIATEK Corp. Device 0717
	Subsystem: Foxconn International, Inc. Device e106
	Flags: bus master, fast devsel, latency 0, IRQ 236
	Memory at bfc00000 (64-bit, non-prefetchable) [size=2M]
	Memory at bfe00000 (64-bit, non-prefetchable) [size=32K]
	Capabilities: [80] Express Endpoint, IntMsgNum 0
	Capabilities: [e0] MSI: Enable+ Count=1/32 Maskable+ 64bit+
	Capabilities: [f8] Power Management version 3
	Capabilities: [100] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
	Capabilities: [108] Latency Tolerance Reporting
	Capabilities: [110] L1 PM Substates
	Capabilities: [200] Advanced Error Reporting
	Kernel driver in use: mt7925e
	Kernel modules: mt7925e
````

ilikenwf, 2024-05-29:
I have the PCIe version.

Yeeler, 2025-06-10:
Anyone knew if there is a plan of supporting it?

morrownr, 2025-06-12:
AMD part number for the mediatek mt7927 chip?

AMD collaborates with MediaTek on WiFi solutions, and the MediaTek MT7927 chip is often referred to with an AMD designation as well. 

According to a Framework Community post, the MediaTek MT7927 is associated with the AMD RZ738 designation. 

Therefore, the AMD part number for the MediaTek MT7927 chip is likely RZ738. 

Note that while the MT7927 supports WiFi 7 (802.11be), the driver support for this chip, especially on platforms like Linux, is still being developed. 

-----

Any confirming information by those that own a system with the mt7927/rz738 chip is appreciated.

DocMAX, 2026-01-15:
I have big issues with MEDIATEK Corp. MT7925 802.11be 160MHz 2x2 PCIe Wireless Network Adapter [Filogic 360], ThinkPad T14 Gen6. Kernel 6.18.5.

I don't get a microphone sink on bluetooth headsets. One Headset is not producing sound even it is connected (KOSS PortaPro Wireless 2.0)