morrownr/USB-WiFi #309: Fenvi FU-AX1800 photos and information dumps

whitslack, 2023-09-21:
Manufacturer's product page: https://fenvi.com/product_detail_44.html

Purchased from: https://www.aliexpress.com/item/1005005935638503.html

[image]
[image]
[image]
[image]
[image]

```
# lsusb -v -d 0e8d:

Bus 001 Device 004: ID 0e8d:7961 MediaTek Inc. Wireless_Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0e8d MediaTek Inc.
  idProduct          0x7961 
  bcdDevice            1.00
  iManufacturer           2 MediaTek Inc.
  iProduct                3 Wireless_Device
  iSerial                 4 000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0051
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          5 Config_01
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           9
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              1 WiFi_If
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
[... 771 lines cut ...]
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		* [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
```

Relevant excerpts from kernel log (Linux 6.1.52+):
```
usb 1-1.2: new high-speed USB device number 4 using dwc_otg
usb 1-1.2: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
usb 1-1.2: New USB device strings: Mfr=2, Product=3, SerialNumber=4
usb 1-1.2: Product: Wireless_Device
usb 1-1.2: Manufacturer: MediaTek Inc.
usb 1-1.2: SerialNumber: 000000000
cfg80211: Loading compiled-in X.509 certificates for regulatory database
cfg80211: Loaded X.509 cert 'benh@debian.org: 577e021cb980e0e820821ba7b54b4961b8b4fadf'
cfg80211: Loaded X.509 cert 'romain.perier@gmail.com: 3abbc6ec146e09d1b6016ab9d6cf71dd233f0328'
cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
usb 1-1.2: reset high-speed USB device number 4 using dwc_otg
usbcore: registered new interface driver mt7921u
mt7921u 1-1.2:1.0: HW/SW Version: 0x8a108a10, Build Time: 20230526130917a
mt7921u 1-1.2:1.0: WM Firmware Version: ____010000, Build Time: 20230526130958
```

whitslack, 2023-09-21:
My adapter has a MAC address beginning with `90:de:80`, which [MACVendors](https://macvendors.com/) says is assigned to Shenzhen Century Xinyang Technology Co., Ltd.

whitslack, 2023-09-21:
This adapter feels extremely cheaply made, to the point that I'm not even certain that the "antennas" aren't placebos, but on the other hand it seems to have ridiculously good receive reception, to the point that it's successfully decoding beacons from neighbors' networks that I've never seen with any other adapter.

I've summarized the scan results of my own networks in the table below, comparing the receive the signal strengths with an old AR9271-based 802.11n USB stick against those with the Fenvi FU-AX1800.

```
 Network           |    Freq. (ch.) |   AR9271   |  FU-AX1800
-------------------+----------------+------------+------------
 garage (backhaul) | 2412 MHz (  1) | -57.00 dBm | -49.00 dBm
 privileged        | 2437 MHz (  6) | -56.00 dBm | -46.00 dBm
 IoT               | 2462 MHz ( 11) | -40.00 dBm | -40.00 dBm
 garage (repeater) | 5765 MHz (153) |      —     | -95.00 dBm
```

* "garage (backhaul)" is an EnGenius ENS202EXT (Atheros AR9341) running OpenWrt 22.03.5. There's a bunch of stuff on the straight line between it and the Fenvi adapter: an interior wall, a gypsum-board ceiling, a ceramic tile floor, a fiberglass bathtub, another interior wall, a wooden chest of drawers, and about 45 feet of distance.
* "privileged" is an ALFA AWUS036AXML (MediaTek MT7921AU) with an [ALFA ARS-WiFi6E-M2 tri-band omni-directional indoor antenna](https://store.rokland.com/collections/wifi-6-6e/products/alfa-ars-wifi6e-m2-2-4ghz-5-ghz-6e-tri-band-omnidirectional-indoor-antenna) — an antenna that I am *not at all* impressed by, especially given its price. It is sitting in a closet of the room above the Fenvi adapter, with only a gypsum-board ceiling, carpeted floor, wooden closet door, and about 20 feet of distance between them.
* "IoT" is an ALFA AWUS036ACM (MediaTek MT7612U) with its stock antennas, located in the same place as "privileged." It blasts out a *much* stronger signal than the AWUS036AXML.
* "garage (repeater)" is a TP-Link RE220 (MediaTek MT7628AN+MT7610E) repeater, configured to connect to "garage (backhaul)" using its 2.4-GHz radio (MT7628AN) and to extend it using its 5-GHz radio (MT7610E). I can't see this network from where the Fenvi is with any other Wi-Fi adapter I own, as there are two interior walls, a glass mirror, an exterior wall, a pickup truck's engine bay, another exterior wall, and about 80 feet of distance between the adapter in my living room and the repeater out in the middle of my detached garage. And this is 5 GHz we're talking about here, which barely punches through anything solid. I doubt I'd actually be able to connect to the network, but just being able to successfully decode the beacons is amazing enough.

Snuupy, 2023-09-22:
I bought 3 for a total of 25usd (aliexpress promo going on) to re-start my rpi portable travel router project, hopefully this will work (unlike those pesky ANDDEAR 7612U adapters that didn't work last time)

 `MaxPower              100mA` is great to see too so I don't have to worry about hitting the 1.2A limit, will report back in a few weeks when I get it

morrownr, 2023-09-22:
@whitslack 

> it seems to have ridiculously good receive reception

My comment here is separate but somewhat related. I have done a few tests with my mt7921au adapters and other adapters. Maybe it is the part of the universe I live in or something I don't understand but even when I see the mt7921au adapters with the same or slightly lower signal reading, they do as good or better when throughput is tested in the same situation. Thoughts?

> ridiculously good receive reception

Also on this topic. A few years ago I bought an Alfa ACHM. I used it for a few months and then one day I was doing a range test with several adapters. I had to redo things several times because I could believe how the ACHM showed range that was so much longer than the other adapters. The Alfa ACH was second place. Alfa does list these two adapter as "MAX Power" so I guess they are putting good quality anteenas and amps in them.

Keep that info coming. We are interested. Thanks.

morrownr, 2023-09-22:
@Snuupy 

> MaxPower              100mA

That is hard to believe. Where is my usb power meter?

I have 2 adapters with the mt7921au chipset and I know how to torture an adapter and I just can't heat them up. How did Mediatek do this?

Snuupy, 2023-09-22:
> > MaxPower              100mA
> 
> That is hard to believe. Where is my usb power meter?
> 
> I have 2 adapters with the mt7921au chipset and I know how to torture an adapter and I just can't heat them up. How did Mediatek do this?

I don't know if that `MaxPower` line is lying or not, I ordered a cheap $2 usb tester on aliexpress, I am just reading what @whitslack posted under `lsusb`

mediatek for their phones has tsmc fabbed 4nm SoCs for their dimensity line which are also very power efficient and have high performance (comparable to flagship qualcomm snapdragon SoCs) so it would not surprise me if usb wifi adapters are made with the same/similar fab process for the transistors that are being used

SoCs for phones include: CPU, RAM, **wifi card**

maybe not leading edge, but 1 or 2 generations behind (since these adapters are so cheap after all, but I think the amount of silicon these usb sticks need is also much lower than a cpu, for example)

If we are getting better process nodes for denser/more efficient transistors by using more up to date lithography processes then of course we will get much better efficiency and lower heat. Have you ever used a AMD 7nm laptop compared to an intel 14nm laptop? The difference is huge in efficiency and performance.

morrownr, 2023-09-27:
Hi @Snuupy 

Thanks for the info.

I finally found my usb power meter! Here are the readings:

Adapter: Alfa AXML
270-300 mA when running iperf3
55-60 mA when idle

When surfing, the numbers sometimes got up to around 100-120 mA but surfing is not really so stressful.

Those are good numbers for those needing to limit power usege and will certainly be welcomed by the RasPi users but it seems to indicate the 100 mA figure provided by lsusb is not entirely accurate. The thermal readings I have seen with my 2 adapters that use the mt7921au chipset are very impressive. I test with extended periods using iperf3 and I have seen no thermal problems at all. This is the same test that will allow adapters with some chipsets and limited ventilation to cook an egg.

Snuupy, 2023-09-28:
for the power meter does data transfer as normal? or do you have to get a specific model?

looks like with 1.2A we can get 3 or 4 simultaneous adapters :D

Anthony96922, 2023-09-28:
I have two of these and both work on my Raspberry Pi 4.

Here is one taken apart:
[image]
Closeup of the chip:
[image]

List of all AC caps from my version of iw:
```
...
                VHT Capabilities (0x339071b2):
                        Max MPDU length: 11454
                        Supported Channel Width: neither 160 nor 80+80
                        RX LDPC
                        short GI (80 MHz)
                        TX STBC
                        RX STBC: up to 1 stream
                        SU Beamformee
                        Compressed Steering Number of Beamformer Antennas: 4
                        Number of Sounding Dimensions: 1
                        MU Beamformee
                        Max A-MPDU Length: 1048575 (exponent: 7)
                        VHT Link Adaptation: No Feedback
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 0 Mbps
                VHT extended NSS: supported
```

whitslack, 2023-09-28:
I guess we know the foldable antennas aren't placebos now! 📶 Thanks for the info, @Anthony96922!

kasinjsh, 2023-10-07:
Hi.

I'm looking into buying this for my (ax) penetration testing needs on irregular basis, learning mainly so 2dbi antenna will be fine, but I'm wondering about modes needed for that and driver issues at this point. Have you tested monitor mode, injection and stability whit airodump-ng?

Anthony96922, 2023-10-07:
It appears to support monitor mode based on the output iw list.

morrownr, 2023-10-07:
@kasinjsh 

I have two adapters with the m7921au chipset. I have done basic monitor mode, injection and deauth testing and did not see any problems. I remember some reports from around July to September of 2022, not long after the adapters with this chipset start to ship. I remember one guy that does a lot of monitor mode work that said the driver works well in monitor mode. He said that it worked well in 6 Ghz.

A lot of Linux users now have adapters with this chipset and I am not seeing any complaints about monitor mode.

kasinjsh, 2023-10-07:
Tnx both, I already ordered this unit as it was in "super deal" and "3 units left" state, for 13eur not that much to loose.

PS: Great repo, lots of useful information about this topic. If this chipset is very good will order some hi gain, better quality unit.

morrownr, 2023-10-11:
@kasinjsh 

> Great repo, lots of useful information about this topic.

Thanks. Something like this site needed to happen and I get a lot of help from users. If you have time to do a review once you have used this adapter, people would appreciate it. I'll put it with the posting in the Plug and Play List.

> If this chipset is very good will order some hi gain, better quality unit.

The Alfa adapters in the Plug and Play list for this chipset both have removal antennas which opens up some options. We don't have a Alfa "MAX Power" adapter yet but hopefully one comes. We do have at least one, maybe two additional chipsets coming for USB adapters at some point in the next year. The new Mediatek mt7925 chipset driver is starting to go into the kernel with kernel 6.7. It is a full up WiFi 7. Another driver in the kernel before product is available. Woo Hoo! We could get the mt7922au chipset before that. It is WiFi 6e like the mt7921au but has 160 Mhz channel width capability. USB Wifi just keeps getting better on Linux.

kasinjsh, 2023-10-12:
> @kasinjsh
>
>
> Thanks. Something like this site needed to happen and I get a lot of help from users. If you have time to do a review once you have used this adapter, people would appreciate it. I'll [...]

As I will be using it for penetration testing and stuff like that I would not be able to comment on regular daily usage. But I can do speedtests and overall impression review, I have ax APs at home.

I would not rush to buy wifi7 at this point as standard is not finalized, but good to see timely development from vendors.

morrownr, 2023-10-12:
> But I can do speedtests and overall impression review, I have ax APs at home.

Anything you can add is appreciated.

> I would not rush to buy wifi7 at this point as standard is not finalized, but good to see timely development from vendors.

I don't think we will have the opportunity to rush out and buy new WiFi 7 USB adapters very soon as USB adapters always come after other products such as cards but we are likely looking at a situation like with the mt7921au chip in that the the driver will be in the kernel for a few months before the prodducts hit the shelves. I think the first time that ever happened was with the mt7921au. I've used usb wifi adapters for many years on Linux and it is the first time I remember it happening. Thanks Mediatek for making it happen.

panecki, 2024-03-26:
Hi, does anyone have an FCC ID or schematic?

Snuupy, 2024-07-10:
just a note that the FU-1800 (and most other adapters of this type) have a maximum transmission signal strength of 20 dBm I saw in the spec sheet

so they will never be as good as an actual AP that can follow regdb guidelines of 27 dBm for example

for $8? it's fine

but any more than that I'd start considering a real AP (or something with pcie) instead.

parkerlreed, 2025-02-26:
Would this be suitable for a dedicated PCVR AP? (Quest 3 on Arch Linux and ALVR)

morrownr, 2025-02-27:
Hi @parkerlreed 

> PCVR AP

I have no idea what that is. Can you enlighten me?

parkerlreed, 2025-02-28:
Using it as a dedicated AP for connecting a VR headset to (Like the Quest 3) 

So your connection would be direct to the computer and not through the normal network which may be worse performing.

morrownr, 2025-02-28:
> Using it as a dedicated AP for connecting a VR headset to (Like the Quest 3)

Okay, now I understand. You want to use a USB WiFi adapter as a dedicated wireless access point for a VR headset.

Can USB WiFi adapters make good APs? Yes, but it can depend on the situation. Given that I am not a gamer, I would have to ask a lot of questions about the details of what you need in order to give good advice:

How far will this dedicated AP being from the VR headset?
What band? 2.4, 5 or 6?
What are the throughput requirements?
Ping speed requirements?
Amount of congestion on the band you want to use?
Will you be able to have line of site between the AP and VR headset?

That should be enough to help me develop more questions.

parkerlreed, 2025-02-28:
It would be fairly close (same room as the laptop)

I was looking at this adapter as it looks like it can do the AP at 6GHz? The headset supports 6E, so I was hoping to take advantage of that.

As long as the speed would be good for same room with minimal latency.

Not a lot of networks around here so interference wouldn't be an issue. 

I guess the overall question is how well is the AP mode working for this with Linux, assumingly at the 6GHz?

morrownr, 2025-02-28:
@parkerlreed 

Okay, so same room.

> I was looking at this adapter as it looks like it can do the AP at 6GHz? The headset supports 6E, so I was hoping to take advantage of that.

The adapter can do 6 GHz but a lot depends on what country you are in because the rules for 6 GHz are all over the place, depend on the laws of the country you are in. Tell me what country you live in.

> I guess the overall question is how well is the AP mode working for this with Linux, assumingly at the 6GHz?

Answer the above question about your country, then we can address this.

Let me reiterate the situation: You game and use a VR headset with a laptop that will be in the same room as the AP you plan to setup with a USB WiFi adapter? Right? Am I good so far?

The USB WiFi adapter will need to be plugged into something like a wifi router with a USB3 port (hopefully one that runs OpenWRT as that will make it possible). Or it could be plugged into something like a RasPi or, well, what are you planning for a system to run the adapter? And what operating system? And how will that system plug into your wifi modem or source of internet?

parkerlreed, 2025-02-28:
United States

This adapter would be plugging directly into said laptop, as to have the least amount of complexity.

AP would be created, headset would connect, and then I could talk TCP between the two.

Thanks.

morrownr, 2025-03-01:
> United States

Ouch! I feel your pain as that is where I am located too. Here is the deal. The below patch that was submitted this week takes No-IR off of the US in the reg db so using our adapters as APs in the US is now something we can do:

https://patchwork.kernel.org/project/linux-wireless/patch/000201db8822$98f28da0$cad7a8e0$@gmail.com/

However, we are still limited to a low txpower here in the US and when combined with 6 GHz being a higher frequency that won't penetrate much, 6 GHz is still a challenge. I've done some testing and 5 Ghz usually smokes 6 GHz. Until such time as we can crank the power up to something reasonable, 6 GHz is kind of a waste here. However 5 GHz is good, unless you have a problem with congestion.

> AP would be created, headset would connect, and then I could talk TCP between the two. Internet passthrough isn't strictly needed but that would be provided by the laptop's integrated card as it stands.

I do AP mode a lot but I'm a little confused here as to your setup. You have a laptop, a VR headset and you want a USB WiFi adapter. Can you clear it up for me exactly what is going where and in what mode? The USB WiFi adapter will be in AP mode? But yet it is connected to the laptop that you will play games on? Yes, I'm lost.

parkerlreed, 2025-03-01:
In this setup, the AP from computer to headset is a dedicated link that allows software like ALVR to connect and be able to play PCVR games. PC encodes and sends a video stream to the headset and controller/tracking data is sent back.

After all of this I found my old OpenWRT router that does 5GHz and it seems to do well, was just curious if I could make it smaller with a dedicated adapter. WiFi 6E/7 is something I haven't tried yet, so was looking for an excuse to give it a go.

Great to hear about the recent patches. Considering that and the low price of the adapter, I'll give this setup a shot and report back how it goes.

Thank you for the help.

morrownr, 2025-03-01:
> Considering that and the low price of the adapter, I'll give this setup a shot and report back how it goes.

Sure. Give it a shot. You may need to do 5 GHz for now as it takes time for patches to flow through.

If you need help with AP mode, I have a lot of experience with hostapd.