morrownr/USB-WiFi #312: How to configure Mediatek MT7610U to support channel 149?

fumin, 2023-10-02:
I bought a [ASUS AC51](https://www.amazon.com/Asus-Dualband-Wirel-AC600-USB-AC51/dp/B00T3DK154)
and found that it cannot connect to my router whose channel is 149.
Out of the box, this ASUS AC51, whose chip is Mediatek MT7610U,  can only go up to channel 136.
How can I change my configuration to make it support channel 149?

I see that people sometimes change their regulatory domain, like "US".
I am in Taiwan, how can I force my Ubuntu to change my ASUS AC51 domain to the US?

```
$ uname -a
Linux topunion-MS-7C96 6.2.0-33-generic #33~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Sep  7 10:33:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
$ lsusb
Bus 001 Device 006: ID 0b05:17d1 ASUSTek Computer, Inc. AC51 802.11a/b/g/n/ac Wireless Adapter [Mediatek MT7610U]
$ sudo iwlist wlx08bfb849ed0a channel
wlx08bfb849ed0a  32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 14 : 2.484 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
```

ZerBea, 2023-10-03:
Regulatory domain "country 00: DFS-UNSET" is the highest restriction. Nearly everything (frequencies, tx power and scan mode is restricted.
To get benefit of all capabilities of the devise, it is mandatory to change the regulatory domain (sudo iw reg set xx - where xx is teh country code.

morrownr, 2023-10-07:
@fumin 

I have two mt7610u adapters and I have no problem with channel 149. I don't think is an adapter problem. I think @ZerBea is headed the right direction. Something with the regdb or something closely related.

fumin, 2023-10-19:
@ZerBea @morrownr Thanks for your kind suggestions.
After doing what you said, my MT7610U wifi dongle is working smoothly.
Thanks!

closed 2023-10-19 by fumin