morrownr/USB-WiFi #313: error: mt76x02u_mcu_wait_resp failed with -108

mirkapmirkap, 2023-10-02:
#`error: mt76x02u_mcu_wait_resp failed with -108`
`MAC error detected`
`MAC stop failed`
`error: mt76x02u_mcu_wait_resp failed with -71`
`vendor request req:07 off:2310 failed:-71`

I have COMFAST CF-WU785ac on RPi B (usb2.0). Everything works great in one year. Suddenly stopped. My system with connected adapter reboot in loop. I touched nothing. After one million times testing and setting up new system again I don't now what's going on.
The problem occurs when I modeswitch.

dmesg before that:

> root@DietPi:~# dmesg
> [    0.000000] Booting Linux on physical CPU 0x0
> [    0.000000] Linux version 6.1.21+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 Mon Apr  3 17:19:14 BST 2023
> [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
> [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
> [    0.000000] OF: fdt: Machine model: Raspberry Pi Model B Rev 2
> [    0.000000] random: crng init done
> [    0.000000] Memory policy: Data cache writeback
> [    0.000000] Reserved memory: created CMA memory pool at 0x1ac00000, size 64 MiB
> [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
> [    0.000000] Zone ranges:
> [    0.000000]   Normal   [mem 0x0000000000000000-0x000000001effffff]
> [    0.000000] Movable zone start for each node
> [    0.000000] Early memory node ranges
> [    0.000000]   node   0: [mem 0x0000000000000000-0x000000001effffff]
> [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001effffff]
> [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
> [    0.000000] pcpu-alloc: [0] 0
> [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 125860
> [    0.000000] Kernel command line: coherent_pool=1M snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=0 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  root=PARTUUID=239f5e33-02 rootfstype=ext4 rootwait net.ifnames=0 logo.nologo console=tty1
> [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
> [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
> [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
> [    0.000000] Memory: 422732K/507904K available (8985K kernel code, 1450K rwdata, 2772K rodata, 432K init, 589K bss, 19636K reserved, 65536K cma-reserved)
> [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
> [    0.000000] ftrace: allocating 33370 entries in 98 pages
> [    0.000000] ftrace: allocated 98 pages with 3 groups
> [    0.000000] trace event string verifier disabled
> [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
> [    0.000007] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
> [    0.000073] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
> [    0.000161] bcm2835: system timer (irq = 27)
> [    0.000897] Console: colour dummy device 80x30
> [    0.001363] printk: console [tty1] enabled
> [    0.001441] Calibrating delay loop... 898.66 BogoMIPS (lpj=4493312)
> [    0.090355] pid_max: default: 32768 minimum: 301
> [    0.090564] LSM: Security Framework initializing
> [    0.090875] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
> [    0.090952] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
> [    0.092569] cgroup: Disabling memory control group subsystem
> [    0.093006] CPU: Testing write buffer coherency: ok
> [    0.094682] cblist_init_generic: Setting adjustable number of callback queues.
> [    0.094762] cblist_init_generic: Setting shift to 0 and lim to 1.
> [    0.095015] cblist_init_generic: Setting shift to 0 and lim to 1.
> [    0.095354] Setting up static identity map for 0x8220 - 0x8258
> [    0.096689] devtmpfs: initialized
> [    0.107414] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
> [    0.107847] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
> [    0.107933] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
> [    0.117591] pinctrl core: initialized pinctrl subsystem
> [    0.119552] NET: Registered PF_NETLINK/PF_ROUTE protocol family
> [    0.122400] DMA: preallocated 1024 KiB pool for atomic coherent allocations
> [    0.128360] audit: initializing netlink subsys (disabled)
> [    0.129949] thermal_sys: Registered thermal governor 'step_wise'
> [    0.130747] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
> [    0.130842] hw-breakpoint: maximum watchpoint size is 4 bytes.
> [    0.131100] Serial: AMBA PL011 UART driver
> [    0.136187] bcm2835-mbox 2000b880.mailbox: mailbox enabled
> [    0.139918] audit: type=2000 audit(0.120:1): state=initialized audit_enabled=0 res=1
[... 239 lines cut ...]
> [   11.582309] systemd[1]: Started systemd-journald.service - Journal Service.
> [   12.091008] EXT4-fs (mmcblk0p2): re-mounted. Quota mode: none.
> [   12.582634] systemd-journald[114]: Received client request to flush runtime journal.
> [   25.054814] smsc95xx 1-1.1:1.0 eth0: Link is Up - 100Mbps/Full - flow control off
> [   25.055146] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
> [   25.056766] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
> [   25.704257] usbcore: registered new interface driver uas
> [   26.499577] usb 1-1.3: USB disconnect, device number 5
> [   27.560973] usb 1-1.3: new high-speed USB device number 6 using dwc_otg
> [   27.693207] usb 1-1.3: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
> [   27.693269] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [   27.693292] usb 1-1.3: Product: 802.11ac WLAN
> [   27.693308] usb 1-1.3: Manufacturer: MediaTek Inc.
> [   27.693324] usb 1-1.3: SerialNumber: 000000000
> [   28.473547] cfg80211: Loading compiled-in X.509 certificates for regulatory database
> [   28.944666] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
> [   30.530922] usb 1-1.3: reset high-speed USB device number 6 using dwc_otg
> [   30.668301] mt76x2u 1-1.3:1.0: ASIC revision: 76120044
> [   30.720965] mt76x2u 1-1.3:1.0: ROM patch build: 20141115060606a
> [   30.936716] mt76x2u 1-1.3:1.0: Firmware Version: 0.0.00
> [   30.936776] mt76x2u 1-1.3:1.0: Build: 1
> [   30.936794] mt76x2u 1-1.3:1.0: Build Time: 201507311614____
> [   31.937058] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
> [   32.015503] usbcore: registered new interface driver mt76x2u
> root@DietPi:~#

morrownr, 2023-10-04:
I took a look around but came up with nothing. Not sure what that is. I have a 785 myself and it has been solid. I'm wondering if a cable has gone bad or maybe a power surge has been too much for the adapter. Do you have another computer, even Windows, to test it on?

mirkapmirkap, 2023-10-04:
Yes, I test it on win10, usb 2.0. Works perfect. Speed demon.

morrownr, 2023-10-04:
Okay.

Your best bet for reporting this might be the downstream OpenWRT MT76 repo:

https://github.com/openwrt/mt76

Some of the Mediatek devs that work on the Mediatek kernel drivers also work at this site and fixes flow back and forth to and from the mainline kernel. You could look in Issues for active and closed reports as well as active and closed PR's to see if anything similar has been reported or fixed. If you don't see anything, create your own issue there.

mirkapmirkap, 2023-10-06:
I maked research before and find not to much. But, I change power supply adapter and now works again like usually.
 Thanks morrownr, that was yours tip!

closed 2023-10-06 by mirkapmirkap

morrownr, 2023-10-06:
@mirkapmirkap 

That is good. 

USB WiFi adaptesr seem to be the Canary in the coal mine as they seem to let us know when there is an electrical issue before other parts of the system.