morrownr/USB-WiFi #317: Random crash in AP mode

fishBone000, 2023-10-09:
This problem is observed on all 3 of my COMFAST adapters: CF-952AX, CF-953AX and CF-924AC.  
When in AP mode, the interface crashes like this: 
```
wlan2: INTERFACE-DISABLED
wlan2: INTERFACE-ENABLED
```
Generally there's more log output around it, like:
```
Failed to set beacon parameters
wlan2: INTERFACE-DISABLED
wlan2: INTERFACE-ENABLED
wlan2: INTERFACE-DISABLED
wlan2: INTERFACE-ENABLED
wlan2: INTERFACE-DISABLED
wlan2: INTERFACE-ENABLED
nl80211: kernel reports: key not allowed
Failed to set beacon parameters
<repeats 4 times more>
```

I tested all 3 USB ports on my laptop, results are the same.  
The crash can happen when I open up a web page (low stress), watch vids (mid stress), or do speed tests (high stress).  

I suspect that it might be the stability problem of the USB hubs or the motherboard in my laptop.  
Any ideas?

closed 2023-10-10 by fishBone000

fishBone000, 2023-10-10:
I figured it out. 
It was NetworkManager that caused inteference. What a mistake of mine...

morrownr, 2023-10-10:
The Main Menu has a AP mode guide that gives guidance to turn NM off for a specific interface or to uninstall it completely. I generally use RasPiOS for my AP mode stuff and NM is off by default but if you use anything that has NM active, you have to handle that first. The guide also has a lot of information that can be useful including example hostapd.conf files for AC. I haven't included the one for AX yet as it is still in testing.