morrownr/USB-WiFi #318: MediaTek-based AC1200 USB adapter with VID_0E8D&PID_761B not working

duc5800, 2023-10-09:
Hi,

I have an EDUP EP-AC1605 USB wireless adapter here that I'd like to use with a computer running Linux Mint (kernel 6.20).

Since this adapter uses a MediaTek "MT7612UN" chip (determined by opening up the casing of my [EP-AC1605](https://ibb.co/3Mt1rVL)) I was hopeful that it would work, but it doesn't. The adapter is recognized as an attached USB device, but apparently no driver is loaded for it.

lsusb shows my device as "Bus 011 Device 002: ID 0e8d:761b MediaTek Inc. 802.11ac WLAN".

I already spent some time on this issue and I think the source of the problem is that the USB ID of my EDUP EP-AC1605 is not listed within the source code of the driver.

The file "https://github.com/torvalds/linux/blob/master/drivers/net/wireless/mediatek/mt76/mt76x2/usb.c" has an entry "USB_DEVICE(0x0e8d, 0x7612)", which should be MediaTek's generic USB ID for the MT7612U(N), but the adapter I've got somehow has the PID 761b.

Is there any way to make this work?

morrownr, 2023-10-09:
Hi @duc5800 

> Is there any way to make this work?

Method 1: Hotplug automation using udev.

Create a file called /etc/udev/rules.d/90-usb-0e8d:761b-mt7612u.rules

$ sudo nano /etc/udev/rules.d/90-usb-0e8d:761b-mt7612u.rules

Note: you can change nano to the text editor of your choice in the above command.

Copy the below lines and paste them into the above file that you are creating:

ACTION=="add", \
	SUBSYSTEM=="usb", \
	ENV{ID_VENDOR_ID}=="0e8d", \
	ENV{ID_MODEL_ID}=="761b", \
	RUN+="/usr/sbin/modprobe mt7612u", \
	RUN+="/bin/sh -c 'echo 0e8d 761b > /sys/bus/usb/drivers/mt7612u/new_id'"
Save file and reboot.

Method 2: From a terminal, enter and execute the following commands:

su
modprobe mt7612u
echo 0e8d 761b > /sys/bus/usb/drivers/mt7612u/new_id

Be aware that method 2 will need to be executed after each reboot.

If this works, it is time to submit a patch to linux-wireless to get this added.

Is this an adapters that currently for sale? If so, please forward a link.

Nick

duc5800, 2023-10-12:
Many thanks!

I've got it working with your instructions (needed to substitute mt7612u with mt76x2u) and have tested it for two days already.

For my purposes it's working really good.

However, one thing I noticed is that in Linux the signal strengths of all nearby wireless networks as indicated in the networking pop-up menu are quite a bit lower than what is shown in the Windows 10 that I can dual-boot on the same computer.

To make sure that isn't just an issue with the signal strength indicator I did a simple test. I placed a Wi-Fi router two rooms away from the computer with the MediaTek wireless adapter. Linux would not even consistently show me both SSIDs of that router, let alone connect to them. However, when Windows was booted up it indicated reasonably strong signals and established solid connections with the router.

For me this is no issue, as I don't need a good wireless range, but I wonder how other users are faring with the m76x2u driver?

I've had this EDUP EP-AC1605 lying around for several years already. Apparently it is not being sold anymore and EDUP appears to have, just like other manufacturers, switched from the MediaTek MT7612U to the Realtek RTL8812BU in later models of their 802.11AC USB wireless adapters.

morrownr, 2023-10-12:
> I've got it working with your instructions (needed to substitute mt7612u with mt76x2u)

Opps. Glad you figured it out.

> but I wonder how other users are faring with the mt76x2u driver?

It is not something that has caught my attention. I have 2 adapters with this chipset.