morrownr/USB-WiFi #319: i got the fenvi ax1800 on me what info u need about it ?

karimHI, 2023-10-11:


morrownr, 2023-10-11:
@karimHI 

Thanks.

A short review that covers some of the things that people who are shopping would like to know such as:

Have you had any problems with the adapter?
What is the vip/pid (ID)? $ lsusb
Is it single-state? Windows driver onboard?
Does it have good range?
Where did you order it? Any problems?
Thermal problems? Does it get hot?
Is it fast?
What bands have you tested? Any problems?
What distro are you using with it?
Does it work with extension cable, powered hub or right angle usb extension?
Cost?
What modes have you tested? Monitor, managed, AP, etc. Any problems?

Basically think about what you want to know if you are shopping. Does not have to be pretty. I can edit if that is okay with you.

Nick

karimHI, 2023-10-12:
id : ID 0e8d:7961 MediaTek Inc. Wireless_Device
it is plug and play in kali but in windows no it's comes with a tinny cd with the driver on it
Does it have good range? : Meduim
Where did you order it : aliexpress / Any problems? : no
Thermal problems? : from the adapters i tested before this is the coolest one i was surprised 
is it fast / what bands ? : everything is good and at described in the box (2.4ghz/5.0ghz)
distro: kali 2023.3 
kernel : 6.5.0-kali2-amd64
Does it work with extension cable : yes I'm using it with 2 extensions hub 
Cost? : 10$
What modes have you tested? :  i tested managed - ap/vlan - monitor everything is good
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * P2P-client
                 * P2P-GO
                 
                 
                 the only problem that i got is it is somehow stuck at 3dbm idk if it's just visually or it is really 
                 
[image]

morrownr, 2023-10-13:
@karimHI 

>   the only problem that i got is it is somehow stuck at 3dbm idk if it's just visually or it is really

As best we can tell that 'txpower 3.00 dBm' reading is not correct. It also happens with the mt7610u chip/driver. We have tested and tested and results appear within normal expectations so the number has to be bad. I guess we need to ask one of the Mediatek devs what's up with that. Maybe on will see this and answer as a couple do stop by at times.

Thanks for the info. I'll move it over to the stop for this adapter in the Plug and Play list. It sounds like you may have a pretty good adapter for 10 USD.

Nick

morrownr, 2023-10-13:
> everything is good and at described in the box (2.4ghz/5.0ghz)

Forgot to mention: The box may not say so but the adapter will do 6 Ghz also.

karimHI, 2023-10-13:
after some few researches I discovered that there's a problem with the driver of this chipset while trying to capture pkmid using hcxdumptool

karimHI, 2023-10-13:
https://github.com/ZerBea/hcxdumptool/issues/352

morrownr, 2023-10-13:
Thanks. Let me familiarize myself. What I am not seeing is an effort to put together a good bug report. The Mediatek devs are very busy these days with drivers for many SoCs and wifi chipsets to include the new mt7925 that is going into the kernel now (WiFi 7). They are not sitting around trying to find stuff to fix. A well done bug report that makes it easy for them to reproduce the issue is what is needed and then it should be submitted.

Keep in mind that we get usb chips from only 2 companies. The only one that we have a viable option to report bugs to is Mediatek.

karimHI, 2023-10-14:
do i need to install any extra driver to make 5ghz work cuz it's not working on kali linux but on wifislax it does ?

karimHI, 2023-10-14:
nevermind i installed the firmware and it's working now

morrownr, 2023-10-14:
The first thing that pops into my mind when when a 5 Ghz issue comes up is whether the country code is properly set and from there I would use my guide about firmware and make sure the very latest is installed.

karimHI, 2023-10-15:
is there any way to make this work on a distro that uses 4.14 kernel ?

morrownr, 2023-10-15:
> is there any way to make this work on a distro that uses 4.14 kernel ?

Sure. It may take hundreds of man-hours to make it happen but it is possible to do it. It is not something I am going to work on.

Backporting modern mac80211 drivers to a kernel that is several years old, all I can say is wow. What brings about the need for this?

curlycrit, 2023-10-16:
I purchased this adapter, and followed the instructions to copy the firmware to the firmware/mediatek directory. 

sudo reboot, and dmesg still returns `new SuperSpeed USB device number 2 using xhci_hcd`

uname -a `Linux kali-raspberry-pi 5.15.44-Re4son-v8l+ #1 SMP PREEMPT Debian kali-pi (2022-07-03) aarch64 GNU/Linux`

Any ideas how to fix this?

morrownr, 2023-10-16:
> Any ideas how to fix this?

Yes

> Linux kali-raspberry-pi 5.15.44-Re4son-v8l+ 

Kali has to get this distro to kernel 5.18 or later.

Read the Main Meni item called Plug and Play list. The first section on the mt7921au chipset provides a lot of information... like how firmware and driver must work together to support the adapter. You may have installed the firmware but you have no driver because your kernel is too old.

kasinjsh, 2023-10-18:
> id : ID 0e8d:7961 MediaTek Inc. Wireless_Device it is plug and play in kali but in windows no it's comes with a tinny cd with the driver on it Does it have good range? : Meduim Where did you order [...]

Happy to hear it, I bought it specifically for use whit kali, waiting for package to arrive.

morrownr, 2023-10-19:
>  the only problem that i got is it is somehow stuck at 3dbm idk if it's just visually...

This has to be a cosmetic problem as the actual performance does not match this number.

If you are using Kali, you might be interested in the following:

https://github.com/morrownr/USB-WiFi

Go to menu item 10, Monitor Mode...

I started this monitor mode project to help Realtek based adapter users because the drivers do not support doing some things the way they are supposed to work but the scripts work with in-kernel drivers also and you can modify them,,, customize them to what you want.

Nick

araknafobia, 2023-10-23:
@karimHI what is the maximum speed you managed to reach with the device? I got mine delivered today and I can't get it above 30 mbit/s. It almost feels like it's capped somehow! 

Just in case it useful here is my system information:

```
[code]
System:
  Kernel: 6.1.0-1024-oem x86_64 bits: 64 compiler: N/A Desktop: Cinnamon 5.6.8 tk: GTK 3.24.33
    wm: muffin dm: LightDM Distro: Linux Mint 21.1 Vera base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop System: Gigabyte product: B450M DS3H v: N/A serial: <superuser required>
  Mobo: Gigabyte model: B450M DS3H-CF v: x.x serial: <superuser required>
    UEFI: American Megatrends v: F41 date: 07/22/2019
Battery:
  Device-1: hidpp_battery_0 model: Logitech Wireless Keyboard ERGO K860 serial: <filter>
    charge: 55% (should be ignored) status: Discharging
  Device-2: hidpp_battery_1 model: Logitech MX Vertical Advanced Ergonomic Mouse
    serial: <filter> charge: 55% (should be ignored) status: Discharging
CPU:
  Info: 6-core model: AMD Ryzen 5 3600 bits: 64 type: MT MCP arch: Zen 2 rev: 0 cache: L1: 384 KiB
    L2: 3 MiB L3: 32 MiB
  Speed (MHz): avg: 3355 high: 4043 min/max: 2200/4208 boost: enabled cores: 1: 4016 2: 4027
    3: 4041 4: 4043 5: 2022 6: 2024 7: 3976 8: 4004 9: 4038 10: 4029 11: 2021 12: 2020
    bogomips: 86240
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm
Graphics:
  Device-1: AMD Baffin [Radeon RX 460/560D / Pro 450/455/460/555/555X/560/560X] vendor: ASUSTeK
    driver: amdgpu v: kernel pcie: speed: 8 GT/s lanes: 8 ports: active: DP-1
    empty: DVI-D-1,HDMI-A-1 bus-ID: 06:00.0 chip-ID: 1002:67ef
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: amdgpu,ati
    unloaded: fbdev,modesetting,vesa gpu: amdgpu display-ID: :0 screens: 1
  Screen-1: 0 s-res: 3840x2160 s-dpi: 96
  Monitor-1: DisplayPort-0 mapped: DP-1 model: BenQ EW2780U res: 3840x2160 dpi: 163
    diag: 685mm (27")
  OpenGL: renderer: AMD Radeon RX 560 Series (polaris11 LLVM 15.0.7 DRM 3.49 6.1.0-1024-oem)
    v: 4.6 Mesa 23.0.4-0ubuntu1~22.04.1 direct render: Yes
Audio:
  Device-1: AMD Baffin HDMI/DP Audio [Radeon RX 550 640SP / 560/560X] vendor: ASUSTeK
    driver: snd_hda_intel v: kernel pcie: speed: 8 GT/s lanes: 8 bus-ID: 06:00.1 chip-ID: 1002:aae0
  Device-2: AMD Starship/Matisse HD Audio vendor: Gigabyte driver: snd_hda_intel v: kernel pcie:
    speed: 16 GT/s lanes: 16 bus-ID: 08:00.4 chip-ID: 1022:1487
  Sound Server-1: ALSA v: k6.1.0-1024-oem running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet vendor: Gigabyte driver: r8169
    v: kernel pcie: speed: 2.5 GT/s lanes: 1 port: f000 bus-ID: 04:00.0 chip-ID: 10ec:8168
  IF: enp4s0 state: down mac: <filter>
  Device-2: MediaTek Wireless_Device type: USB driver: mt7921u bus-ID: 1-6:2 chip-ID: 0e8d:7961
  IF: wlx90de80d684ad state: up mac: <filter>
  IF-ID-1: br-4f16bc2b10a7 state: down mac: <filter>
  IF-ID-2: br-82bd4ccc4401 state: down mac: <filter>
  IF-ID-3: br-9daee90625ab state: down mac: <filter>
  IF-ID-4: br-e52e5db09ad6 state: down mac: <filter>
  IF-ID-5: docker0 state: down mac: <filter>
  IF-ID-6: wg-mullvad state: unknown speed: N/A duplex: N/A mac: N/A
Bluetooth:
  Device-1: Cambridge Silicon Radio Bluetooth Dongle (HCI mode) type: USB driver: btusb v: 0.8
    bus-ID: 3-2:2 chip-ID: 0a12:0001
  Report: hciconfig ID: hci0 rfk-id: 0 state: up address: <filter> bt-v: 2.1 lmp-v: 4.0
    sub-v: 22bb
Drives:
  Local Storage: total: 1.82 TiB used: 1.02 TiB (55.9%)
  ID-1: /dev/sda vendor: Western Digital model: WDS200T2B0B size: 1.82 TiB speed: 6.0 Gb/s
    serial: <filter>
Partition:
  ID-1: / size: 457.28 GiB used: 315.96 GiB (69.1%) fs: ext4 dev: /dev/sda1
  ID-2: /boot/efi size: 279.33 GiB used: 8.9 MiB (0.0%) fs: vfat dev: /dev/sda2
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 0 KiB (0.0%) priority: -2 file: /swapfile
[... 26 lines cut ...]
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/jcfp-nobetas-jammy.list
    1: deb http: //ppa.launchpad.net/jcfp/nobetas/ubuntu jammy main
  Active apt repos in: /etc/apt/sources.list.d/lutris-team-lutris-jammy.list
    1: deb http: //ppa.launchpad.net/lutris-team/lutris/ubuntu jammy main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb https: //mirrors.evoluso.com/linuxmint-packages/linuxmint-packages vera main upstream import backport
    2: deb https: //mirrors.evoluso.com/ubuntu jammy main restricted universe multiverse
    3: deb https: //mirrors.evoluso.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb https: //mirrors.evoluso.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
  Active apt repos in: /etc/apt/sources.list.d/plexmediaserver.list
    1: deb [arch=amd64 signed-by=/usr/share/keyrings/plexmediaserver.gpg] https: //downloads.plex.tv/repo/deb public main
  No active apt repos in: /etc/apt/sources.list.d/skype-stable.list
  No active apt repos in: /etc/apt/sources.list.d/team-xbmc-ppa-focal.list
  Active apt repos in: /etc/apt/sources.list.d/ubuntu-mozilla-daily-ppa-focal.list
    1: deb http: //ppa.launchpad.net/ubuntu-mozilla-daily/ppa/ubuntu jammy main
  Active apt repos in: /etc/apt/sources.list.d/vscode.list
    1: deb [arch=amd64,arm64,armhf] http: //packages.microsoft.com/repos/code stable main
Info:
  Processes: 385 Uptime: 13m Memory: 31.28 GiB used: 3.62 GiB (11.6%) Init: systemd v: 249
  runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12/9 Client: Unknown python3.10 client inxi: 3.3.13
[/code]

```

morrownr, 2023-10-23:
> Device-1: 1-6:2 info: MediaTek Wireless_Device type: Network driver: mt7921u rev: 2.1
    speed: 480 Mb/s chip-ID: 0e8d:7961

The adapter is running in USB2 mode. Did you plug it into a USB3 port?

We probably need info about your AP/WiFi router to do much more.

@morrownr

karimHI, 2023-10-23:
I got a 50mbps on me and I'm getting solid 54mbps from it in either linux and windows maybe ur missing smthn

araknafobia, 2023-10-23:
@morrownr I was trying different ports and the last one I tried was apparently usb2 :) 

I made sure that it's connected to usb3 now and I also adjusted the direction of the device now I am getting 80mbps! And yes, angel of the antennas and where the device is looking at/located makes a huge difference! This things is a beast! :+1:

morrownr, 2023-10-23:
@araknafobia 

You can check usb mode with:

$ lsusb -t

480 = usb2
5000 = usb3

>This things is a beast! 

Only the best for us Linux users!

morrownr, 2023-10-23:
@araknafobia 

If you want to tweak it...

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.127 port 33600 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  92.8 MBytes   778 Mbits/sec    0   1.49 MBytes       
[  5]   1.00-2.00   sec  92.5 MBytes   776 Mbits/sec    0   1.80 MBytes       
[  5]   2.00-3.00   sec  95.0 MBytes   797 Mbits/sec    0   1.89 MBytes       
[  5]   3.00-4.00   sec  92.5 MBytes   776 Mbits/sec    0   1.89 MBytes       
[  5]   4.00-5.00   sec  91.2 MBytes   765 Mbits/sec    0   1.89 MBytes       
[  5]   5.00-6.00   sec  92.5 MBytes   776 Mbits/sec    0   1.89 MBytes       
[  5]   6.00-7.00   sec  93.8 MBytes   786 Mbits/sec    0   1.89 MBytes       
[  5]   7.00-8.00   sec  93.8 MBytes   786 Mbits/sec    0   1.89 MBytes       
[  5]   8.00-9.00   sec  93.8 MBytes   786 Mbits/sec    0   1.89 MBytes       
[  5]   9.00-10.00  sec  92.5 MBytes   776 Mbits/sec    0   1.89 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   930 MBytes   780 Mbits/sec    0             sender

```

karimHI, 2023-10-24:
also another problem with it rather that the latest hcxdumptool isn't working
you can't deauth people while evil twin attack is running

morrownr, 2023-10-24:
> also another problem with it rather that the latest hcxdumptool isn't working

Have you talked to @ZerBea ?

karimHI, 2023-10-24:
no since it has nothing to do with hcxdumptool this error occur while launching an evil twin attack using a tool like airgeddon
the deauthing isn't working anymore no matter with aireplay or mdk3 
some ppl told me that there's a problem with vif

ZerBea, 2023-10-24:
A VIF could be a real problem. That is mentioned in --help:
```
        do not set monitor mode by third party tools (iwconfig, iw, airmon-ng)
        do not use logical (NETLINK) interfaces (monx, wlanxmon, prismx, ...) created by airmon-ng and iw
        do not use virtual machines or emulators
        do not run other tools that take access to the interface in parallel (except: tshark, wireshark, tcpdump)
        do not use tools to change MAC (like macchanger)
        do not merge (pcapng) dump files, because this destroys assigned hash values!
```

BTW:
Is hcxdumptool's REASSOCIATION attack still working?

morrownr, 2023-10-24:
What is a VIF? 

If there is not a bug report about this, maybe it is time to put one together.

How would i duplicate this?

ZerBea, 2023-10-24:
VIF == Virtual Interface

ZerBea, 2023-10-24:
@karimHI 
Please notice that neither aireplay nor mdk3 take care about Management Frame Protection (MFP).
If MFP is activated on target side, stupid injecting DEAUTHENTICATION or DISASSOCIATION frames will not have an effect.

TeddyShmosby, 2024-03-20:
Hello! Hopefully you're still there :D Can you please tell me if you can use it as a hotspot? I'm having issues activating it for some reason

morrownr, 2024-03-20:
@TeddyShmosby 

> Can you please tell me if you can use it as a hotspot?

It might be best if you start a new issue as this one is old.

Please include what adapter you have, your distro and exactly what it is you are trying to do.

@morrownr