morrownr/USB-WiFi #320: Regarding CF-952AX, I can provide some information

cneyi, 2023-10-12:
The 952AX comes in two different versions 
[image]

The difference between them is that the V2 version is driver-free
------------------
Also, I want to buy mt7921au for my kali virtual machine.
Do you still need some information about COMFAST?
I think I can help you
**Translated from google translate**
[http://en.comfast.com.cn/index.php?m=content&c=index&a=lists&catid=13&itype=1&mbps=1300&page=1&newkeys=itype](url)
[https://comfastsz.en.alibaba.com/productgrouplist-928296598/WiFi_6.html](url)

GeGerardo, 2023-10-19:
Hi @morrownr 

I will use this issue to give some feedback regarding CX-952AX:

---
General notes:
- It is much smaller than the CX-953AX. I am trying to open it to check what is inside
- The antennas are shorter than 953AX
- I have 2 units and a collegue has also 2 units. The LED does not work in any of them
- Using airodump I had some instances where the device turned off.

---

> ID :  ID 3574:6211 MediaTek Inc. Wireless_Device
> It is plug and play in kali not in windows or MacOS. Windows worked using the generic driver
> Where did you order it : [Aliexpress](https://www.aliexpress.com/item/1005005897678080.html?spm=a2g0o.order_list.order_list_main.35.157818022A33lB)
> Kernel : 6.5.0-kali2-amd64
> Does it work with extension cable : Yes
> Cost? : 15$
> What modes have you tested? : Managed, AP and Monitor

---

```bash
iwconfig |& grep -Eo 'Tx-Power=.*dBm'
Tx-Power=3 dBm
```
---

Using iPerf from a Mac pro 2023 -> Kali in a network with Eeero 6 I had almost the same results as CX-953AX

```bash
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-30.00  sec   765 MBytes   214 Mbits/sec                  receiver
[  8]   0.00-30.00  sec   789 MBytes   221 Mbits/sec                  receiver
[ 10]   0.00-30.00  sec   666 MBytes   186 Mbits/sec                  receiver
[ 12]   0.00-30.00  sec   606 MBytes   169 Mbits/sec                  receiver
[ 14]   0.00-30.00  sec   472 MBytes   132 Mbits/sec                  receiver
[SUM]   0.00-30.00  sec  3.22 GBytes  1.06 Gbits/sec                  receiver
```

<details>
  <summary>iw list</summary>
  
  ```bash
  Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * P2P-client
                 * P2P-GO
        Band 1:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
[... 619 lines cut ...]
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * mesh point: 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-device: 0x40 0xd0
        Supported extended features:
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
                * [ BEACON_RATE_HT ]: HT beacon rate setting
                * [ BEACON_RATE_VHT ]: VHT beacon rate setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ ACK_SIGNAL_SUPPORT ]: ack signal level support
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ CAN_REPLACE_PTK0 ]: can safely replace PTK 0 when rekeying
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
                * [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
  ```
</details>

morrownr, 2023-10-19:
Thanks to both of you for the info.

I have resisted adding the 952ax to the Plug and Play list for 2 primary reasons:

- I have not been that comfortable with the sellers that are available. If you have a link and can make a case that the experience was good, that will help.

- The fact that there is a v2 that is multi-state (Windows driver is onboard) is not good because many retailer are sloppy and users may get a v2. I try to warn Linux users away from multi-state adapters as that can cause problems. If I had a retailer to list where it is very sure that users will only get v1, I can see the adapter being listed.

As both of you use the adapters, give me your thoughts on whether the 952 should be in the Plug and Play list. I am very picky with the Plug and Play list. I have to be convinced that the adapters listed give Linux users the highest probability of satisfaction.

Nick

GeGerardo, 2023-10-19:
Taking that in consideration it is better to put it out. That seller I put he only sells the v1. But you know, maybe one day it will be a v2.

What I like on the 952 is the size.

By the way, sorry for the stupid question: what is the problem with v2 and driver onboard @morrownr ?

morrownr, 2023-10-20:
> By the way, sorry for the stupid question: what is the problem with v2 and driver onboard?

I don't think there is such a thing as a stupid question around here. This wifi stuff can be complicated. I started this site to help with things like you are asking.

The problem with adapters that have the onboard Windows driver is that it can cause problems. Linux has a good utility called usb_modeswitch that generally handles the switch without problem in most cases but not always. Main Menu item 7 has more info:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Modeswitch.md

I've helped a lot of usb with wifi issues over the last several years. Buying adapters that are multi-state (with Windows driver onboard) is just one more little things that can cause problems and we have enough single-state adapters that are available that it is an issue to avoid.

The problem I see with listing the CF-952AX is that retailers won't accurately show whether the adapter is v1 or v2 and most retailers won't care what they stock.

I am very picky about what goes into the Plug and Play list. There are makers and retailers that I will not list because it may not be a good experience for Linux users. I hope you understand.

Nick

bjlockie, 2023-10-21:
Isn't v1 mediatek but v2 is a realtek chipset?

morrownr, 2023-10-21:
> Isn't v1 mediatek but v2 is a realtek chipset?

As best I can tell at this point, both use the mt7921au chipset and the only difference is v1 is single state and v2 is multi-state.

Many Windows users have problems with anything beyond putting adapters in a usb port. It may not be the best usb port but they can usually find a usb port. Then they turn into a tech support nightmare and we end up with multi-state adapters.

Netgear, with their A8000 adapter, handled this Windows driver issue about as well as it can be handled. Other makers SHOULD take note. Netgear included a very small capacity (cheap) usb flash drive with the Windows driver. I have yet to see a complaint. I suspect most adapter makers are not rocket scientist so we end up with a problem.

I'd love to include the 952AX in the Plug and Play list because it is cute and I hear good things. Why did Comfast not use a different model number like Alfa and Panda ALWAYS do?

bjlockie, 2023-10-21:
https://www.aliexpress.com/item/1005006115196167.html?spm=a2g0o.productlist.main.13.2dbc6jom6jomhw&algo_pvid=1b623e69-b443-49bd-89d8-16354058bc96&algo_exp_id=1b623e69-b443-49bd-89d8-16354058bc96-6&pdp_npi=4%40dis%21CAD%2133.89%2123.73%21%21%2124.14%21%21%40211b5e2216978649360811489e1930%2112000035821412330%21sea%21CA%214077304562%21&curPageLogUid=XovjJuPDcoRP

RTL8832BU

CF-952AX V2

Anything that is "driverless" seems to be Realtek in my experience.

cneyi, 2023-10-21:
> > v1不是mediatek，v2是realtek芯片组吗？
> 
> 据我目前所知，两者都使用 mt7921au 芯片组，唯一的区别是 v1 是单状态，v2 是多状态。
> 
> 除了将适配器插入 USB 端口之外，许多 Windows 用户还遇到其他问题。它可能不是最好的 USB 端口，但他们通常可以找到 USB 端口。然后它们变成了技术支持的噩梦，我们最终得到了多状态适配器。
> 
> Netgear 及其 A8000 适配器很好地解决了这个 Windows 驱动程序问题。其他制造商应该注意。Netgear 附带了一个带有 Windows 驱动程序的非常小容量（便宜）的 USB 闪存驱动器。我还没有看到投诉。我怀疑大多数适配器制造商都不是火箭科学家，所以我们最终遇到了问题。
> 
> 我很乐意将 952AX 纳入即插即用列表中，因为它很可爱，而且我听到了很多好消息。为什么 Comfast 不像 Alfa 和 Panda 那样使用不同的型号？

What I learned from comfast here in China are:
952 (unspecified version) 7921au requires driver
953 7921au requires driver
951 7921au requires driver
However
962 (8832bu) 955v2 965 957v2 are all driver-free, and are most likely reltek model wireless network cards.
----------
I have a guess that 952v2 has been renamed and continues to be sold.
Proof is easy, just have a 952v2
I think it is unlikely that comfast will use a different wireless network card chip on the 952 because its sales volume is very large. And the comfast Chinese official website has shown that there is no channel to purchase 952v2.
But I hope I can mark it in the article and remind readers that the v2 version may not be 7921au
In fact, there is another point. I don’t think Comfast’s technology is very good. Making a drive-free 7921au may be a bit difficult for them. By the way, have you received feedback that 951/2/3 (7921au from comfast) has a plug-and-play version?

cneyi, 2023-10-21:
[image]
translate:
I would like to ask if this 962ax is from 7921au?
no
rtl8852’s?
no
Then I don’t want to guess. Can you give me an accurate answer?
rtl8832bu
So what does 7921au have?
(952, 951, 953)
okok

bjlockie, 2023-10-21:
The 962 looks to be Realtek:
https://fccid.io/OYR-CF-962AX/User-Manual/User-manual-6685940

The 951 and 953 are probably Mediatek.

The 952 could be either (both have been seen).
https://github.com/morrownr/USB-WiFi/issues/137