morrownr/USB-WiFi #322: TEROW ROW02CD on Newagg is actually TEROW ROW12CD

lauchokyip, 2023-10-12:
I purchased from the link provided but after I received the product, it turned out to be using rtl8812bu not mt7612u chipset.

morrownr, 2023-10-13:
@lauchokyip 

> We should take the link down to prevent people buying the wrong product again.

The link is gone.

My humble apologies but it seems you understand that at much as I try, some resellers will make changes without notice. I do my best to use links to really dependable sources so newegg is on the blacklist now.

At least there is a reasonable good in-kernel driver for that chipset these days in case you keep the adapter. USB support for that chipset went into kernel 6.2.

Nick

lauchokyip, 2023-10-13:
> @lauchokyip
>
>
> The link is gone.
>
> My humble apologies but it seems you understand that at much as I try, some resellers will make changes without notice. I do my best to use links to really [...]

Thanks for your fast response. I appreciate you keeping the lists maintained so just want to help out others as well!

Will close this issues

closed 2023-10-13 by lauchokyip