morrownr/USB-WiFi #323: ALFA AWUS036AXM usb wifi adapter generating xhci_hcd transfer error

culliard, 2023-10-13:
Setup:

- Ubuntu 22.04 (kernel 6.3.5) guest running on VirtualBox VM
- MacOS (Big Sur) Host
- ALFA AWUS036AXM connected via USB (Mediatek MT7921aun chipset). 

The VM detects the Wifi adapter fine

```
$ lsusb
Bus 002 Device 012: ID 0e8d:7961 MediaTek Inc. Wireless_Device
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

When I run airodump-ng, it does work (although not convinced it is stable/reliable), however there are many error being generated identical to:

```
$ dmesg | grep xhci
[44519.918798] xhci_hcd 0000:00:0c.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 8 comp_code 13
[44519.918858] xhci_hcd 0000:00:0c.0: Looking for event-dma 00000000160e5bb0 trb-start 00000000160e5bc0 trb-end 00000000160e5bf0 seg-start 00000000160e5000 seg-end 00000000160e5ff0
```

If I use another ALFA adapter, AWUS036AC (with [aircrack driver](https://github.com/aircrack-ng/rtl8812au)) then there are no errors

Appreciate any suggestions of how to debug or what to try.

Thanks.

morrownr, 2023-10-13:
Hi @culliard

I am working on rebuilding one of the test systems in my lab today and I can add a partition to do some testing regarding this issue. Here are some questions that will help:

- What version of VirtualBox are you using?
- Did you follow a guide to properly setup VirtualBox and the usb port settings in VirtualBox? If so, can I have a link?
- What airodump-ng commands are you using?

With this info, I will try to duplicate what you are seeing so that I have a better idea of what is going on. I can't go fast due to health issues but will do my best. The errors you are showing are showing up as usb problems so this could be an adventure.

Nick

culliard, 2023-10-13:
Hi Nick

To answer your questions:
- Virtualbox 7.0.10 r158379 (which is the latest)
- to add usb devices this [guide ](https://www.makeuseof.com/windows-virtualbox-add-usb/) works
- command: `airodump-ng -a --channel 1-165 <interface>` 
- to swap the kernel, I followed this [guide](https://linux.how2shout.com/how-to-install-linux-kernal-5-19-on-ubuntu-22-04-or-20-04/) and selected 6.3.5.

BTW the same error also happens with Windows host and the errors may also appear even before running airodump

I'm wondering if the same happens with native ubuntu install but I'm not able to test that. If there is anything more I can provide, please let me know.

Thanks a lot for your help!

bjlockie, 2023-10-13:
Does USB pass through go through a virtualbox software filter?
I wonder too if it happens on native systems.
A wifi 5 device works.
Is the wifi 6 device connecting to a wifi 5 router?
Are both devices connected at USB 3 speeds (maybe the wifi 5 device is 
connecting  via USB 2 (on a USB 3 port) and the wifi 6 device is trying 
to use USB 3?

morrownr, 2023-10-13:
> I wonder too if it happens on native systems.

I tested that within 5 minutes of getting the answers from @culliard . The answer is no. Test system:

- amd64 i7
- LMDE 6 (Debian 12 plus Cinnimon)
- kernels 6.1 and 6.4
- wifi adapter is Alfa AXML (same chipset as AXM)

I was not running a vm, this is bare metal. This is my main dev box.

This result suggests the problem may not be in the mt7921au driver but that is not for sure. Adding a vm to a system and trying to run usb wifi adapters via the vm complicates things many fold. If a user is looking to learn, this can be a hard knocks learning experience but if you are just wanting to learn pen testing and security analysis, get a cheap little dedicated machine like a RasPi so you can run bare metal or dual boot so you can run bare metal.

culliard, 2023-10-13:
Thanks for the test and info. Unfortunately a VM is required in our solution. I guess I will raise a query with VirtualBox.

morrownr, 2023-10-13:
@culliard 

My answer to @bjlockie was not meant to send you to VB as that was just a quick bare metal test to establish a baseline. However, it might be a good idea to check in with VB to see if there is information available. I'll go ahead and set up to do a test with VB on a test system. It just won't happen fast as I mentioned.

I don't know much about your requirement but if base metal doesn't work and dual boot doesn't work, what about using Linux for the host and Windows as the client?

ZerBea, 2023-11-11:
Please take a look at this report:
https://bugzilla.kernel.org/show_bug.cgi?id=202541

morrownr, 2023-11-12:
@ZerBea 

Yes sir, that is the issue from hell. I think I remember reading it at the time.

Some observations that I have had over the years:

- USB WiFi adapters and their drivers get blamed for a LOT of things that are not their fault.
- USB3 is not mankind's greatest invention.
- USB2 is far more stable than USB3.
- USB WiFi adapters tend to have less support because they are categorized as a HOME product.

That observation is one of the prime reasons that I started this site. After starting this site and watching the analytical data, I can add some more observations:

- USB WiFi is used extensively outside of the HOME market.
- USB WiFi is used for critical servers far more places than might be expected.
- There are a lot of USB WiFi adapters that are poorly made and poorly supported.
- Misinformation in ads and forums about USB WiFi is the norm.
- Bad design, such as the RasPi4B USB subsystem, only supporting a total of 1200 mA of power, is a major contributor to problems.

Getting good information to users is of paramount importance. I think I need to add a paragraph to the mt7921au section in the Plug and Play list even though I already say it in other places:

Low cost adapters may work well depending on your use case but low cost adapters do use low cost components and will not perform well in many use cases. 

In the stickied post, I am doing some testing to try to help get the bottom of an issue. To see the issue, you have to scroll to the bottom and look at the last few messages. I've been testing today. I have tried to get close to the hardware the OP's were using but I started testing with an Alfa AXML. The AXML is in AP mode, 5 GHz, and the client is showing a signal of about -70. Damn if the AXML is not hanging in there and producting stable results while I pound the hell out of it over time. I can't get it to go down.

I do have one low cost mt7921au based adapter, cf-951ax, but I'l either have to rearrange things or use my notebook computer because the range is far less than the AXML. Testing will continue but will take time.

@morrownr

ZerBea, 2023-11-12:
I agree because I've had the same experience.
Luckily hcxdumptool does not rely on upper bands or a high transmission rate. A (WPA-PSK) CLIENT can't resist to connect.
We are talking about layer 2 attacks only, because it is not acting on higher layers (e.g. like Evil Twin). And it is not acting on layer 1, because this need to modify the firmware and the driver.

To compare the AXML with an ACM and an ACHM, I ordered such a device.
With regard to a penetration testing task, the ACM is on the first rank, followed by the ACHM. Let's see how the AXML performs.

BTW:
I solved the USB3 problem by using a simple USB2 hub (quick and dirty solution, but it is working as expected).

morrownr, 2023-11-12:
@ZerBea 

> To compare the AXML with an ACM and an ACHM, I ordered such a device.

Something interesting that I have noticed about the AXML is that the actually performance in managed and AP modes is better that what you would expect given the dBm signal you will see. It took a while for this to click with me. I haven't check it with monitor mode as far as signal goes. I did do some basic monitor mode checking like injection and it worked. I would like to see your results.

ZerBea, 2023-11-13:
I started the test series here:
https://github.com/ZerBea/hcxdumptool/discussions/361

ZerBea, 2023-11-13:
Logilink 0151
ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter

driver: rt2800usb (Linux stock kernel driver)

```
328 packet(s) captured
1 SHB written to pcapng dumpfile
1 IDB written to pcapng dumpfile
1 ECB written to pcapng dumpfile
25 EPB written to pcapng dumpfile

exit on EAPOL M1M2
bye-bye

real	0m15,383s
user	0m0,016s
sys	0m0,031s
```

Important notice:
After first plugin, the device has to be plugged out and again to be plugged in.
Otherwise no packets will be received (neither tshark nor Wireshark nor hcxdumptool).